openhab/org.openhab.binding.zwave #810: Failing unit tests

kaikreuzer, 2018-01-19:
See https://openhab.ci.cloudbees.com/job/openHAB2-Bundles/259/org.openhab.binding$org.openhab.binding.zwave/testReport/

cdjackson, 2018-01-19:
This is very strange given there have been no changes to code for a month!

kaikreuzer, 2018-01-19:
Don't ask me - but it needs to be fixed anyhow ;-)

cdjackson, 2018-01-19:
Agreed :)

I  think it's probably trigged by something changing in ESH as it's failing to create a thing in a test. I probably won't get the chance to look at this tonight but will check it out tomorrow.

```
ThingType thingType = ThingTypeBuilder.instance("bindingId", "thingTypeId", "label").build();
Thing thing = ThingFactory.createThing(thingType, new ThingUID(thingType.getUID(), "thingId"), new Configuration());
```

cdjackson, 2018-01-20:
The issue seems to be triggered by a change in Eclipse - for now I've just ignored the tests as I can't debug as tests are not able to be run in the IDE (see #812).

closed 2018-01-20 by cdjackson