openhab/org.openhab.binding.zwave #812: Tests no longer run in IDE

cdjackson, 2018-01-20:
@kaikreuzer it looks like following #753 it is no longer possible to run tests in the IDE. I've only just noticed this since the development version still has the JARs included so works fine, and since there have been no changes in the master branch for a couple of months, this hasn't come up yet.

Are you able to run tests in the IDE?

The following is the error I get on every test. I've tried searching around the net, but not found any solution that works (we already use the ```mockito-all``` dependancy which seems to be the recommended solution).

```
java.lang.NoClassDefFoundError: org/objenesis/ObjenesisStd
	at org.mockito.internal.creation.jmock.ClassImposterizer.<init>(ClassImposterizer.java:36)
	at org.mockito.internal.creation.jmock.ClassImposterizer.<clinit>(ClassImposterizer.java:29)
	at org.mockito.internal.util.MockCreationValidator.isTypeMockable(MockCreationValidator.java:17)
	at org.mockito.internal.util.MockCreationValidator.validateType(MockCreationValidator.java:21)
	at org.mockito.internal.creation.MockSettingsImpl.validatedSettings(MockSettingsImpl.java:133)
	at org.mockito.internal.creation.MockSettingsImpl.confirm(MockSettingsImpl.java:127)
	at org.mockito.internal.MockitoCore.mock(MockitoCore.java:50)
	at org.mockito.Mockito.mock(Mockito.java:1243)
	at org.mockito.Mockito.mock(Mockito.java:1120)
	at org.openhab.binding.zwave.test.internal.protocol.commandclass.ZWaveCommandClassTest.getCommandClass(ZWaveCommandClassTest.java:109)
	at org.openhab.binding.zwave.test.internal.protocol.commandclass.ZWaveConfigurationCommandClassTest.getValueMessage_WriteOnly(ZWaveConfigurationCommandClassTest.java:42)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:497)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.eclipse.jdt.internal.junit4.runner.JUnit4TestReference.run(JUnit4TestReference.java:86)
	at org.eclipse.jdt.internal.junit.runner.TestExecution.run(TestExecution.java:38)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.runTests(RemoteTestRunner.java:538)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.runTests(RemoteTestRunner.java:760)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.run(RemoteTestRunner.java:460)
	at org.eclipse.jdt.internal.junit.runner.RemoteTestRunner.main(RemoteTestRunner.java:206)
Caused by: java.lang.ClassNotFoundException: org.objenesis.ObjenesisStd
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 34 more
```

cdjackson, 2018-01-21:
Ref https://github.com/eclipse/smarthome/issues/3918

cdjackson, 2018-01-25:
https://github.com/openhab/org.openhab.binding.zigbee/pull/69

cdjackson, 2018-01-25:
@htreu @davidgraeff I see you guys involved in solving this in ESH - do you have any ideas here? Following recent changes to the ZigBee and ZWave projects to remove the Mockito libraries I'm now unable to run tests in the IDE which is causing a lot of problems.

Any help with this would be appreciated as I've spent quite a bit of time trying to fix this. I could revert the change that @kaikreuzer made but I'd prefer not to do that if I can avoid it...

kaikreuzer, 2018-01-25:
@cdjackson I have spent several hours on this today, but not yet with a solution.
According to @sjka, it should work if you launch the tests as plug-in tests and not pure JUnit tests.
Unfortunately, we have PDE JUnit bundles in the openhab.target platform, which cause issues with launching Plug-in Tests from PDE... My hope is that I can resolve it by cleaning the target platform.
Please stay tuned...

cdjackson, 2018-01-25:
Thanks @kaikreuzer. I think I tried using plug-in tests as I saw that as the recommended solution in the ESH issue. I also completely reinstalled everything - including clearing all caches to try and get this working, and it didn’t seem to help so I’m not convinced cleaning the TP will help (but I’ll keep my fingers crossed for you :) ).

Thanks.

htreu, 2018-01-26:
@cdjackson what did the trick in ESH for running JUnit tests with mockito mocks is to provide

```
Require-Bundle: 
 org.hamcrest,
 org.junit,
 org.mockito,
 org.objenesis
```

in MANIFEST.MF. This will fix the classpath during test execution (Run As -> JUnit Test).

kaikreuzer, 2018-01-26:
Thanks @htreu - that's what I already did, but as mentioned above there is another issue with PDE+JUnit in general in the openHAB 2 IDE, which needs to be resolved...

closed 2018-02-05 by cdjackson