openhab/org.openhab.binding.zwave #832: Textual Thing stays in INITIALIZING status

kaikreuzer, 2018-02-19:
When creating a Z-Wave Thing via DSL like
```
zwave:fibaro_fgd212_03_005:test "My Test Device"
```
the Thing stays in INITIALIZING status forever. Instead, the handler is expected to set the status to UNKNOWN, ONLINE or OFFLINE immediately.

cdjackson, 2018-03-03:
I think the reason for this is that the required configuration is not set - ie you have not set the node ID. I expected that the system would not call the handler initialize() if this wasn't set, but it seems that's not the case here.

I think the reason is that the startup is checking the configuration for the ```thing-type``` and my config handler is providing configuration for the ```thing``` so it misses this check.

I'll look to update the config provider to always provide required config descriptions for the ```thing-type```.

closed 2018-03-03 by cdjackson

kaikreuzer, 2018-03-03:
> I think the reason is that the startup is checking the configuration for the thing-type and my config handler is providing configuration for the thing so it misses this check.

If this is really the case, you should definitely enter a bug for the ESH ThingManager about this! //cc @sjka

cdjackson, 2018-03-04:
I've opened and issue (https://github.com/eclipse/smarthome/issues/5180). For now at least, to get around this I'm handling ```thing-type``` configuration in my config provider as well as ```thing``` configuration.

PR #844 (2018-03-03): Remove dependancy on thing uid in config handler
Use thing configuration to establish node id rather than assuming this is in the uid.
Split the config response to now handle both ```thing``` and ```thing-type``` separately which ensures that the required configuration is correctly passed to the system before the thing is created.

Closes #833 
Closes #832