openhab/org.openhab.binding.zwave #834: [DEV] Battery devices with wakup set to 0 should not be initialized at startup or included in heal

openhab-5iver, 2018-02-20:
* OH2 Version: 2.3.0 snapshot 1212
* Binding Version: 2.3.0.201802082157

I'm not sure if the title of the issue describes it well. [Aeon Minimotes](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/74) have a default wakeup interval of 0. It is also a controller, so a heal doesn't seem to make sense. After each restart or heal, they are marked as Node initialising: REQUEST NIF. These devices can be woken up manually, which allows them to complete the initialization. I'm sure there are other devices that would behave the same. The devices function and I'm not sure how much of a performance hit this could be, but seems like it is something that could be cleaned up.

[image]

cdjackson, 2018-02-20:
How would you propose to initialise such devices? Until it is initialised, it doesn't know that it's a controller, or that the wakeup is 0.

I guess after an initial initialisation it could be skipped under such circumstances, but there does need to be an initialisation if there's no XML saved for the device - right?

mhilbush, 2018-02-20:
@cdjackson Recalling our discussion from [here](https://community.openhab.org/t/zwave-device-sensative-strip-11-01-011-not-recogniced/11111/172), there are a number of devices that default to a wakeup interval of 0. I thought we concluded (although I never actually tested it) that the initial inclusion into the network would cause the wakeup interval to be set to the default wakeup interval.

@openhab-5iver How recently was this device included into you network. From the discussion Chris and I had in the above thread, the wakeup interval should've been set to the default wakeup interval when the device was included into the network.

I think I have a brand new Minimote. I can try to test the inclusion behavior today.

> I guess after an initial initialisation it could be skipped under such circumstances, but there does need to be an initialisation if there's no XML saved for the device - right?

Notwithstanding the above...  Yes, it does need to be initialized once (and this device is a real PITA to initialize, requiring many presses of the `Join` button). However, once it is initialized and there is a node XML, the best thing to do might be to skip it. But, how would you know to skip it? An indicator in the database, or just use wakeupInterval==0?

cdjackson, 2018-02-20:
I think it's a different issue here. The previous discussion was about wether or not to set the wake up period - if I understand the request here it's about wether or not to do any initialisation at all. 

Maybe I'm wrong though?

mhilbush, 2018-02-20:
Ok. I thought the issues were related (both had to do with wakeup interval of 0), but maybe I linked them together unnecessarily. I reread this issue again, and I guess I'm not sure what @openhab-5iver is asking.

I do think, however, that there should be a strategy for devices that have a wakeup interval of 0. Currently, after a restart, these devices will perpetually show in HABmin that they're being initialized, which can be confusing. I have around 7 or 8 of these devices (Wallmotes, Buttons, Minimotes, etc.) that all have a default of 0. I'm pretty sure they default to 0 to conserve battery, so my preference is to leave them that way.

cdjackson, 2018-02-20:
> I do think, however, that there should be a strategy for devices that have a wakeup interval of 0. Currently, after a restart, these devices will perpetually show in HABmin that they're being initialized, which can be confusing

Well I think that is what this refers to - but that's different (I think) to the previous discussion which was more about setting the default...

For this, as I said, it's not possible to avoid initialising at least once. Otherwise we don't know the device, and we don't know that the wakeup is 0. In fact, following first initialisation, wakeup interval will be set to something non-0 anyway - as will the wakeup node. If you then set the period manually back to 0, then it will stay there. This is what we discussed in the previous thread and was shown in your logs if I remember correctly.

I think what is being asked here is to completely skip initialisation for devices that have a wakeup period of 0. This is possible, but only after this first initialisation, and only assuming that the XML file is not deleted - if it is, then it will have to reinitialise as again it won't know that the wakeup is 0.

Does that make sense?

mhilbush, 2018-02-20:
> I think what is being asked here is to completely skip initialisation for devices that have a wakeup period of 0. This is possible, but only after this first initialisation, and only assuming that [...]

Yes, this makes perfect sense to me.

> In fact, following first initialisation, wakeup interval will be set to something non-0 anyway - as will the wakeup node. If you then set the period manually back to 0, then it will stay there. This is what we discussed in the previous thread and was shown in your logs if I remember correctly.

I never actually verified this behavior because my node was already included in the network. The log I provided was only for a reinitialization of the node, which doesn't change the wakeup value. 

My memory might be a little foggy, but I thought we concluded that the wakeup value would be set only when the device is first included into the network; however, I never actually tested that scenario. I can test that scenario today, since I have a new Minimote.

cdjackson, 2018-02-20:
> The log I provided was only for a reinitialization of the node, which doesn't change the wakeup value.

The reinitialisation of the node does change the wakeup value. The reason that it didn't change the wakeup value in your log is because the wakeup node was already set to the controller so the binding knows that it's already been set and therefore leaves the value at 0.

> My memory might be a little foggy, but I thought we concluded that the wakeup value would be set only when the device is first included into the network

The binding doesn't know if a device is included for the first time or not, so it always sets the wakeup if it's not set (as above).

openhab-5iver, 2018-02-20:
Just noticed this was getting attention! Sorry I'm late...

> I guess after an initial initialisation it could be skipped under such circumstances, but there does need to be an initialisation if there's no XML saved for the device - right?
Yes. My thought was that after the first initialization, that devices with a wakeup interval of 0 should not need to be initialized again, because it would need to be maintained manually. For the Minimote, it is best for the Minimote to not be waking up on its own in order to conserve battery.

> @openhab-5iver How recently was this device included into you network.
It was back in November when Buydig was dumping them.

> I think what is being asked here is to completely skip initialisation for devices that have a wakeup period of 0. This is possible, but only after this first initialisation, and only assuming that [...]
Exactly what I was thinking.

PR #846 (2018-03-03): Don't initialise sleeping devices that are already initialised
This adds logic to not perform device initialisation if the device has previously been initialised, has the wakeup node set to the controller, and has the wakeup time set to 0.
Closes #834
closed 2018-03-03 by cdjackson