openhab/org.openhab.binding.zwave #870: [DEV] Alarm V2 alarm_entry channel updates with incorrect value during wakeup

openhab-5iver, 2018-04-09:
* OH2 Version: 2.3.0 snaPSHOT 1250
* Binding Version: 2.3.0.201804022210

I'm specifically seeing this on [Linear WADWAZ-1](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/9) window/door sensors, but I do not think this is device specific.

I recently changed all of my WADWAZ-1 sensors to use the alarm_entry channel, so that they could be setup as a Switch item like my motion sensors, so that I could have motion sensors and door sensors in the same group with state and aggregation function. Without changing anything else, I'm seeing all of my door sensors being updated to OFF (closed) after waking up, even though some are open. 

In this example, the node of interest is 68. [040918 alarmv2.zip](https://github.com/openhab/org.openhab.binding.zwave/files/1890463/040918.alarmv2.zip)

Here is the sensor being closed and opened...
[image]

Here is the sensor after a wakeup (manually initiated but same occurs when sensor initiates)...
[image]

The door is physically open, but after the wakeup, the GET is asking for the wrong thing or the binding is interpreting the response incorrectly. I haven't been able to decode these yet to determine where the error is. Before moving the contacts to alarm_entry, I noticed something similar happening with sensor_door, but did not put time into troubleshooting it, so this may not be limited to Alarm V2.

openhab-5iver, 2018-04-14:
I see two oddities. First, the reports sent while opening/closing include an Alarm Type of 07 and Z-Wave Alarm Type of 07. Should the GET include Alarm Type 07 instead of Type 00? I ask because the response is coming back for Alarm Type 00, like what was requested.
```
NODE 68: Sending REQUEST Message = 01 0B 00 13 44 04 71 04 00 07 25 94 64
```
```
Receive Message = 01 0F 00 04 00 44 09 71 05 00 00 00 FF 07 02 00 37
```
Second, I'm still reading the code to try to make better sense of it, but should one of these lines of code caught this...
https://github.com/openhab/org.openhab.binding.zwave/blob/development/src/main/java/org/openhab/binding/zwave/internal/converter/ZWaveAlarmConverter.java#L199
and
https://github.com/openhab/org.openhab.binding.zwave/blob/development/src/main/java/org/openhab/binding/zwave/internal/converter/ZWaveAlarmConverter.java#L204
... since the alarm_entry channel is configured for type=BURGLAR, event=2?

I'm surprised nobody else has reported this, but after I changed all of the similar door sensors in the database, my guess is people are probably using the sensor_door channel.

Could this just be the device not sticking to standards, or could there be something missing from the [AlarmV2](https://github.com/openhab/org.openhab.binding.zwave/pull/825/files) code for Alarm GET? I've now just modified the binding jar to include a sensor_binary channel so I could add these sensors to groups with the motion sensors. I haven't seen any issues so far, so this might be OK to add to the database, as an option for others. My concern was with having two channels with COMMAND_CLASS_BASIC. Here's the modified XML... [wadwaz1_0_0.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1910461/wadwaz1_0_0.txt). This doesn't fix THIS problem, but it's a workaround for what I was trying to achieve.

cdjackson, 2018-04-14:
I'll try and take a look at this tonight - it does look like V2 alarm implementation isn't complete for this request.

cdjackson, 2018-04-14:
Actually, the problem is a bit messier :( .

With V2, the binding sets V1 alarm to 0 - on the assumption that the device uses V2 alarm IDs. In this case it screws up the response.

I'll try using the V2 alarm type in the V1 code as well and we can see what that does.

ZWave is a lot messier than ZigBee when it comes to backward compatibility! :( ZigBee is generally pretty clean by using new command IDs, but ZWave uses the old command IDs and adds more data into the command which makes it hard to know which bits to use as every device can do its own thing!

openhab-5iver, 2018-04-14:
I've been poking at this for days, but it hasn't come completely clear as to where the issue was. I'm glad you're seeing it! I can't wait to see the code changes to understand it better. Anyhow, thank you!

I'll change one of the sensors back when you have an updated binding. Do you see any issue with adding the other channel to SENSOR_BINARY, with two channels set to Treat as Basic? I'm actually preferring switches vs contacts, since I can manually flick a switch to test or correct issues.

cdjackson, 2018-04-14:
I've put a version for you to test in the usual place (org.openhab.binding.zwave_2.3.0.201804141724).

I'll have a think about what happens if you add two links to basic - in general, I wouldn't have thought it a good idea, but it might be ok...

openhab-5iver, 2018-04-14:
```
2018-04-14 14:33:46.013 [DEBUG] [e.internal.protocol.commandclass.ZWaveCommandClass] - NODE 123: ArrayIndexOutOfBoundsException COMMAND_CLASS_ALARM V2 NOTIFICATION_REPORT 71 05 07 : null
```
Everything seemed OK, but I wasn't seeing any GETs, so I put the poll interval to 10m and let it sit for a while. It seems like maybe the queue was full of them?
[zwave1.log](https://github.com/openhab/org.openhab.binding.zwave/files/1910739/zwave1.log)
But then things seemed OK again. I'll let it sit a while and see if the error comes back. But there is no update for the alarm_entry channel...
[zwave2.log](https://github.com/openhab/org.openhab.binding.zwave/files/1910740/zwave2.log)
[zwave3.log](https://github.com/openhab/org.openhab.binding.zwave/files/1910746/zwave3.log)

openhab-5iver, 2018-04-14:
> I don't think this is binding related unless you can see something I've missed?
You may have posted before I edited my post. I do not see them being related to the change because I have these in the log before I updated the binding.

> Why do you think the queue was full of GETs? This shouldn't be possible (the queue shouldn't allow multiple requests of the same frame), and I don't see that in the log (there's only 3 GETs sent.
I just saw the repeated reports, so was assuming there were requests made for all of them, but I should have looked harder because there's only the one ALARM_GET.

So, the update you made solves the issue of the state getting set incorrectly, but the response to the GET is not updating alarm_entry at all. Is this what you were expecting?

cdjackson, 2018-04-14:
It’s not updating the state because the responses are corrupt. 

Can you delete the XML and get a debug log of the startup (or reinitialise the device so I get the initialisation logged). I want to check something during initialisation of this command class...

openhab-5iver, 2018-04-14:
> It’s not updating the state because the responses are corrupt.

I didn't think all of the responses were corrupt. From zwave3.log...
[image]
Here is a log from reinit... [zwave.log.zip](https://github.com/openhab/org.openhab.binding.zwave/files/1911454/zwave.log.zip). Next restart I'll delete the XML.

openhab-5iver, 2018-04-15:
I created a ticket on your website and submitted a log from an OH restart after deleting the XML.

cdjackson, 2018-04-15:
I didn’t see this one - in this case the device is responding using V1 report format. This is processed ok, but because the V1 alarm doesn’t contain the event, and the channel definition on the database DOES contain the event, it doesn’t update the state.  I don’t see an easy way around this since you need the event to decide if it’s a door or tamper event. I suspect we will have to go back to setting the V1 type to 0 and then look at why the response isn’t matching properly.

The log of the reinit didn’t have anything for node 122 - I’ll take a look at the one on the ticket once I’ve had a look at this V1/V2 report issue...

openhab-5iver, 2018-04-15:
This is probably unrelated, but maybe the change you did worked some magic... or it was there the whole time. I just got the external switch to report events through a channel! Too bad alarm_flood is opposite of what I need (on/open=OK, off/closed=Alarm). Is there any way to invert a binary channel? [wadwaz1_0_0.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1911826/wadwaz1_0_0.txt)

```
2018-04-15 03:25:48.799 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 10 00 04 00 7B 0A 71 05 07 00 00 FF 07 FE 00 00 EF 
2018-04-15 03:25:48.802 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=123, callback=0, payload=00 7B 0A 71 05 07 00 00 FF 07 FE 00 00 
2018-04-15 03:25:48.802 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - processReceiveMessage past lock Message: class=ApplicationCommandHandler[4], type=Request[0], dest=123, callback=0, payload=00 7B 0A 71 05 07 00 00 FF 07 FE 00 00 
2018-04-15 03:25:48.803 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=123, callback=0, payload=00 7B 0A 71 05 07 00 00 FF 07 FE 00 00 
2018-04-15 03:25:48.803 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - lastTransaction TID 1370: [WAIT_DATA] requiresResponse=true callback: 0
2018-04-15 03:25:48.803 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 123: Application Command Request (ALIVE:DONE)
2018-04-15 03:25:48.803 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 123: resetResendCount initComplete=true isDead=false
2018-04-15 03:25:48.803 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 123: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
2018-04-15 03:25:48.803 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 123: SECURITY not supported
2018-04-15 03:25:48.803 [DEBUG] [e.internal.protocol.commandclass.ZWaveCommandClass] - NODE 123: Received COMMAND_CLASS_ALARM V2 NOTIFICATION_REPORT
2018-04-15 03:25:48.803 [DEBUG] [ernal.protocol.commandclass.ZWaveAlarmCommandClass] - NODE 123: ALARM report - 7 = 0
2018-04-15 03:25:48.803 [DEBUG] [ernal.protocol.commandclass.ZWaveAlarmCommandClass] - NODE 123: Alarm Type = BURGLAR (7)
2018-04-15 03:25:48.803 [DEBUG] [ab.binding.zwave.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveAlarmValueEvent
2018-04-15 03:25:48.803 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 123: Got an event from Z-Wave network: ZWaveAlarmValueEvent
2018-04-15 03:25:48.803 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 123: Got a value event from Z-Wave network, endpoint = 0, command class = COMMAND_CLASS_ALARM, value = 0
2018-04-15 03:25:48.803 [DEBUG] [nding.zwave.internal.converter.ZWaveAlarmConverter] - Alarm converter processing ALARM
2018-04-15 03:25:48.803 [DEBUG] [nding.zwave.internal.converter.ZWaveAlarmConverter] - Alarm converter processing ALARM
2018-04-15 03:25:48.803 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 123: Updating channel state zwave:device:07cb40a2:node123:alarm_flood to OFF [OnOffType]
2018-04-15 03:25:48.804 [DEBUG] [nding.zwave.internal.converter.ZWaveAlarmConverter] - Alarm converter processing ALARM
2018-04-15 03:25:48.804 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 123: Commands processed 1.

2018-04-15 03:25:52.428 [DEBUG] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 10 00 04 00 7B 0A 71 05 07 FF 00 FF 07 FE 00 00 10 
2018-04-15 03:25:52.432 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=123, callback=0, payload=00 7B 0A 71 05 07 FF 00 FF 07 FE 00 00 
2018-04-15 03:25:52.432 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - processReceiveMessage past lock Message: class=ApplicationCommandHandler[4], type=Request[0], dest=123, callback=0, payload=00 7B 0A 71 05 07 FF 00 FF 07 FE 00 00 
2018-04-15 03:25:52.433 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=123, callback=0, payload=00 7B 0A 71 05 07 FF 00 FF 07 FE 00 00 
2018-04-15 03:25:52.433 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - lastTransaction TID 1370: [WAIT_DATA] requiresResponse=true callback: 0
2018-04-15 03:25:52.433 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 123: Application Command Request (ALIVE:DONE)
2018-04-15 03:25:52.433 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 123: resetResendCount initComplete=true isDead=false
2018-04-15 03:25:52.433 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 123: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
2018-04-15 03:25:52.433 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 123: SECURITY not supported
2018-04-15 03:25:52.434 [DEBUG] [e.internal.protocol.commandclass.ZWaveCommandClass] - NODE 123: Received COMMAND_CLASS_ALARM V2 NOTIFICATION_REPORT
2018-04-15 03:25:52.434 [DEBUG] [ernal.protocol.commandclass.ZWaveAlarmCommandClass] - NODE 123: ALARM report - 7 = 255
[... cut ...]
2018-04-15 03:25:52.436 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 123: Updating channel state zwave:device:07cb40a2:node123:alarm_flood to ON [OnOffType]
2018-04-15 03:25:52.437 [DEBUG] [nding.zwave.internal.converter.ZWaveAlarmConverter] - Alarm converter processing ALARM
2018-04-15 03:25:52.444 [DEBUG] [ng.zwave.internal.protocol.ZWaveTransactionManager] - NODE 123: Commands processed 1.
```

> The log of the reinit didn’t have anything for node 122

Is that a typo? The device is node 123, and there are some logs in there for it.

cdjackson, 2018-04-15:
Let's go right back to the beginning -:
The reason that the door is showing closed and not open is because the V1 status is showing 0, and the V2 status is showing FF. I think we previously decided that to keep this device happy, we needed to use the V1 alarm type? This was the change I made a month or so back - now it looks like it needs reverting in this case.

Can you revisit your previous analyses with this in mind? I seem to recall you had a spreadsheet of what was used where?

cdjackson, 2018-04-15:
Mails crossed...

> Is that a typo? The device is node 123, and there are some logs in there for it.

Yes, it was a typo - there was nothing in there for node 123.

cdjackson, 2018-04-15:
The log in the ticket had what I needed. It shows that this device does NOT support V1 alarms - so I will revert the change I made yesterday to request the V1 alarm in the GET and restore it to 0. This will put us back at the top of this issue, and we should then revisit the need to use the V1 alarm status in any of the status updates. As mentioned above - this reopens the question from https://github.com/openhab/org.openhab.binding.zwave/pull/825 <https://github.com/openhab/org.openhab.binding.zwave/pull/825> 2 months back where in order to solve another problem with this device, I believe we needed to use the V1 alarm. I might be wrong with my recollection though so can you double check - if it’s as simple as ignoring the V1 alarm in all circumstances, then I think we can find a solution…..

openhab-5iver, 2018-04-15:
> where in order to solve another problem with this device, I believe we needed to use the V1 alarm. I might be wrong with my recollection though so can you double check - if it’s as simple as ignoring the V1 alarm in all circumstances, then I think we can find a solution…..

This was one of the previous issues that you were addressing, but I recall there being another one related to AlarmV2 that I'm still looking for.
https://community.openhab.org/t/vision-door-windows-sensor-zd2102-open-close-and-tamper-are-not-triggered-to-different-events/27972/54?u=5iver
Everything was working with the changes you did back then, but it appears nobody tested the polling. Worst case, maybe not polling AlarmV2 is an option.

cdjackson, 2018-04-15:
Can you get a log of the response from polling when closed to complete the following table -:

```
Closed
71 05 07 00 00 FF 07 02 00 00

Opened
71 05 07 FF 00 FF 07 02 00 00

Polled when open
71 05 00 00 00 FF 07 02 00 00

Polled when closed
XX
```

openhab-5iver, 2018-04-15:
ALARM_GET:
01 0B 00 13 7B 04 71 04 07 07 25 DC 14
ALARM_REPORT:
01 0A 00 04 00 7B 04 71 05 07 00 FD

openhab-5iver, 2018-04-15:
OPEN
ALARM_GET: 01 0B 00 13 7B 04 71 04 00 07 25 55 9A
ALARM_REPORT: 01 0F 00 04 00 7B 09 71 05 00 00 00 FF 07 02 00 08

CLOSED
ALARM_GET: 01 0B 00 13 7B 04 71 04 00 07 25 D6 19
ALARM_REPORT: 01 0F 00 04 00 7B 09 71 05 00 00 00 FF 07 02 00 08

I hadn't deleted the Thing and the external switch still worked! Just need a good channel for it. Is there a secret configuration for a channel to invert the value? There was a poll added for it too, but I don't see a GET for it...

```
2018-04-15 06:28:05.151 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 123: Polling...
2018-04-15 06:28:05.151 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 123: Polling zwave:device:07cb40a2:node123:alarm_entry
2018-04-15 06:28:05.151 [DEBUG] [nding.zwave.internal.converter.ZWaveAlarmConverter] - NODE 123: Generating poll message for COMMAND_CLASS_ALARM, endpoint 0, alarm BURGLAR, event 2
2018-04-15 06:28:05.151 [DEBUG] [ernal.protocol.commandclass.ZWaveAlarmCommandClass] - NODE 123: Creating new message for application command NOTIFICATION_GET V2
2018-04-15 06:28:05.151 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 123: Encapsulating message, endpoint 0
2018-04-15 06:28:05.151 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 123: SECURITY not supported
2018-04-15 06:28:05.151 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 123: Command Class COMMAND_CLASS_ALARM is NOT required to be secured
2018-04-15 06:28:05.151 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 123: Polling zwave:device:07cb40a2:node123:alarm_flood
2018-04-15 06:28:05.151 [DEBUG] [nding.zwave.internal.converter.ZWaveAlarmConverter] - NODE 123: Generating poll message for COMMAND_CLASS_ALARM, endpoint 0, alarm BURGLAR, event 254
2018-04-15 06:28:05.151 [DEBUG] [ernal.protocol.commandclass.ZWaveAlarmCommandClass] - NODE 123: Creating new message for application command NOTIFICATION_GET V2
2018-04-15 06:28:05.151 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 123: Encapsulating message, endpoint 0
2018-04-15 06:28:05.151 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 123: SECURITY not supported
2018-04-15 06:28:05.152 [DEBUG] [.openhab.binding.zwave.internal.protocol.ZWaveNode] - NODE 123: Command Class COMMAND_CLASS_ALARM is NOT required to be secured
2018-04-15 06:28:05.152 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 123: Polling zwave:device:07cb40a2:node123:battery-level
2018-04-15 06:28:05.152 [DEBUG] [ing.zwave.internal.converter.ZWaveBatteryConverter] - NODE 123: Generating poll message for COMMAND_CLASS_BATTERY endpoint 0
```

openhab-5iver, 2018-04-15:
> So the GET here is correct, but the state is always the same.

Out of curiosity... what are the last three bytes in the GET? Well, the last one is checksum, but what are the other two? 

> I propose to simple disable GET requests in the database.

See! It's one of those secret channel configurations! It's unfortunate, but I'm not a fan of polling anyhow. I don't see how your V1 option would work with the external switch in play too, since they'd both be reporting through ALARM CC. I wonder what the response would be to Alarm Type FF...

I'll use alarm_entry for the reed switch and add the sensor_binary channel for the external switch in the database. And I'll add NOGET to alarm_entry... although, I should probably add it to both for good measure.

cdjackson, 2018-04-15:
> On 15 Apr 2018, at 16:00, Scott Rushworth <notifications@github.com> wrote:
> 
> Out of curiosity... what are the last three bytes in the GET? Well, the last one is checksum, but what are the other two?

From memory, it’s the  transmit options, transaction ID, and checksum.

openhab-5iver, 2018-04-15:
Database is updated. I put NOGET in all of the ALARM CC channels. Thank you for your help straightening this out!

openhab-5iver, 2018-04-16:
Chris, should the NOGET go into the ALARM CC configuration, or into the individual channels? I suspect the fomrer, because when I tested with the XML as it is now ([individual channels](www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/9)), I get the following error for every WADWAZ-1...
```
2018-04-16 02:21:58.529 [ERROR] [eclipse.smarthome.core.thing.internal.ThingManager] - Exception occurred during notification about bridge status change on thing 'zwave:device:07cb40a2:node86': 1
java.lang.ArrayIndexOutOfBoundsException: 1
        at org.openhab.binding.zwave.handler.ZWaveThingHandler.initialiseNode(ZWaveThingHandler.java:210) [15:org.openhab.binding.zwave:2.3.0.201804090748]
        at org.openhab.binding.zwave.handler.ZWaveThingHandler.bridgeStatusChanged(ZWaveThingHandler.java:488) [15:org.openhab.binding.zwave:2.3.0.201804090748]
        at org.eclipse.smarthome.core.thing.internal.ThingManager$4.run(ThingManager.java:815) [115:org.eclipse.smarthome.core.thing:0.10.0.201804081055]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) [?:?]
        at java.util.concurrent.FutureTask.run(FutureTask.java:266) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [?:?]
        at java.lang.Thread.run(Thread.java:748) [?:?]
```

cdjackson, 2018-04-16:
> should the NOGET go into the ALARM CC configuration, or into the individual channels?

It should be in the CC config.

openhab-5iver, 2018-04-16:
OK, database is corrected. That could have been a mess!

openhab-5iver, 2018-04-18:
Chris, I'm finding that the NOGET is not being saved to the XML. Using the new binding, I've deleted Things and done reinits of a WADWAZ-1, but the XML looks like this [network_f2b3b7e3__node_65.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1924830/network_f2b3b7e3__node_65.txt) (isGetSupported is set to true). The devices are also polling and being having their states set incorrectly. If I manually change it to false, the polling appears to behave properly. Would you like me to open another issue for this, or use this one?

cdjackson, 2018-04-18:
Did you delete the XML after you updated the binding? If not, it would not have implemented the changes from the database.

I just had a look at http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/9?layout=openhab2 <http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/9?layout=openhab2> and the info is being exported into the database XML so it should be ok. If it’s not working, then please get a log during the initialisation (ie after deleting the XML) so I can see if (or not) this is being applied.

openhab-5iver, 2018-04-18:
I've only done a reinit, but expect that deleting the XML and restarting would do the same. After a reinit, the XML has isGetSupported set to true... here is a log.
[node74_reinit.log](https://github.com/openhab/org.openhab.binding.zwave/files/1925045/node74_reinit.log)

cdjackson, 2018-04-18:
Yep - it’s not implemented!

I’ll update this in the next day or so.

openhab-5iver, 2018-04-18:
> I’ll update this in the next day or so.
Cool beans!

In looking at [that code](https://github.com/openhab/org.openhab.binding.zwave/blob/development/src/main/java/org/openhab/binding/zwave/internal/protocol/initialization/ZWaveNodeInitStageAdvancer.java#L716), it got me wondering if a check couldn't be added to make sure the CC was also in the nodeInformationFrame section. My XMLs get filled with inaccuracies that I think we deduced the cause to be from events being received with the wrong nodeID in them. Some of these WADWAZ XMLs had security, multilevel_switch, switch_all, etc. Verifying the CC would help, but not eliminate them, since I also saw things like access_control in the ALARM CC section. I'm not sure what the side effects are to having this bad info in the XML, but I periodically delete them because of it. More than half of my 22 WADWAZ had inaccuracies! Just a thought.

cdjackson, 2018-04-18:
I don’t think there is any need to add the CC to the NIF section. This is not used anywhere - it’s just provided as information for debugging.

openhab-5iver, 2018-04-18:
> I don’t think there is any need to add the CC to the NIF section.
I was meaning to verify that the CC already existed in the NIF before adding it elsewhere in the XML.

openhab-5iver, 2018-04-18:
[network_f2b3b7e3__node_86.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1925286/network_f2b3b7e3__node_86.txt)

Hopefully this archived XML will do a better job of explaining than I have! There are CCs added to endpoint 0 that are obviously not in the NIF. If a check was added to verify the CC in the NIF before they were put into the file, these could be eliminated. I don't know the harm of having them, so I don't know how useful that would be... but it should be a simple check.

cdjackson, 2018-04-18:
No - we can’t do that. Unfortunately, many devices don’t report all command classes they support in the NIF. This violates the ZW requirements, but hey-ho - it still happens. If we implement what I think you’re suggesting (ie only supporting CCs that a device reports in the NIF) then many devices will stop working.

openhab-5iver, 2018-04-18:
Point taken. Sorry for the tangent... well... actually, one for the road... what are the side effects of this? Deleting my XMLs and Things is part of my regular maintenance, but I doubt it is something others consider.

cdjackson, 2018-04-18:
Sorry - what are the side effects of what? You mean regularly deleting XMLs and things? If so, then no major side effects I think. Deleting the XMLs will mean that all devices need to complete the service discovery (interrogation) again which will clog the network for a bit. Deleting the things shouldn’t really matter too much so long as you have item/thing naming that’s always consistent so that everything restores afterwards…

Or did you mean something else?

openhab-5iver, 2018-04-18:
> Or did you mean something else?

I meant side effects of having a bunch of unsupported CCs in the XML, like that example. 

On second thought though, you've got the ccAdd option which could be used on a device that uses CCs that it does not include in the NIF. So maybe there _could_ be a verification step? But again, maybe there's no harm. :smile:

cdjackson, 2018-04-18:
Yes, I could always generate another list of command classes for use in this way. I can’t use the NIF, and wouldn’t add CCs to the NIF from the ccAdd option as the NIF is there solely to represent the NIF. I’m not sure it’s worth it though - I know to you it is because you have this issue with corrupted node IDs (or something strange anyway ;) ), but it adds complexity that probably isn’t needed for 99.9% of cases.

There shouldn’t be any major issue if extra command classes get added...

openhab-5iver, 2018-04-18:
> Yes, I could always generate another list of command classes for use in this way. I can’t use the NIF, and wouldn’t add CCs to the NIF from the ccAdd option as the NIF is there solely to represent the NIF

I don't see a need for another list. When adding or updating a CC in the XML where there is no ccAdd, check the NIF list first to make sure the CC is supposed to be in the XML. I still think you're misunderstanding what I'm suggesting, or I'm just talking _complete_ crap. But it doesn't really matter if it doesn't cause issues.

> There shouldn’t be any major issue if extra command classes get added...

This has occurred with the Aeon gen5 Zstick and the HUSBZB1. And without error checking on the stick, although the checksums are good so not sure how the stick would do a check, I suspect this is not isolated to my system and quite common, so I'm glad there shouldn't be any harm in it!

Cheers!

cdjackson, 2018-04-18:
Ok, I think what you’re suggesting is simply not to add nodes that aren’t either in the NIF, or in already added by the ccAdd command? This doesn’t need any check - it just needs some code removing to not add the command classes in the first place. I’ll have a think about this - maybe it’s ok, but I’d like to think about it some more before doing anything.

I would also assume that you aren’t alone with this error - other than I don’t think anyone else has reported it ;)

openhab-5iver, 2018-04-18:
> Ok, I think what you’re suggesting is simply not to add nodes that aren’t either in the NIF, or in already added by the ccAdd command?
Bing bing bing!

> I would also assume that you aren’t alone with this error - other than I don’t think anyone else has reported it ;)

The best bug is one that nobody ever knew existed! I'm probably imagining it, but my system seems to run a lot better with a fresh set of XMLs. There may not be any harm in them being in there, but maybe there's something else to it.

openhab-5iver, 2018-04-25:
> Yep - it’s not implemented! I’ll update this in the next day or so.

@cdjackson, is NOGET/isGetSupported still on your radar?

cdjackson, 2018-04-25:
It’s done in the latest version.

openhab-5iver, 2018-04-25:
TY!

openhab-5iver, 2018-05-06:
After testing this for over a week, the fix appears to be working. Would you prefer I leave this open as a reminder to look into CCs being incorrectly added to XMLs, or close this one and open a new issue for that?

As always, THANK YOU for your time and effort!

cdjackson, 2018-05-06:
Thanks.

Let's close this - please feel free to open another issue about not adding CCs. I'd prefer not to have that hidden in here - it's a simple change, but when someone complains I'd like to be able to remind myself why we changed it ;)

(ps did you get my email with a link to a test version with the UoM change?).

closed 2018-05-06 by cdjackson