openhab/org.openhab.binding.zwave #881: Supported, yet not supported?  TKB Home wall switch

Tontze, 2018-04-22:
Hi ! 

My first time with openhab here :) A problem with TKB home tz55s dimmer wall switch. It is not regognized, yet it has correct manufacturer id. When added as thing, it gives this:

**Z-Wave Node 15 (0118:0311:0201:2.4)
Unknown Device

This device has not been fully discovered by the binding. There are a few possible reasons for this -:
The device is not in the database. If the device attributes show that this device has a valid manufacturer ID, device ID and type, then this is likely the case (eg. you see a label like "Z-Wave node 1 (0082:6015:020D::2.0)"). Even if the device appears to be in the database, some manufacturers use multiple sets of references for different regions or versions, and your device references may not be in the database. In either case, the database must be updated and you should raise an issue to get this addressed.**

How do i get it supported, and in use ?)

cdjackson, 2018-04-22:
I’ve added it to the database, it will be added to the binding in the coming days.

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/516 <http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/516>

Tontze, 2018-04-22:
Well that was fast, tyvm :)

cdjackson, 2018-04-25:
If the devices have different type/id parameters then they will not be recognised as the same device. If they have different channels, then it would be a bad idea for them to be the same as, well, they aren't ;)

cdjackson, 2018-05-30:
Should be closed as above.

closed 2018-05-30 by cdjackson

sumitu25, 2018-06-09:
Hi,

Ive recently installed Z wave PIR Sensor from Neo. And the inclusion was successful with my Aeotec Wave Stick 5G. However in the openhab binding I see an incomplete message:
"NeoMSensor
Unknown Device

This device has not been fully discovered by the binding. There are a few possible reasons for this -:
The device is not in the database. If the device attributes show that this device has a valid manufacturer ID, device ID and type, then this is likely the case (eg. you see a label like "Z-Wave node 1 (0082:6015:020D::2.0)"). Even if the device appears to be in the database, some manufacturers use multiple sets of references for different regions or versions, and your device references may not be in the database. In either case, the database must be updated and you should raise an issue to get this addressed.
The device initialisation is not complete. Once the device is included into the network, the binding must interrogate it to find out what type of device it is. One part of this process is to get the manufacturer information required to identify the device, and until this is done, the device will remain unknown. For mains powered devices, this will occur quickly, however for battery devices the device must be woken up a number of times to allow the discovery phase to complete. This must be performed with the device close to the controller."

This node is online but I cannot see what more I can do. Please kindly assist. I also have another PIR sensor from Zooz (4-in-1 Sensor) and its the same story.

cdjackson, 2018-06-09:
Please discuss on the community forum where there will be more people able to support you.