openhab/org.openhab.binding.zwave #891: ZWave plus mandatory classes

aromanro, 2018-05-03:
In ZWavePlusDeviceClass.java, ZWavePlusDeviceClass, ZWavePlusDeviceType enum:

They appear to be in order, but at about line 81

```
        SEISMIC_INTENSITY_SENSOR(0x0d19, DEVICE_RESET_LOCALLY, ZWAVE_PLUS_INFO, ASSOCIATION_GROUP_INFO, MANUFACTURER_SPECIFIC, POWERLEVEL, ASSOCIATION, VERSION, SENSOR_BINARY, SENSOR_MULTILEVEL),
        SEISMIC_MAGNITUDE_SENSOR(0x0d1a, DEVICE_RESET_LOCALLY, ZWAVE_PLUS_INFO, ASSOCIATION_GROUP_INFO, MANUFACTURER_SPECIFIC, POWERLEVEL, ASSOCIATION, VERSION, SENSOR_BINARY, SENSOR_MULTILEVEL),
        ULTRAVIOLET_SENSOR(0x0d1b, DEVICE_RESET_LOCALLY, ZWAVE_PLUS_INFO, ASSOCIATION_GROUP_INFO, MANUFACTURER_SPECIFIC, POWERLEVEL, ASSOCIATION, VERSION, SENSOR_BINARY, SENSOR_MULTILEVEL),
        ELECTRICAL_RESISTIVITY_SENSOR(0x0d1c, DEVICE_RESET_LOCALLY, ZWAVE_PLUS_INFO, ASSOCIATION_GROUP_INFO, MANUFACTURER_SPECIFIC, POWERLEVEL, ASSOCIATION, VERSION, SENSOR_BINARY, SENSOR_MULTILEVEL),
        ELECTRICAL_CONDUCTIVITY_SENSOR(0x0b1d, DEVICE_RESET_LOCALLY, ZWAVE_PLUS_INFO, ASSOCIATION_GROUP_INFO, MANUFACTURER_SPECIFIC, POWERLEVEL, ASSOCIATION, VERSION, SENSOR_BINARY, SENSOR_MULTILEVEL),
        LOUDNESS_SENSOR(0x0b1e, DEVICE_RESET_LOCALLY, ZWAVE_PLUS_INFO, ASSOCIATION_GROUP_INFO, MANUFACTURER_SPECIFIC, POWERLEVEL, ASSOCIATION, VERSION, SENSOR_BINARY, SENSOR_MULTILEVEL),
        MOISTURE_SENSOR(0x0b1f, DEVICE_RESET_LOCALLY, ZWAVE_PLUS_INFO, ASSOCIATION_GROUP_INFO, MANUFACTURER_SPECIFIC, POWERLEVEL, ASSOCIATION, VERSION, SENSOR_BINARY, SENSOR_MULTILEVEL),
        FREQUENCY_SENSOR(0x0b20, DEVICE_RESET_LOCALLY, ZWAVE_PLUS_INFO, ASSOCIATION_GROUP_INFO, MANUFACTURER_SPECIFIC, POWERLEVEL, ASSOCIATION, VERSION...
```

The device type is basically counted there 0x0d19, 0x0d1a, 0x0d1b, 0x0d1c, but then, instead of 0x0d1d it's 0x0b1b and it continues from there. Is that correct? Shouldn't be there 0x0d.. too?
They seem to be in the same group, by the mandatory supported classes...

cdjackson, 2018-05-30:
NA at the moment for new version since these classes don't exist.

aromanro, 2018-11-08:
Despite that, I think it should be fixed, they might in the future.

Here is a very similar bug being confirmed for open zwave: https://github.com/OpenZWave/open-zwave/issues/1500

cdjackson, 2018-11-08:
At the moment, it makes no difference, and probably for any forseeable future it will not matter as it's fundamentally not the way the binding works, but please feel free to provide the update.