openhab/org.openhab.binding.zwave #893: [DEV] Error logged for qubino/zmnhod_0_0.xml

openhab-5iver, 2018-05-06:
OH snapshot 2.3.0 build 1272
Zwave 2.3.0.201804302009
device: http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/214#

I've noticed this in the last few snapshot builds, but did not note when it first started. I do not have this device. The following error is logged at startup:
```
2018-05-06 15:57:43.241 [WARN ] [smarthome.config.xml.osgi.XmlDocumentBundleTracker] - The XML document '/ESH-INF/thing/qubino/zmnhod_0_0.xml' in module 'org.openhab.binding.zwave' could not be parsed:  : ParseError at [row,col]:[174,9]
Message: /opt/openhab2/userdata/cache/org.eclipse.osgi/15/0/bundleFile :  : ParseError at [row,col]:[174,9]
Message: /opt/openhab2/userdata/cache/org.eclipse.osgi/15/0/bundleFile
---- Debugging information ----
message             :  : ParseError at [row,col]:[174,9]
Message: /opt/openhab2/userdata/cache/org.eclipse.osgi/15/0/bundleFile
cause-exception     : com.thoughtworks.xstream.io.StreamException
cause-message       :  : ParseError at [row,col]:[174,9]
Message: /opt/openhab2/userdata/cache/org.eclipse.osgi/15/0/bundleFile
class               : org.eclipse.smarthome.config.core.ConfigDescriptionParameter
required-type       : org.eclipse.smarthome.config.core.ConfigDescriptionParameter
converter-type      : org.eclipse.smarthome.config.xml.ConfigDescriptionParameterConverter
path                : /thing-descriptions/thing-type/config-description/parameter[7]
line number         : 174
class[1]            : java.util.ArrayList
converter-type[1]   : com.thoughtworks.xstream.converters.collections.CollectionConverter
class[2]            : org.eclipse.smarthome.config.core.ConfigDescription
converter-type[2]   : org.eclipse.smarthome.config.xml.ConfigDescriptionConverter
class[3]            : org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlResult
converter-type[3]   : org.eclipse.smarthome.core.thing.xml.internal.ThingTypeConverter
class[4]            : org.eclipse.smarthome.core.thing.xml.internal.ThingDescriptionList
converter-type[4]   : org.eclipse.smarthome.core.thing.xml.internal.ThingDescriptionConverter
version             : 1.4.7
-------------------------------
com.thoughtworks.xstream.converters.ConversionException:  : ParseError at [row,col]:[174,9]
Message: /opt/openhab2/userdata/cache/org.eclipse.osgi/15/0/bundleFile :  : ParseError at [row,col]:[174,9]
Message: /opt/openhab2/userdata/cache/org.eclipse.osgi/15/0/bundleFile
---- Debugging information ----
message             :  : ParseError at [row,col]:[174,9]
Message: /opt/openhab2/userdata/cache/org.eclipse.osgi/15/0/bundleFile
cause-exception     : com.thoughtworks.xstream.io.StreamException
cause-message       :  : ParseError at [row,col]:[174,9]
Message: /opt/openhab2/userdata/cache/org.eclipse.osgi/15/0/bundleFile
class               : org.eclipse.smarthome.config.core.ConfigDescriptionParameter
required-type       : org.eclipse.smarthome.config.core.ConfigDescriptionParameter
converter-type      : org.eclipse.smarthome.config.xml.ConfigDescriptionParameterConverter
path                : /thing-descriptions/thing-type/config-description/parameter[7]
line number         : 174
class[1]            : java.util.ArrayList
converter-type[1]   : com.thoughtworks.xstream.converters.collections.CollectionConverter
class[2]            : org.eclipse.smarthome.config.core.ConfigDescription
converter-type[2]   : org.eclipse.smarthome.config.xml.ConfigDescriptionConverter
class[3]            : org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlResult
converter-type[3]   : org.eclipse.smarthome.core.thing.xml.internal.ThingTypeConverter
class[4]            : org.eclipse.smarthome.core.thing.xml.internal.ThingDescriptionList
converter-type[4]   : org.eclipse.smarthome.core.thing.xml.internal.ThingDescriptionConverter
version             : 1.4.7
-------------------------------
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:79) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:65) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at com.thoughtworks.xstream.converters.collections.AbstractCollectionConverter.readItem(AbstractCollectionConverter.java:71) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at com.thoughtworks.xstream.converters.collections.CollectionConverter.addCurrentElementToCollection(CollectionConverter.java:98) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:91) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at com.thoughtworks.xstream.converters.collections.CollectionConverter.populateCollection(CollectionConverter.java:85) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at com.thoughtworks.xstream.converters.collections.CollectionConverter.unmarshal(CollectionConverter.java:80) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:65) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:66) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convertAnother(TreeUnmarshaller.java:50) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at org.eclipse.smarthome.config.xml.ConfigDescriptionConverter.unmarshal(ConfigDescriptionConverter.java:85) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at com.thoughtworks.xstream.core.AbstractReferenceUnmarshaller.convert(AbstractReferenceUnmarshaller.java:65) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
[... 53 lines cut ...]
        at com.thoughtworks.xstream.io.xml.AbstractPullReader.hasMoreChildren(AbstractPullReader.java:87) ~[?:?]
        at com.thoughtworks.xstream.io.ReaderWrapper.hasMoreChildren(ReaderWrapper.java:32) ~[?:?]
        at com.thoughtworks.xstream.io.ReaderWrapper.hasMoreChildren(ReaderWrapper.java:32) ~[?:?]
        at org.eclipse.smarthome.config.xml.util.ConverterValueMap.readValueMap(ConverterValueMap.java:87) ~[?:?]
        at org.eclipse.smarthome.config.xml.util.ConverterValueMap.<init>(ConverterValueMap.java:61) ~[?:?]
        at org.eclipse.smarthome.config.xml.util.ConverterValueMap.<init>(ConverterValueMap.java:46) ~[?:?]
        at org.eclipse.smarthome.config.xml.ConfigDescriptionParameterConverter.unmarshal(ConfigDescriptionParameterConverter.java:110) ~[?:?]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72) ~[?:?]
        ... 63 more
Caused by: javax.xml.stream.XMLStreamException: ParseError at [row,col]:[174,9]
Message: /opt/openhab2/userdata/cache/org.eclipse.osgi/15/0/bundleFile
        at com.sun.org.apache.xerces.internal.impl.XMLStreamReaderImpl.next(XMLStreamReaderImpl.java:599) ~[?:?]
        at com.thoughtworks.xstream.io.xml.StaxReader.pullNextEvent(StaxReader.java:58) ~[?:?]
        at com.thoughtworks.xstream.io.xml.AbstractPullReader.readRealEvent(AbstractPullReader.java:148) ~[?:?]
        at com.thoughtworks.xstream.io.xml.AbstractPullReader.readEvent(AbstractPullReader.java:135) ~[?:?]
        at com.thoughtworks.xstream.io.xml.AbstractPullReader.hasMoreChildren(AbstractPullReader.java:87) ~[?:?]
        at com.thoughtworks.xstream.io.ReaderWrapper.hasMoreChildren(ReaderWrapper.java:32) ~[?:?]
        at com.thoughtworks.xstream.io.ReaderWrapper.hasMoreChildren(ReaderWrapper.java:32) ~[?:?]
[... cut ...] ap.java:87) ~[?:?]
        at org.eclipse.smarthome.config.xml.util.ConverterValueMap.<init>(ConverterValueMap.java:61) ~[?:?]
        at org.eclipse.smarthome.config.xml.util.ConverterValueMap.<init>(ConverterValueMap.java:46) ~[?:?]
        at org.eclipse.smarthome.config.xml.ConfigDescriptionParameterConverter.unmarshal(ConfigDescriptionParameterConverter.java:110) ~[?:?]
        at com.thoughtworks.xstream.core.TreeUnmarshaller.convert(TreeUnmarshaller.java:72) ~[?:?]
        ... 63 more
```

openhab-5iver, 2018-05-07:
Here is another possibly related error from startup. The previous error seems to only appear after the first startup after an upgrade, but this one occurs thereafter. Maybe they are related?
```
2018-05-07 02:48:06.012 [WARN ] [smarthome.config.xml.osgi.XmlDocumentBundleTracker] - The XML document '/ESH-INF/binding/binding.xml' in module 'org.openhab.binding.zwave' could not be parsed: The XmlConfigDescriptionProvider must not be null!
java.lang.IllegalArgumentException: The XmlConfigDescriptionProvider must not be null!
        at org.eclipse.smarthome.core.binding.xml.internal.BindingInfoXmlProvider.<init>(BindingInfoXmlProvider.java:59) [105:org.eclipse.smarthome.core.binding.xml:0.10.0.201805051943]
        at org.eclipse.smarthome.core.binding.xml.internal.XmlBindingInfoProvider.createDocumentProvider(XmlBindingInfoProvider.java:141) [105:org.eclipse.smarthome.core.binding.xml:0.10.0.201805051943]
        at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.acquireXmlDocumentProvider(XmlDocumentBundleTracker.java:174) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.addingObject(XmlDocumentBundleTracker.java:199) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.parseDocuments(XmlDocumentBundleTracker.java:343) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.processBundle(XmlDocumentBundleTracker.java:329) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.access$3(XmlDocumentBundleTracker.java:324) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker$2.run(XmlDocumentBundleTracker.java:300) [101:org.eclipse.smarthome.config.xml:0.10.0.201805051943]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) [?:?]
        at java.util.concurrent.FutureTask.run(FutureTask.java:266) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180) [?:?]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293) [?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [?:?]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [?:?]
        at java.lang.Thread.run(Thread.java:748) [?:?]
```

cdjackson, 2018-05-07:
Strange that this database entry hasn't changed for 2 months...

https://github.com/openhab/org.openhab.binding.zwave/blob/development/ESH-INF/thing/qubino/zmnhod_0_0.xml#L174

The line is -:
```
        <label>74: Motor moving up/down time</label>
```

Col 9 is the word ```label```. I don't see why it doesn't like this - it's presumably fine just above in the previous parameter definition.

I also don't have the issue here - it's loading fine -:

```
2018-05-07 10:33:27.521 [DEBUG] [c.x.o.XmlDocumentBundleTracker:340  ] - Reading the XML document '/ESH-INF/thing/qubino/zmnhod_0_0.xml' in module 'org.openhab.binding.zwave'...
```

Not sure what else to suggest - I can't see any problem with the binding or this XML file...

openhab-5iver, 2018-05-20:
I have not seen this again with OH 2.3.0 snapshot 1281 and zwave 2.3.0.201805072038.

closed 2018-05-20 by openhab-5iver