openhab/org.openhab.binding.zwave #900: Support discovery of USB dongles

bodiroga, 2018-05-08:
Hi @cdjackson!

I know that you are busy, but it would be awesome if USB dongle discovery could be added to the ZWave binding, just like it is now in ZigBee binding. Perhaps @hsudbrock could give a helping hand here, as he has develop the USB discovery API :wink: I can try to implement it myself, but I'm not sure if I should push it to master or development branch. How far are we from merging both branches? I suppose that it's imposible to get it for openHAB 2.3 release.

Anyway, here you can find the properties of the two Aeon ZWave sticks:

- ZStick Gen5 -> Vendor_id: 0x0658, Model_id/Product_id: 0x0200
- ZStick Gen2 -> Vendor_id: 0x10c4, Model_id/Product_id: 0xea60

Thank you for the amazing work and best regards,

Aitor

cdjackson, 2018-05-08:
> I know that you are busy, but it would be awesome if USB dongle discovery could be added to the ZWave binding, just like it is now in ZigBee binding. Perhaps @hsudbrock <https://github.com/hsudbrock> could give a helping hand here, as he has develop the USB discovery API 😉 I can try to implement it myself, but I'm not sure if I should push it to master or development branch. How far are we from merging both branches? I suppose that it's imposible to get it for openHAB 2.3 release.

It’s not something I will likely do as I can’t test this. If you want to add it, then it must be added to the development version. It will not be in 2.3.

hsudbrock, 2018-05-09:
I have this already on my "to do sometime in the future" list, but it has not the topmost priority currently... (I have one of the Aeon sticks myself, so I will be able to test at least with that stick.)

If you want to try yourself at this, @bodiroga , you can have a look at the implementation for the ZigBee binding as (probably the only current) example using the USB dongle discovery (cf. https://github.com/openhab/org.openhab.binding.zigbee/pull/179).

If I would code this, I would probably do a pull request against master branch (but @cdjackson will be able to tell you against which branch to do a PR if this is wrong...)

cdjackson, 2018-05-09:
As above, please don't do a PR against master - use the development branch only for this sort of feature. The master will likely be dropped (or force merged or something!) in a few weeks after 2.3 is released - it's impossible to merge this so any significant changes will have to be manually ported if they are merged into master.

hsudbrock, 2018-05-09:
thanks @cdjackson (I must have misread your comment above somehow, although when reading it now your comment appears very clear to me)

bodiroga, 2018-05-09:
Hi guys!

I will try to get some time and mimic what you @hsudbrock did with the ZigBee binding, it should be quite similar. In my case, I can test with both Aeon sticks (Gen5 and Gen2).

Thanks for your help!

cdjackson, 2018-05-09:
Thanks @bodiroga

cdjackson, 2018-06-01:
#902

closed 2018-06-01 by cdjackson