openhab/org.openhab.binding.zwave #905: [DEV] Attempt to set association in group reported as not supported by the device.

sjw04, 2018-05-15:
Some devices, such as Fibaro Relays (eg FGS-222) report a different number of association groups between COMMAND_CLASS_ASSOCIATION (V2) and MULTI_INSTANCE_ASSOCIATION (V1).

In the case of the example device(s) there are 3 groups reported by COMMAND_CLASS_ASSOCIATION and 2 groups reported by MULTI_INSTANCE_ASSOCIATION - see [zwave.include.log](https://github.com/openhab/org.openhab.binding.zwave/files/2004456/zwave.include.log) around timestamp 21:28:54.

By default, if the device supports multiple endpoints, the current development uses MULTI_INSTANCE_ASSOCIATION to set the association groups, otherwise COMMAND_CLASS_ASSOCIATION is used.

The example device(s) uses association group 3 for controller updates. 

The ZWave spec for MULTI_INSTANCE_ASSOCIATION - V2 says both command classes should operate on the same groups. However this device uses V1 of the command class, which may have different rules (not publicly available).

When initialising the binding attempts to set the OpenHAB node in group 3 using MULTI_INSTANCE_ASSOCIATION for controller updates.  The group information is then read back.

In the example device case, this means setting the OpenHAB node in group 3, which is greater than the highest group the device reports as supported in MULTI_INSTANCE_ASSOCIATION, and the read back returns an empty group. Therefore controller updates are not sent, unless updates to other groups are done.

There could be multiple ways to work around this device behavior:
* Check the group being updated or read is supported by the command class being used.  If not try to falling back to the next command class (eg check MULTI_INSTANCE_ASSOCIATION, then check COMMAND_CLASS_ASSOCIATION, then fail to set).
* If only nodes are being set always use COMMAND_CLASS_ASSOCIATION. If endpoints are also being set then use  MULTI_INSTANCE_ASSOCIATION. Note: When the device supports multiple endpoints, the controller lifeline is always setup using a node and endpoint 1, however as OH2 does not appear to use the end point - so could be just a node?

Attached is a patch [group_range_check.zip](https://github.com/openhab/org.openhab.binding.zwave/files/2004469/group_range_check.zip) with a proof of concept for the first option above, and log [zwave.limit.log](https://github.com/openhab/org.openhab.binding.zwave/files/2004470/zwave.limit.log) with the patch included.

Thanks,
Steve.

cdjackson, 2018-07-01:
Thanks - I'll try and look over this soon.

For reference, the second option wouldn't work. It might work if we added a check for the versions (which could be an option of course). The problem is that on newer devices, using the MULTI_INSTANCE_ASSOCIATION is required in order to get multi-channel reports.

It's really a (big!) bit of a messy area in ZWave :(

sjw04, 2018-07-09:
I've just tried the dev binding (2.4.0.201807032158) and it appears to work well.

Another device that reports different numbers of association groups in different ways is the Fibaro Universal Binary Sensor (FGBS-001), and that works well now.

Thanks,
Steve.