openhab/org.openhab.binding.zwave #918: Z-Wave Serial-over-TCP Controller

gersilex, 2018-05-30:
* OH2 Version: 2.2.0
* Binding Version: 2.2.0

While moving the whole home IT to stateless, dockerized systems I noticed, that there is no way to connect to a TCP socket instead of a serial device. I found those functionalities in the MySensors and RFXCOM Binding and found that using them is very easy and reliable.

The hardware device that has the Z-Wave Serial Controller connected can forward that serial port to a listening TCP socket using [socat](https://linux.die.net/man/1/socat) or [ser2net](https://linux.die.net/man/8/ser2net) and connecting to that socket from another device yields the same exact data like talking to the serial port directliy, minus the serial settings (baud rate, stop bits, ...).

I am very aware of alternatives like using socat to create a pty. All of this is not possible when the Serial Controller is not running on the same host as the openHAB docker container. So this is a feature request by design.

While I really hope that this would be relatively easy to implement, I can't take a shot at it right now and ask for help while continuing to get my development environment up and running.

--------------

## Creating a test environment on Raspbian

### Prerequisites for the device host:
- Z-Wave Serial Controller
- root access

I am using a raspberry pi as a realistic and common example.

### Installation
- Log in to the Host
- Become root, if not already: `sudo -i`
- Install ser2net: `apt-get update && apt-get install ser2net`

### Configuration
- Open the config file of ser2net: `vi /etc/ser2net.conf`
- Replace the content of the file with this:
```
2002:raw:0:/dev/zwave:115200 8DATABITS NONE 1STOPBIT HANGUP_WHEN_DONE
```
where `/dev/zwave` is the device or a symlink to the device.
where `2002` is the TCP port you want to expose the device on the network.

- Save and quit
- Restart ser2net (after every change): `systemctl restart set2net`
- `netstat -lntp` should show a tcp LISTEN on socket 0.0.0.0:2002 by program `ser2net`

### Usage
- Connect to the port and talk to Z-Wave Controller over TCP

#### Linux netcat example
`nc openhab 2002` where `openhab` is the host name of the host device that has the serial controller connected.

--------------

Please, let me know if I can do more to help and thank you very much!

bodiroga, 2018-05-31:
Hi @gersilex!

I like that you have written this message, because that is what I am currently doing and its working great. To start with, I'm using this python program/script in the Raspberry Pi that acts as a gateway (I call them seriatl2tcp gateways): https://github.com/Tknika/serial2tcp-gateway

Both common Z-Wave controllers are supported (Z-Wave Gen2 and new Z-Wave+ usb devices) but it is really easy to add more just editing this file: https://github.com/Tknika/serial2tcp-gateway/blob/master/serial2tcp-gateway/device_definitions.py The TCP ports are anounced using multicast DNS (mDNS) and I have edited the binding to support this new bridge type and its discovery. I will try to push the code to my own repo tomorrow and will compile a .jar file so that you can test it and give me your opinion.

Anyway, my idea was to wait until @cdjackson merges the development branch into master and then have a conversation about how he would like to address it. But, yeah, first I should show my code and then discuss it, so tomorrow I will come back here and we can talk about it using my code as the starting point.

As always, thanks to Chris for all the time he spends for the openHAB community and best regards,

AItor

cdjackson, 2018-05-31:
I'm not super keen on this I must say. To me, it's not the right architecture to go and add such code to every binding that wants to have a serial port over IP. Personally, I think a better option (if possible) is to provide support through a virtual serial port - this means that any binding can use it without change, and there's no hacking of bindings required.

I'm not sure how easy this is, but I think I saw a similar conversation recently in ESH about using different serial drivers and I think this was the same sort of use case.

Maybe I can be convinced to add something to the binding, but really I think it would be good to look at a more universal option if possible. What do you think?

cdjackson, 2018-05-31:
For information - I think this is a similar use case and IMHO it provides a more flexible system as it would allow any binding to use IP services without messing with the binding.

https://github.com/eclipse/smarthome/pull/5560

gersilex, 2018-05-31:
> I think a better option (if possible) is to provide support through a virtual serial port.
This is a solution. But it's a solution with strict barriers. When considering the current development in computer and network architecture, running applications on machines - be it hardware or virtual machines - is getting less and less attractive. Everything get's packed up in containers and can be run anywhere. Usually one doesn't even know on which machine the software (in this case openHAB) is running. It could even run in the cloud and have all the operating system stuff abstract and inaccessable to the user.

What I am trying to say is, that things are moving away from fiddling in the operating systems and moving to just starting pre-packaged software on abstract operating-system-like providers. This is also true in relatively small environments at home where I operate 2 Docker hosts, with no common operating system at all and the openHAB Docker container that just run whereever there is RAM and CPU provided. This could even change throughout the day.

Thinking of larger scale networks like that of my company and many other modern companies. Usually there are hundreds of those hosts that just provide some CPU and RAM to run containers. Working on the underlaying operating system on all of these hosts (that often differ in OS type and versions) would be a huge configuration workload. This doesn't scale at all. But what does scale is the good old TCP/IP stack. All of those hosts have a lot of computing power and a lot of network bandwidth - something that is commonly needed by almost everything today.

If we can implement the Serial-over-TCP Gateway, TelnetSerial Gateway, or however we want to call it, we automatically benefit from all of those possiblities and also - I think we can agree that talking to a TCP socket is way more stable (as in fault-tolerant) that having to deal with baud rates, stop bits, parity and so on.

I hope this makes any sense as this is clearly my corporate datacenter (and homelab) view point and could also be way over the top. But I really think the cost-benefit ratio looks promising without compromising anything in the binding at all. We would have a differen TCP Controller thing next to the Serial Controller which might seem like duplicate code, but we also have a Serial Controller in every single binding, so this is also duplicated already.

> https://github.com/eclipse/smarthome/pull/5560
Thank you for pointing me to this pull request. I absolutely think this could be a solution for my request. I assume we would also remove the Serial Controller and hand it over to the `SerialPort`? In that case, we also could drop the TCP Controller support (if we would have built one when the `SerialPort` becomes usable). My only fear is that the pull request in the Eclipse Smarthome Project will take a long before we could even start trying out how well this would work.

To conclude, I absolutely agree that eclipse/smarthome#5560 should be the long-term solution, and I also think that we should try an implementation in the binding to see how well it works (maybe it never makes it to the master, maybe it does).

@bodiroga having said this, I would happily see and test your work and help you with it whereever I can, while still trying to keep on (or near) track with @cdjackson roadmap so that we can contribute back to his upstream, if we can make it fit. :)

Thanks to all of you. I am really interested in your thoughts about this!

cdjackson, 2018-06-01:
> What I am trying to say is, that things are moving away from fiddling in the operating systems and moving to just starting pre-packaged software on abstract operating-system-like providers. 

I don't disagree - I simply stated that I thought there was a better architecture that could be employed.  Bindings still need to interface with serial devices - this can't change. We already have a standard interface (SerialComm) that can be used here - why not use standard interfaces that everyone knows and can use, rather than bespoke additions to each binding?!?

> To conclude, I absolutely agree that eclipse/smarthome#5560 should be the long-term solution

Great - so I suggest to continue the discussion there and close this. Or, if you were suggesting that this should be the longterm solution, but we should ALSO add it to the binding, then I don't agree it's a wise move. I would suggest to work on the above solution which I think is trying to achieve the same goal.

gersilex, 2018-06-01:
> Great - so I suggest to continue the discussion there and close this.
Alright, I will try to to the proof-of-concept with @bodiroga and then transfer the findings to eclipse/smarthome#5560 whereever I can.

closed 2018-06-01 by gersilex

cdjackson, 2018-06-01:
I would suggest to comment on the ESH issue as well - at least so that people know this work is being looked at as I think it's likely to be of interest to other bindings.

bodiroga, 2018-06-01:
Here you have a link to my branch: https://github.com/bodiroga/org.openhab.binding.zwave/tree/development_tcp_controller

I had to make some small changes to the ZWaveThingHandler and ZWaveConfigProvider, mainly because I wanted to supported the reinitialization of the network without having to restart the binding.

I will try to follow the ESH issue, although in the first read I haven't been able to understand quite well what's the goal. I have to read it more carefully :wink: 

Thanks both for your work and comments!

gersilex, 2018-06-02:
@bodiroga Amazingly, everything worked right out of the box, including the discovery.

Can you please open the `Issues` area on your fork, so we can continue discussing there? Or whatever fits you.

bodiroga, 2018-06-03:
Hi @gersilex!

Perfect, I'm glad that the binding also works fine for you! I also have opened the "Issues" area in my fork (I didn't know that could be done, thanks), so feel free to open a new topic therel

Best regards,

Aitor