openhab/org.openhab.binding.zwave #938: [DEV] Can't disable Command Repoll Period

mhilbush, 2018-06-29:
It appears that the Command Repoll period cannot be set to **Disable**

Setting to `Disable` in HABmin sends this in the REST request payload. This looks correct given the definition of the config parameter.
```
{binding_cmdrepollperiod: 0}
```
However, when viewing in HABmin, it shows 100. It appears that the config coming back from the binding is 100, not 0.
```
		"configuration": {
			"config_84_3_0000FF00": 50,
			"binding_cmdrepollperiod": 100,
			"config_100_1_wo": 0,
```
And is 100 in the jsondb
```
      "configuration": {
        "properties": {
          "config_84_3_0000FF00": 50,
          "binding_cmdrepollperiod": 100,
          "config_100_1_wo": 0,
          "config_91_2": 25,
          "config_92_1": 5,
          "config_90_1": 0,
          "group_1": [
            "node_1"
          ],
```

It also would be nice if this could be set as a default, similar to the way you can set a default for the poll period.

cdjackson, 2018-06-30:
This is a disconnect in my thinking. I hadn’t actually intended this to be disable-able, except I added this as an option (probably due to cut and paste). Do you need to disable this or is it just that you found there was a disconnect?

cdjackson, 2018-06-30:
> I had tried to disable it for my lock because when I operated the lock the item would change ON / OFF / ON or OFF / ON / OFF because the lock hadn't yet changed status when it received the command repoll. I worked around it by lengthening the repoll period.
> 
This isn’t a workaround - this is exactly what the delay is for.

In some instances the poll after command is absolutely required - eg if you have he “restore to last value” set - then turning a dimmer on will go to some value, and without a poll, the UI doesn’t know what the value is. In other cases, it seems good practice to confirm that the command you sent actually happened. Maybe there are other cases where it’s not a good idea - hence my question about if you really needed to disable it or not.

The binding has always done this - it’s just that now the time is configurable rather than it being sent immediately after the command.

mhilbush, 2018-06-30:
Thanks for the explanation. I'm glad to hear I'm using the feature as intended. 😁

Given that, I'm fine with it. Want me to leave this open as a reminder to remove the disable option?

cdjackson, 2018-06-30:
I’ve removed it from the code, but let’s leave it open and I’ll close it once it’s merged.

Thanks.

cdjackson, 2018-07-01:
Will be closed by #943.

closed 2018-07-01 by cdjackson