openhab/org.openhab.binding.zwave #939: Color command processing is incompatible with EzMultiPli

hexamer, 2018-06-30:
This device only has discrete red, green, and blue LEDs (no white LEDs), but also uses the CW color state to turn on or off all LEDs to simulate white.  So if a color switch command is sent with a CW component (after the R/G/B components), as is the case with the current binding, there is no way to get any state other than off or white.

Per discussion in this thread, the proper fix is to refactor the color commands such that RGB and CW/WW are sent separately: https://community.openhab.org/t/problems-commanding-color-of-ezmultipli/46683

* OH2 Version: 2.3.0
* Binding Version: 2.3.0

* Device name: EzMultiPli 2.0
* Device version: 2.2
* Link to [database]:https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/268

cdjackson, 2018-06-30:
I will take a look at a PR for this tomorrow - it's partly coded already, but need to spend another hour or so to complete it.

cdjackson, 2018-07-15:
Closing as the above PR is merged - thanks for your testing...

PR #942 (2018-07-15): Refactor color converter
closes #939 
replaces #940
closed 2018-07-15 by cdjackson