openhab/org.openhab.binding.zwave #959: Incorrect conversion from HSB to RGB

fh-eureka, 2018-07-14:
**Expected Behavior**
Send command "0,100,100" to COLOR item. RGB values should be (255,0,0) meaning (100%, 0%, 0%)

**Current Behavior**
Send command "0,100,100" to COLOR item. RGB values will become (100,0,0) which is (39%, 0%, 0%)

**Steps to Reproduce**
1. Open REST API UI (misc-restdocs - 2.3.0)
2. Go to Item --> post /items/{itemname} (Sends a command to an item)
3. Fill parameters: itemname = {the color item}, body = 0,100,100
4. Click "Try it out"
5. Check the status of the RGB channels (on any OpenHAB UI)

**Detailed Description**
The current implementation for **receiveCommand** uses HSBType to convert from OpenHAB color item HSB (this is actually an HSV not to be confused with HSL) to Z-Wave RGB color components.
The Z-Wave command expects each component to be a value between 0 and 255
But HSBType actually returns a value between 0 and 100 for each of getRed(), getGreen(), getBlue()
`    public PercentType getRed() {
        return byteToPercentType(toColor().getRed());
}
`

**Possible Solution**
Quick solution:
At line 135
`rawMessages = commandClass.setColor(color.getRed().intValue(), color.getGreen().intValue(),
color.getBlue().intValue(), 0, 0);`
Multiply each RGB component by 2.55  
But double conversion from 255 to percentage and back means we'll lose some precision. 
Better solution:
Do not use HSBType for the conversion (maybe use java.awt.Color)

Also the IF statement at line 132 
`if (color.getSaturation().equals(PercentType.ZERO))`
Should be removed. There is no special case for saturation 0. the result should be RGB(255,255,255).
Setting RGB to 0 and assuming there is a WHITE channel and also assuming that the RGB LED light strip has White LEDs is not always valid. If the user is using normal RGB LED strips (not RGBW led strips, the white channel from the RGBW controller will not be connected to anything)
When processing an HSB command the RGB channels should be set. the W channel should either be set to 0 (which is the current implementation for all HSB values except *,0,*) or even better not set at all (left as-is).

Also regarding the comment at line 22
`        // Since we get an HSB, there is brightness information. However, we only deal with the color class here`
`        // so we need to scale the color and let the brightness be handled by the multi_level command class
`
We get an HSB and the color class takes RGB.
HSB can be converted to RGB and vise versa.
RGB has all the components including the brightness.
Nothing needs to be scaled, and nothing needs to be handled by the multi_level command.

ULTRA ideal solution: Do not use HSL.
A new channel type / item type should be created for RGBW values. (the value could be integer 0xwwrrggbb or just a string W,R,G,B) 

**Context (Environment)**
- OH2 Version: 2.3.0
- Binding Version: 2.3.0
- Device name: Fibaro RGBW Controller
- Device version: 25.25
- Link to [database]:https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/131

cdjackson, 2018-07-14:
Please can you take a look at the development branch as all changes are going in to that branch and this may be fixed already.

fh-eureka, 2018-07-15:
OK.
I looked at the development branch.
The first issue has been fixed. 
But it would be much better not to use HSBType which converts from 255 to percentage then we convert back from percentage to 255. (This could cause a problem in some cases where the value reported back from the device is not equal to the value we expect we have sent)

The other two issue are still there:
The special case for Saturation == 0 will still cause a problem for anyone using RGB (not RGBW) light. Sending HSB(0,0,100) which should be white (RGB(255,255,255)) but instead they get black (all off)

And the comment is confusing.
The only reason we need the "scaleColor" function is because HSBType returns percentage for RGB component which is incorrect behavior IMO.

fh-eureka, 2018-07-15:
Actually looking at the source code for HSBType:
`    public PercentType getRed() {
        return byteToPercentType(toColor().getRed());
}`

Since `toColor()` is public there is an easy fix for the first issue.
instead of using `HSBType.getRed()` we should use `HSBType.toColor().getRed()`
So instead of 
`rawMessages = commandClass.setColor(scaleColor(color.getRed()), scaleColor(color.getGreen()),
scaleColor(color.getBlue()), 0, 0);`
Should be:
`rawMessages = commandClass.setColor(color.toColor().getRed(), color.toColor().getGreen(),
color.toColor().getBlue(), 0, 0);`

Or even better:
The `color` variable should be `HSBType.toColor()`
So instead of
`HSBType color = (HSBType) command;`
should be
`java.awt.Color color = ((HSBType) command).toColor();`
And fix the rest of the code accordingly.

cdjackson, 2018-07-15:
If there is an issue with HSBType, then it should be fixed there. The binding should use the standard libraries, and not create new functions just because the framework is wrong.

fh-eureka, 2018-07-15:
Yes exactly.
No need to create new functions. (the current implementation has a new workaround function called `scaleColor()` )  
HSBType has a function called `getRed()` which returns a percentage. And also has a **public** function called `toColor()` which returns a color that we can use to `getRed()` 
If we use `toColor().getRed()` there is no need to use the workaround function `scaleColor()`

fh-eureka, 2018-07-15:
Yes, you're right I was looking at a different implementation of HSBType.
As you said the new implementation of HSBType has `getRGB()` which returns the correct scale.
It is converting from percentage to 0-255, but I think the percentage was not converted to integer so we should not lose any precision.

Other than that. If the saturation special case (saturation == 0) can be removed, we can close this issue.

cdjackson, 2018-07-15:
I have left the scaleColor method as is. using ```getRGB``` also requires further processing since it returns a single integer and then needs to be bitshifted so there is always two operations required no matter what.

I agree that this isn't so nice, but I suspect that changing HSBType may not be possible - maybe adding more methods _might_ be acceptable, but for now, I'll leave this as is unless someone strongly disagrees...

closed 2018-07-15 by cdjackson

fh-eureka, 2018-07-15:
Fair enough.
But can you please fix the `scaleColor` function so it does not convert percentage to integer.
Should get the float percentage value, do the calculation using float numbers then round the result.
We could use something similar to this function in HSBType:
`    private int convertPercentToByte(PercentType percent) {`
`        return percent.value.multiply(BigDecimal.valueOf(255))`
`                .divide(BigDecimal.valueOf(100), 2, BigDecimal.ROUND_HALF_UP).intValue();`
`}`

And what about the white color issue (saturation == 0) ?

cdjackson, 2018-07-15:
Why does using the float percentage matter? Does it really need to be that accurate as it’s being converted to an int anyway?

I already removed the saturation == 0 code.

cdjackson, 2018-07-15:
> It matters if someone is creating an app and using the REST API.
> So if the app needs to send a color to OpenHAB, the app has the RGB value, then convert that to HSV send it to OpenHAB, then Z-Wave binding will convert that back to RGB. If you are using integers instead of float the resulting RGB will not be the same as the source RGB in some cases.
> So the value we send and the actual value we get are not the same (close but not the same). This could cause an issue or just a strange behavior. And it can be easily avoided.
> (As you can see from the HSBType implementation, they had this in mind when they implemented convertPercentToByte function)
>
> Thank you very much Chris.

Sorry - I forgot to change to using floats before merging this. I have changed it though and it will flow through in the coming days.

Cheers
Chris