openhab/org.openhab.binding.zwave #993: heatit/thermofloor z-trm2 & z-relay have duplicate channel defs.

olemr, 2018-09-09:
OH2 Build #1344
180 | Active   |  80 | 2.4.0.201809081506     | ZWave Binding

Hi Chris. 
These two new devices have some duplicate channel definitions that is somewhat confusing.
Should/could it be cleaned up?

z-trm2 
The Furnace set-point should be removed.
Sensor (temp) 2 is temp 1 (AIR, internal sensor) on the device.
Sensor (temp) 3 is temp 2 (External sensor))
Sensor (temp) 4 is temp 3 (Floor sensor)
[image]

z-relay 
Sensor (temp) 2 is input 1 on the device
Sensor (temp) 3 is input 2 on the device
Alarm 4 is input 3 on the device
[image]

cdjackson, 2018-09-09:
Yes, please feel free to mark duplicates for deletion in the database (just rename the channels label to DELETE or something similar), and myself or @sihui can remove them. (I’m assuming here that you have database access?).

Firstly though, a few comments as you are marking the multi-channel encapsulated channels for deletion…

It’s not uncommon for ZWave devices to do this - normally, they will have functionality provided in an endpoint, and then some will also be duplicated in the root node (for backward compatibility reasons). The thing to be careful of is to get the right ones - with the new association scheme, new devices will normally send data that is linked to association groups using the multi instance encapsulation - so they will appear in the numbered endpoints, and not the root endpoint (ie endpoint 0 in the database).  Just be sure that you are getting the correct ones and any data sent to associations is really working before marking channels for deletion ;)

Otherwise - I definitely am happy to remove duplicates to reduce the confusion. Ideally, we’d do this on all devices, but it takes people with the devices to update the database :)

Any questions, just ask…

Thanks.

olemr, 2018-09-09:
OK, but I must honestly say that I don't fully understand all aspect about the endpoints, and I have never done any associations, so it seems to be a hard making sure I don't break anything.
Probably best to leave it as-is.
Thanks.

cdjackson, 2018-09-09:
It’s fine that you haven’t done any associations - the binding will set the lifeline for you. The thing you need to be sure of is that the channels that you want to keep are being updated when the device changes - ie not just when you change something in the OH UI, but also when the temperature updates, or whatever else updates, and the device sends a status without you doing anything.

If you have the wrong set of channels, then you will only get updates when the binding polls (normally every 30 minutes) and not when the device sends out an update on its own.

I think if you can make sure of this, then getting rid of the other channels is a good idea :)

cdjackson, 2018-09-29:
@olemr did we close this out already?

closed 2018-11-15 by cdjackson