openhab/org.openhab.binding.zwave #1002: Burglar Alarm Not Set to ON When Receiving Event = 3

fh-eureka, 2018-09-16:
**Expected Behavior**
Burglar channel state (and linked item) should be updated to ON when shaking the Sensor.

**Current Behavior**
Burglar channel state (and linked item) remains OFF when shaking the Sensor.

**Steps to Reproduce**
1. Include Fibaro Flood Sensor (version 3.2)
2. Link Alarm (burglar) to an item
3. Shake the sensor
4. Sensor gives beep-beep sound 
5. Alarm (burglar) items remains OFF

**Detailed Description**
This is the relevant section of the log file:
```
14:03:29.434 [DEBUG] [ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0F 00 04 00 15 09 71 05 00 03 00 FF 07 03 00 64
14:03:29.440 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=21, callback=0, payload=00 15 09 71 05 00 03 00 FF 07 03 00
14:03:29.452 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - processReceiveMessage past lock Message: class=ApplicationCommandHandler[4], type=Request[0], dest=21, callback=0, payload=00 15 09 71 05 00 03 00 FF 07 03 00
14:03:29.461 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=21, callback=0, payload=00 15 09 71 05 00 03 00 FF 07 03 00
14:03:29.469 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - lastTransaction null
14:03:29.473 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 21: Application Command Request (ALIVE:DONE)
14:03:29.478 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 21: resetResendCount initComplete=true isDead=false
14:03:29.484 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 21: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
14:03:29.490 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 21: SECURITY NOT required on COMMAND_CLASS_ALARM
14:03:29.497 [DEBUG] [otocol.commandclass.ZWaveCommandClass] - NODE 21: Received COMMAND_CLASS_ALARM V5 NOTIFICATION_REPORT
14:03:29.505 [DEBUG] [l.commandclass.ZWaveAlarmCommandClass] - NODE 21: NOTIFICATION report - 0 = 3, event=3, status=255, plen=0
14:03:29.511 [DEBUG] [l.commandclass.ZWaveAlarmCommandClass] - NODE 21: Alarm Type = BURGLAR (0)
14:03:29.517 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 21: Got an event from Z-Wave network: ZWaveAlarmValueEvent
14:03:29.523 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 21: Got a value event from Z-Wave network, endpoint = 0, command class = COMMAND_CLASS_ALARM, value = 255
14:03:29.532 [DEBUG] [nternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
14:03:29.538 [DEBUG] [nternal.converter.ZWaveAlarmConverter] - Alarm converter NOTIFICATION event is 3, type OnOffType
14:03:29.545 [DEBUG] [nternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
14:03:29.550 [DEBUG] [nternal.converter.ZWaveAlarmConverter] - Alarm converter NOTIFICATION event is 3, type OnOffType
14:03:29.557 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 21: Commands processed 1.
14:03:29.561 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 21: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@356fe6a4.
14:03:29.568 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
14:03:29.572 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
14:03:29.578 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
14:03:29.582 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
14:03:53.829 [DEBUG] [ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0F 00 04 00 15 09 71 05 00 00 00 FF 07 00 00 64
14:03:53.845 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=21, callback=0, payload=00 15 09 71 05 00 00 00 FF 07 00 00
14:03:53.856 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - processReceiveMessage past lock Message: class=ApplicationCommandHandler[4], type=Request[0], dest=21, callback=0, payload=00 15 09 71 05 00 00 00 FF 07 00 00
14:03:53.867 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=21, callback=0, payload=00 15 09 71 05 00 00 00 FF 07 00 00
14:03:53.875 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - lastTransaction null
14:03:53.879 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 21: Application Command Request (ALIVE:DONE)
14:03:53.885 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 21: resetResendCount initComplete=true isDead=false
14:03:53.890 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 21: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
14:03:53.896 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 21: SECURITY NOT required on COMMAND_CLASS_ALARM
14:03:53.901 [DEBUG] [otocol.commandclass.ZWaveCommandClass] - NODE 21: Received COMMAND_CLASS_ALARM V5 NOTIFICATION_REPORT
14:03:53.907 [DEBUG] [l.commandclass.ZWaveAlarmCommandClass] - NODE 21: NOTIFICATION report - 0 = 0, event=0, status=255, plen=0
14:03:53.913 [DEBUG] [l.commandclass.ZWaveAlarmCommandClass] - NODE 21: Alarm Type = BURGLAR (0)
14:03:53.918 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 21: Got an event from Z-Wave network: ZWaveAlarmValueEvent
14:03:53.925 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 21: Got a value event from Z-Wave network, endpoint = 0, command class = COMMAND_CLASS_ALARM, value = 255
14:03:53.933 [DEBUG] [nternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
14:03:53.939 [DEBUG] [nternal.converter.ZWaveAlarmConverter] - Alarm converter NOTIFICATION event is 0, type OnOffType
14:03:53.946 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 21: Updating channel state zwave:device:32303136-3131-3033-3030-303031353434:node21:alarm_burglar to OFF [OnOffType]
14:03:53.955 [DEBUG] [nternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
14:03:53.961 [DEBUG] [nternal.converter.ZWaveAlarmConverter] - Alarm converter NOTIFICATION event is 0, type OnOffType
14:03:53.967 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 21: Commands processed 1.
14:03:53.972 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 21: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@7f2ed49f.
14:03:53.978 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
14:03:53.983 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
14:03:53.988 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
14:03:53.992 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
```
When the sensor is shaken OpenHAB recives:
NODE 21: NOTIFICATION report - 0 = 3, **event=3**, status=255, plen=0
(This does not update the channel state)
Then After ~23 seconds it receives:
NODE 21: NOTIFICATION report - 0 = 0, **event=0**, status=255, plen=0 
(This updates the channel state to OFF)

**Context (Environment)**
    OH2 Version: 2.3.0
    Binding Version:  2.4.0.201807242215
    Device name: Fibaro Flood Sensor
    Device version: 3.2
    Link to [database]: https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/392

cdjackson, 2018-09-16:
Firstly, thank you for writing a good issue report and providing all the information :)

The issue is that this channel is configured as a burglar alarm, and it’s really a tamper alarm. I will update the database and it should fix the problem. Note that you will need to delete the thing and add it back again once the database is updated in the binding. This is to allow your system to pick up the latest definitions.

cdjackson, 2018-09-16:
While I’m looking at this - can you tell me what the binary sensor does? I assume it is another flood indicator and is therefore not used? If so, I’ll delete it from the database to avoid confusion.

cdjackson, 2018-09-16:
Oh - and same question for the Alarm (general) channel - is that used for anything or is it another flood indicator that can be removed?

cdjackson, 2018-09-16:
> Thank you Chris for your quick response.
> The binary sensor and the Alarm (general) are not related to the flood sensor.
> I was not able to activate them.
> Here is the log when the flood sensor is activated then deactivated:
> 15:06:23.161 [DEBUG] [ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0F 00 04 00 15 09 71 05 05 02 00 FF 05 02 00 63
> 15:06:23.172 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=21, callback=0, payload=00 15 09 71 05 05 02 00 FF 05 02 00
> 15:06:23.183 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - processReceiveMessage past lock Message: class=ApplicationCommandHandler[4], type=Request[0], dest=21, callback=0, payload=00 15 09 71 05 05 02 00 FF 05 02 00
> 15:06:23.192 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=21, callback=0, payload=00 15 09 71 05 05 02 00 FF 05 02 00
> 15:06:23.200 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - lastTransaction null
> 15:06:23.204 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 21: Application Command Request (ALIVE:DONE)
> 15:06:23.209 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 21: resetResendCount initComplete=true isDead=false
> 15:06:23.214 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 21: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
> 15:06:23.219 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 21: SECURITY NOT required on COMMAND_CLASS_ALARM
> 15:06:23.225 [DEBUG] [otocol.commandclass.ZWaveCommandClass] - NODE 21: Received COMMAND_CLASS_ALARM V5 NOTIFICATION_REPORT
> [... cut ...]
> 15:06:26.170 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
> 15:06:26.174 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
> The device itself support external sensor / alarm
> (see Connection with a wired alarm system)
> So I suspect these channels are related to that.

Thanks. Reading the manual, I’m unclear if these are inputs, or outputs -: may be connected to any external system (potential free output terminal)

I read this as though the flood sensor can output its status TO a wired alarm system, but I might be reading it wrong.

I think for now, I will remove these channels and if someone needs them, we can understand this better.

Thanks.

fh-eureka, 2018-09-16:
BTW.
Same issue applies to Fibaro Smoke Detector (version 3.2)
https://manuals.fibaro.com/smoke-sensor/
NODE 22: NOTIFICATION report - 0 = 0, **event=3**, status=255, plen=0
(this one is activated by opening the case)
Link to [database]: https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/878

cdjackson, 2018-09-16:
Can you open another issue for this please - and provide a short log. The database looks ok for this - it is already set to alarm_tamper.

Can you confirm that the flood sensor is working now? (note you will need to delete the thing and add it back to get the latest definition).

cdjackson, 2018-09-16:
> But I'm using the development branch (Locks, RGB fixes, etc.)

There is no development branch any more - it was merged into the master, so you can use the standard snapshot builds.

fh-eureka, 2018-09-16:
Thank you Chris.
Using the latest JAR file from here:
https://openhab.ci.cloudbees.com/job/openHAB2-Bundles/lastBuild/org.openhab.binding%24org.openhab.binding.zwave/
Tamper alarm is working fine for both Fibaro Flood Sensor and Fibaro Smoke Detector.

BTW
Looking at the xml files inside the latest JAR files I've noticed there is a difference in tamper alarm channel definition between FGFS and  FGSD
```
      <channel id="alarm_tamper" typeId="alarm_tamper">
        <label>Tamper Alarm</label>
        <properties>
          <property name="binding:*:OnOffType">COMMAND_CLASS_ALARM;type=BURGLAR</property>
        </properties>
      </channel>
```
vs
```
      <channel id="alarm_tamper" typeId="alarm_tamper">
        <label>Tamper Alarm</label>
        <properties>
          <property name="binding:*:OnOffType">COMMAND_CLASS_ALARM;type=BURGLAR, event=3</property>
        </properties>
      </channel>
```
Is the last part (event=3) optional? or is it missing from FGFS definition?

closed 2018-09-17 by fh-eureka