openhab/org.openhab.binding.zwave #1015: Eurotronic Spirit Z-wave - Configuration External Air Temperature

szewczykus, 2018-09-23:
* OH2 Version: 2.4.0-SNAPSHOT, build: 1369
* Binding Version: 2.4.0.201809081506
* Device name:  Eurtronic Spirit Z-wave

I would like to set Eurortronic Spirit Z-wave into external air temperature sensor mode, by setting configuration parameter 8, Measured Temperature Offset to 128. 

When i am doing this via habmin it seems that incorrect value is being sent to device = "-128".

**2018-09-23 06:33:49.303 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Configuration update set config_8_1 to 128**
**2018-09-23 06:33:50.787 [DEBUG] [class.ZWaveConfigurationCommandClass] - NODE 18: Node configuration report, parameter = 8, value = -128, size = 1**

```
2018-09-23 06:33:11.203 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Configuration update received
2018-09-23 06:33:49.296 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Configuration update received
2018-09-23 06:33:49.303 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Configuration update set config_8_1 to 128
2018-09-23 06:33:49.306 [DEBUG] [class.ZWaveConfigurationCommandClass] - NODE 18: Creating new message for application command CONFIGURATION_SET
2018-09-23 06:33:49.309 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 18: SECURITY NOT required on COMMAND_CLASS_CONFIGURATION
2018-09-23 06:33:49.311 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 18: Command Class COMMAND_CLASS_CONFIGURATION is NOT required to be secured
2018-09-23 06:33:49.314 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: Bump transaction 324 priority from Config to Immediate
2018-09-23 06:33:49.316 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: Adding to device queue
2018-09-23 06:33:49.319 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
2018-09-23 06:33:49.322 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0C 00 13 12 05 70 04 08 01 80 25 3A 15 
2018-09-23 06:33:49.324 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 18: Sending REQUEST Message = 01 0C 00 13 12 05 70 04 08 01 80 25 3A 15 
2018-09-23 06:33:49.326 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - Message SENT
2018-09-23 06:33:49.327 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage started: TID 324: [WAIT_RESPONSE] priority=Immediate, requiresResponse=true, callback: 58
2018-09-23 06:33:49.328 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 06
2018-09-23 06:33:49.329 [DEBUG] [class.ZWaveConfigurationCommandClass] - NODE 18: Creating new message for application command CONFIGURATIONCMD_GET
2018-09-23 06:33:49.330 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=null[0], type=ACK[2], dest=255, callback=0, payload=
2018-09-23 06:33:49.331 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage past lock Message: class=null[0], type=ACK[2], dest=255, callback=0, payload=
2018-09-23 06:33:49.331 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 18: SECURITY NOT required on COMMAND_CLASS_CONFIGURATION
2018-09-23 06:33:49.332 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 18: Command Class COMMAND_CLASS_CONFIGURATION is NOT required to be secured
2018-09-23 06:33:49.332 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=null[0], type=ACK[2], dest=255, callback=0, payload=
2018-09-23 06:33:49.333 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction TID 324: [WAIT_RESPONSE] priority=Immediate, requiresResponse=true, callback: 58
2018-09-23 06:33:49.333 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: Bump transaction 325 priority from Config to Immediate
2018-09-23 06:33:49.335 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg: ACK
2018-09-23 06:33:49.335 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 04 01 13 01 E8 
2018-09-23 06:33:49.335 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: Adding to device queue
2018-09-23 06:33:49.336 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2018-09-23 06:33:49.337 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start. Holdoff false.
2018-09-23 06:33:49.337 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=SendData[19], type=Response[1], dest=255, callback=0, payload=01 
2018-09-23 06:33:49.338 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage past lock Message: class=SendData[19], type=Response[1], dest=255, callback=0, payload=01 
2018-09-23 06:33:49.338 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start. Holdoff false.
2018-09-23 06:33:49.340 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=SendData[19], type=Response[1], dest=255, callback=0, payload=01 
2018-09-23 06:33:49.341 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction TID 324: [WAIT_RESPONSE] priority=Immediate, requiresResponse=true, callback: 58
2018-09-23 06:33:49.342 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Checking outstanding transactions: 1
2018-09-23 06:33:49.343 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Last transaction: TID 324: [WAIT_RESPONSE] priority=Immediate, requiresResponse=true, callback: 58
2018-09-23 06:33:49.345 [DEBUG] [ve.internal.protocol.ZWaveController] - Incoming Message: Message: class=SendData[19], type=Response[1], dest=255, callback=0, payload=01 
2018-09-23 06:33:49.346 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 18: sentData successfully placed on stack.
2018-09-23 06:33:49.348 [DEBUG] [nal.protocol.ZWaveTransactionManager] - TID 324: Advanced to WAIT_REQUEST
2018-09-23 06:33:49.349 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: TID 324: Transaction not completed
2018-09-23 06:33:49.351 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2018-09-23 06:33:49.352 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start. Holdoff false.
2018-09-23 06:33:50.649 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 07 00 13 3A 00 00 84 55 
2018-09-23 06:33:50.653 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=SendData[19], type=Request[0], dest=0, callback=58, payload=3A 00 00 84 
2018-09-23 06:33:50.656 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage past lock Message: class=SendData[19], type=Request[0], dest=0, callback=58, payload=3A 00 00 84 
2018-09-23 06:33:50.659 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=SendData[19], type=Request[0], dest=0, callback=58, payload=3A 00 00 84 
2018-09-23 06:33:50.661 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction TID 324: [WAIT_REQUEST] priority=Immediate, requiresResponse=true, callback: 58
2018-09-23 06:33:50.663 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Checking outstanding transactions: 1
2018-09-23 06:33:50.666 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Last transaction: TID 324: [WAIT_REQUEST] priority=Immediate, requiresResponse=true, callback: 58
2018-09-23 06:33:50.667 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Checking TID 324: (Callback 58)
2018-09-23 06:33:50.670 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Callback match!
2018-09-23 06:33:50.671 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Correlated to TID 324: callback 58
2018-09-23 06:33:50.674 [DEBUG] [ve.internal.protocol.ZWaveController] - Incoming Message: Message: class=SendData[19], type=Request[0], dest=0, callback=58, payload=3A 00 00 84 
2018-09-23 06:33:50.677 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 18: SendData Request. CallBack ID = 58, Status = Transmission complete and ACK received(0)
2018-09-23 06:33:50.680 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 18: resetResendCount initComplete=true isDead=false
2018-09-23 06:33:50.681 [DEBUG] [e.internal.protocol.ZWaveTransaction] - TID 324: Transaction COMPLETED
2018-09-23 06:33:50.684 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: Response processed after 1357ms
2018-09-23 06:33:50.686 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: TID 324: Transaction completed
[... cut ...]
2018-09-23 06:33:50.801 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Got an event from Z-Wave network: ZWaveTransactionCompletedEvent
2018-09-23 06:33:50.802 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 1
2018-09-23 06:33:50.803 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2018-09-23 06:33:50.804 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2018-09-23 06:33:50.804 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
```

cdjackson, 2018-09-23:
This is a problem area with ZWave as ZWave defines values to be unsigned, but many suppliers don't respect this and it's a little messy.

However, that aside, the log looks fine - what is the actual problem? The value of 0x80 is being set, and seems to be returned in the REPORT.

szewczykus, 2018-09-24:
Fine.
Generally speaking, i have got a problem with channel sensor_report, so i am finding what is not correct...

When i am trying to set a value to item linked with channel sensor_report nothing happens, and after some time value has been restored to current measured by valve temperature.

```
2018-09-24 18:57:59.810 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Command received zwave:device:512:node18:sensor_report --> 15
2018-09-24 18:57:59.824 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Polling intialised at 86400 seconds - start in 15000 milliseconds.
2018-09-24 18:58:14.814 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Polling...
2018-09-24 18:58:14.817 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Polling zwave:device:512:node18:sensor_temperature
2018-09-24 18:58:14.818 [DEBUG] [erter.ZWaveMultiLevelSensorConverter] - NODE 18: Generating poll message for COMMAND_CLASS_SENSOR_MULTILEVEL, endpoint 0
2018-09-24 18:58:14.820 [DEBUG] [ss.ZWaveMultiLevelSensorCommandClass] - NODE 18: Creating new message for command SENSOR_MULTILEVEL_GET
2018-09-24 18:58:14.821 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 18: SECURITY NOT required on COMMAND_CLASS_SENSOR_MULTILEVEL
2018-09-24 18:58:14.823 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 18: Command Class COMMAND_CLASS_SENSOR_MULTILEVEL is NOT required to be secured
2018-09-24 18:58:14.824 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Polling zwave:device:512:node18:thermostat_setpoint_heating
2018-09-24 18:58:14.826 [DEBUG] [ter.ZWaveThermostatSetpointConverter] - NODE 18: Generating poll message for COMMAND_CLASS_THERMOSTAT_SETPOINT, endpoint 0
2018-09-24 18:58:14.828 [DEBUG] [.ZWaveThermostatSetpointCommandClass] - NODE 18: Creating new message for application command THERMOSTAT_SETPOINT_GET (Heating)
2018-09-24 18:58:14.831 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 18: SECURITY NOT required on COMMAND_CLASS_THERMOSTAT_SETPOINT
2018-09-24 18:58:14.832 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 18: Command Class COMMAND_CLASS_THERMOSTAT_SETPOINT is NOT required to be secured
2018-09-24 18:58:14.835 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: Bump transaction 283 priority from Get to Immediate
2018-09-24 18:58:14.837 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: Adding to device queue
2018-09-24 18:58:14.839 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
2018-09-24 18:58:14.843 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0B 00 13 12 04 31 04 01 00 25 FB 1B
2018-09-24 18:58:14.846 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 18: Sending REQUEST Message = 01 0B 00 13 12 04 31 04 01 00 25 FB 1B
2018-09-24 18:58:14.849 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - Message SENT
```
It seems that device is pooled for current values and that is why my external temperature was overwritten...

szewczykus, 2018-10-02:
@kaikreuzer can you support?

cdjackson, 2018-10-02:
What are you wanting help on? I'm not sure I understand your latest statement - I assume that the initial issue is resolved (or at least understood).

> It seems that device is pooled for current values and that is why my external temperature was overwritten...

What does "pooled" mean?

szewczykus, 2018-10-02:
As you could see in log after channel sensor_report received command it triggers "polling"
```
2018-09-24 18:57:59.810 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Command received zwave:device:512:node18:sensor_report --> 15
2018-09-24 18:57:59.824 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Polling intialised at 86400 seconds - start in 15000 milliseconds.
2018-09-24 18:58:14.814 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Polling...
```
As far as i understand polling is reading data from device - in the wake value set from channel sensor_report is being ignored by overwriting it value from device.

cdjackson, 2018-10-02:
> in the wake value set from channel sensor_report is being ignored due to being overwritten by value read from device.

Sorry, I don't understand what you mean. Polling should not impact on the update of other channels - other than of course it will update any channels to read the latest value from the device.