openhab/org.openhab.binding.zwave #1024: Offline / Failed Devices Are Not Marked Offline / Failed

fh-eureka, 2018-10-07:
**Expected Behavior**
Offline Z-Wave devices (not powered) should be marked as OFFLINE.
Failed Z-Wave devices (factory reset devices, failure during inclusion, etc.) should be marked as FAILED.

**Current Behavior**
Some offline devices are showing as ONLINE. 
Some failed devices are showing as INITIALIZING / ONLINE.

**Steps to Reproduce**
Remove the power/battery of a Z-Wave Device
Device remains online
Restart OpenHAB
Device remains online (or stuck on initializing)

**Detailed Description**
Some (most) offline Z-Wave devices are not getting marked as offline (or failed) even after a few days of disconnecting their power.
Invalid devices, (for example a device was not included correctly the inclusion process did complete, or a device that was factory reset) are not getting marked as failed or offline.
As a result of this we cannot remove these failed devices from the controller (in HABmin when we try to remove them from controller we get a message that says the node was not found)
I have attached a log file that demonstrates these issue:
In the log file (this is a 2 hour log file) there are these nodes:
Node 9 (Fibaro Roller Shutter) and Node 10 (Fibaro Double Switch) are disconnected from power, but they are still showing as online in HABmin.
Nodes 28, 29 and 31 are "Unknown Devices" these are failed attempts to add a Fibaro Motion Sensor, are showing online or initializing, and cannot be removed from controller using HABmin.
Node 30 (Fibaro Motion Sensor) is factory reset then battery removed, but it's still showing as Node Initializing REQUEST_NIF in HABmin and cannot be removed from controller.
OpenHAB keeps trying to poll these devices but they are never marked offline/failed even after a few days.

**Context (Environment)**
- OH2 Version: 2.3.0
- Binding Version: 2.4.0.201809261735

Log File:
[openhab.zip](https://github.com/openhab/org.openhab.binding.zwave/files/2453872/openhab.zip)

cdjackson, 2018-10-07:
I just had a quick look at the binding, and there is an issue with processing these events. However you should also understand that removing the battery will not cause a battery device to be marked offline. Battery devices only communicate with the binding very occasionally, so it will take a while for them to be detected as offline.

Regarding the removal of failed devices, you need to use the “Move device to failed list”, and then “Delete device” (I forget the exact names) options in the thing menu.

fh-eureka, 2018-10-08:
Thank you very much Chris.
Yes, I understand that battery devices could take a while to be marked offline.
What is the exact condition that will cause a device (battery or main powered) to be considered offline? Is it related to polling interval? or wake up interval? for example if we failed to poll 3 times then it's considered offline or something like that?

Regarding the removal. 
I did try "move device to failed list" (aka "Set device as FAILED") but this action also failed. 
Interestingly the error message that you get in HABmin says "**Remove** node 30 failed - node was not found" which is the exact same message you get when you use "Remove device from controller"
Maybe it's just a UI issue, or maybe there is something wrong with "Set device as FAILED"

cdjackson, 2018-10-08:
> What is the exact condition that will cause a device (battery or main powered) to be considered offline?

For non-battery devices, it is the failure of a transaction, which is actually 3 attempts to send a command to the device.

For battery devices, the above is also true, but in addition, if no data is received from the device within twice the wakeup period it is also considered DEAD (ie offline).

cdjackson, 2018-10-11:
@fh-eureka with the latest snapshot is the marking devices offline any better? I've not had the chance to fully test this so would welcome your feedback.

fh-eureka, 2018-10-11:
I have installed the latest snapshot (version: 2.4.0.201810100057)
(I've dropped the the new jar file in the addons folder replacing the old file, and OH automatically restarted the binding)
Now after 3 hours, 3 devices are marked offline (2 main powered and 1 battery powered)
Other devices (that should be offline) are still marked online or REQUEST_NIF

I have attached the log file (starting from the time of zwave binding restart and ending after 3 hours)

Examples:
**Node 9** (Fibaro Roller Shutter) (this should be offline)
This node was marked offline at 13:43:33 (2 minutes after restart)
Then it was set **online!** at 14:04:19 (20 minutes after restart)
And it remained online.

**Node 10** (Fibaro Double Switch) (this should be offline)
This node was marked offline at 13:49:48 (8 minutes after restart)
And it remained offline.
(Every 30 minutes I see a "NODE 10: sendTransaction" in the log file, I expect this is retrying to communicate with the node to check if it became online?) 
I think this is the correct expected behavior, but 8 minutes is a long time to wait for the device status. 

**Node 14** (Fibaro RGBW) (this should be offline)
This node was marked offline at 113:47:49 (6 minutes after restart)
Then it was set **online!** at 14:41:38 (60 minutes after restart)
And it remained online.
Then after the device was marked online:
I see in the log file that every 30 minutes there is a "NODE 14: sendTransaction" 
And every 30 minutes there is a polling attempt
I think both are failing but the device remains online!

**Node 8** (MCOHOME MH-P311) (this device should be online)
This node was marked online at 13:45:28 (4 minutes after restart)
And it remained online.
Then after the device was marked online:
I see in the log file that every 30 minutes there is a polling request, I think these requests are succeeding.
And unlike Node 14 ("fake" online) there is no "sendTransaction" every 30 minutes!

Log File:
[openhab.zip](https://github.com/openhab/org.openhab.binding.zwave/files/2469207/openhab.zip)

fh-eureka, 2018-10-16:
@cdjackson Did you get a chance to look into the above log?

cdjackson, 2018-10-16:
Hopefully I've found the cause for this - at least it will solve some of the issues once merged.

cdjackson, 2018-10-16:
Oh, and I should have said - thanks for taking the time to provide a really useful bug report and analyses.

closed 2018-10-17 by cdjackson

fh-eureka, 2018-10-29:
Sorry for the late reply.
@cdjackson Thank you for your quick responses and great efforts.

Today I've got a chance to download and test the latest snapshot build (**2.4.0.201810282146**)
Unfortunately it looks like the issue still persist.
 
I have attached the log file (starting from the time of zwave binding restart and ending after 2 hours)

Looking at one example node:
**Node 9** (Fibaro Roller Shutter) (this should be offline)
This node was marked offline at 11:07:18.711 (3 minutes after restart)
`NODE 9: Setting OFFLINE`
Then immediately at 11:07:18.871 it was changed to REQUEST_NIF 
`NODE 9: Node advancer - advancing to REQUEST_NIF`
Is this the correct behavior? After it was set to OFFLINE it was changed to REQUEST_NIF?
Then at 11:08:18.990:
`NODE 9: No data from device, but it was ACK'd. Possibly not supported? (Try 0)`
Can you get ACK'd if the device is completely offline (power disconnected)?
Then at 11:41:31.639 (37 minutes after restart):
`NODE 9: Node advancer: Restored from file - skipping static initialisation`
This looks strange to me. That after 37 minutes it was decided to skip static initialisation and load from file!
Then the last Node advancer entry at 11:43:44.915:
`NODE 9: Node advancer - advancing to DONE`
I guess this did set the node to ONLINE! (HABmin shows that the node is online)
Then after that, there is a polling request every 30 minutes
And also a sendTransaction every ~15 minutes
All of them are failing but the device remains online.

Log file:
[openhab.zip](https://github.com/openhab/org.openhab.binding.zwave/files/2526218/openhab.zip)

PR #1035 (2018-10-17): Don't set the device online once static is complete
(hopefully!) fixes #1024