openhab/org.openhab.binding.zwave #1031: TKB home TZ65D

aviamax, 2018-10-12:
Hi, afer upgrading openhab all my switches (TKB home TZ65D) appear in the INBOX with wrong type - TZ35S.

I completed all the steps after the update specified in this document:
https://community.openhab.org/t/zwave-binding-updates/51080

Вefore updating the device appeared with the correct type

This issue list is to report bugs or enhancements for the openHAB Z-Wave binding. If you're not sure about something, please consider discussing it on the [forum](http://community.openhab.org) before raising an issue - more people follow the forum than monitor the issues list, so if you are having trouble with a device, or want to ask a question, you will have a much larger audience there.

In order to ensure we have the best information to deal with the issue, please fill in the following information (delete as necessary depending on the issue).

* OH2 Version: 2.4.0~20181011203025-1
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console):

If the problem is associated with a device, please provide the following -:

* Device name: TKB home TZ65D
* Device version:
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/155
* If the device is not in the database, provide a link to the manual.
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.

Provide a full debug level log of the problem. Preferably, you should try use the [log viewer](http://www.cd-jackson.com/index.php/openhab/zwave-log-viewer) to see if you can identify the problem.

If a device will not initialise, ensure the log is provided from the time OH starts.

Thank you.

cdjackson, 2018-10-12:
Please discuss on the forum. Often devices of similar types are lumped together as the manufacturer uses the same device types. If you raise this on the forum there will be more people to help you to resolve the issue . If it is just the name, then probably there is no need to change this.

Anyway, please discuss on the forum - thanks.

closed 2018-10-12 by cdjackson

aviamax, 2018-10-18:
I have three kinds of TKB Home devices - TZ65D,TZ65S,TZ66D. Before updating openhab their type was correctly determined by openhab. But now all of this devices was appeared as TZ35S in inbox.