openhab/org.openhab.binding.zwave #1041: Send external temperature to Spirit ZWave as Report instead of Get

maxberger, 2018-10-24:
* OH2 Version: SNAPSHOT 2.4.0.201810082025
* Binding Version: SNAPSHOT 2.4.0.201810082025
If the problem is associated with a device, please provide the following -:
* Device name: Spirit Z-Wave
* Device version: 0.15
* Link to [database]: https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/710

---

This is a thermostat which supports external temperature sensors. My external temperature sensor can only send information to the controller, so I need to forward that information to the thermostat.

Issue: At this time, providing external temperature does not work, the thermostat just keeps heating

What is the issue: The thermostat supports an "incoming" channel for temperature, which is mapped in openhab (was added last december). However, when setting the temperature, openhab send a "get" command. Instead, it should send a REPORT command.

---

I have started looking at the code, but I am unsure where to start. At the moment, I don't even see any code to create a Multilevel sensort report command.

**I am willing to start creating and testing a patch, but I would need some guidance as where to start with.**

---

References:
Snipped from the doc showing this needs REPORT: https://community.openhab.org/t/zwave-eurotronic-spirit-with-external-temperatur-dont-work/35330/2

Another person finding the same issue: https://community.openhab.org/t/external-temperature-report-send-sensor-multilevel-report-spirit-radiator-thermostat/36065

Full Manual: http://www.produktinfo.conrad.com/datenblaetter/1500000-1599999/001559295-an-01-de-EUROTRONIC_SPIRIT_Z_WAVE_PLUS.pdf

First implementation: https://github.com/openhab/org.openhab.binding.zwave/issues/720 [does not fully work yet]

cdjackson, 2018-10-24:
This feature should already be in the binding - it was added a long time ago, although I don't recall if anyone ever tested it. A new channel was created to support this, but please feel free to debug.

See #720

maxberger, 2018-10-24:
#720 added the channel to the binding, but it triggers a "GET" instead of a REPORT. We should send a report to this device.

Logs that show "GET"
20:09:19.560 [INFO ] [smarthome.event.ItemCommandEvent     ] - Item 'zwave_device_710d4836_node12_sensor_report' received command 22
20:09:19.566 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 12: Command received zwave:device:710d4836:node12:sensor_report --> 22 [DecimalType]
20:09:19.574 [INFO ] [arthome.event.ItemStatePredictedEvent] - zwave_device_710d4836_node12_sensor_report predicted to become 22
20:09:19.585 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 12: Polling intialised at 86400 seconds - start in 1500 milliseconds.
20:09:19.601 [INFO ] [smarthome.event.RuleStatusInfoEvent  ] - 47fc2df1-3eda-430b-9ddd-9603ad0c9754 updated: RUNNING
20:09:19.608 [INFO ] [smarthome.event.ItemStateChangedEvent] - zwave_device_710d4836_node12_sensor_report changed from 21.4 to 22
20:09:19.623 [INFO ] [smarthome.event.RuleStatusInfoEvent  ] - 47fc2df1-3eda-430b-9ddd-9603ad0c9754 updated: IDLE
20:09:21.085 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 12: Polling...
20:09:21.092 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 12: Polling zwave:device:710d4836:node12:sensor_temperature
20:09:21.102 [DEBUG] [verter.ZWaveMultiLevelSensorConverter] - NODE 12: Generating poll message for COMMAND_CLASS_SENSOR_MULTILEVEL, endpoint 0
20:09:21.114 [DEBUG] [ass.ZWaveMultiLevelSensorCommandClass] - NODE 12: Creating new message for command SENSOR_MULTILEVEL_GET
20:09:21.144 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 12: SECURITY NOT required on COMMAND_CLASS_SENSOR_MULTILEVEL
20:09:21.164 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 12: Command Class COMMAND_CLASS_SENSOR_MULTILEVEL is NOT required to be secured
20:09:21.182 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 12: Polling zwave:device:710d4836:node12:sensor_report
20:09:21.192 [DEBUG] [verter.ZWaveMultiLevelSensorConverter] - NODE 12: Generating poll message for COMMAND_CLASS_SENSOR_MULTILEVEL, endpoint 0
20:09:21.203 [DEBUG] [ass.ZWaveMultiLevelSensorCommandClass] - NODE 12: Creating new message for command SENSOR_MULTILEVEL_GET
20:09:21.211 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 12: SECURITY NOT required on COMMAND_CLASS_SENSOR_MULTILEVEL
20:09:21.218 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 12: Command Class COMMAND_CLASS_SENSOR_MULTILEVEL is NOT required to be secured
20:09:21.223 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 12: Polling zwave:device:710d4836:node12:thermostat_setpoint_heating
20:09:21.230 [DEBUG] [rter.ZWaveThermostatSetpointConverter] - NODE 12: Generating poll message for COMMAND_CLASS_THERMOSTAT_SETPOINT, endpoint 0
20:09:21.237 [DEBUG] [s.ZWaveThermostatSetpointCommandClass] - NODE 12: Creating new message for application command THERMOSTAT_SETPOINT_GET (Heating)
20:09:21.245 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 12: SECURITY NOT required on COMMAND_CLASS_THERMOSTAT_SETPOINT
20:09:21.250 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 12: Command Class COMMAND_CLASS_THERMOSTAT_SETPOINT is NOT required to be secured
20:09:21.256 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 12: Polling zwave:device:710d4836:node12:battery-level
20:09:21.260 [DEBUG] [ernal.converter.ZWaveBatteryConverter] - NODE 12: Generating poll message for COMMAND_CLASS_BATTERY endpoint 0
20:09:21.265 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 12: SECURITY NOT required on COMMAND_CLASS_BATTERY
20:09:21.273 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 12: Command Class COMMAND_CLASS_BATTERY is NOT required to be secured
20:09:21.278 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 12: Bump transaction 27183 priority from Get to Immediate
20:09:21.283 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 12: Adding to device queue
20:09:21.290 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 12: Added 27183 to queue - size 12
20:09:21.295 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
20:09:21.302 [DEBUG] [zwave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0B 00 13 0C 04 31 04 01 00 25 C3 3D
20:09:21.308 [DEBUG] [ding.zwave.handler.ZWaveSerialHandler] - NODE 12: Sending REQUEST Message = 01 0B 00 13 0C 04 31 04 01 00 25 C3 3D
20:09:21.316 [DEBUG] [ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 06
20:09:21.320 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=null[0], type=ACK[2], dest=255, callback=0, payload=
20:09:21.325 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=null[0], type=ACK[2], dest=255, callback=0, payload=
20:09:21.327 [DEBUG] [ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 04 01 13 01 E8
20:09:21.330 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - lastTransaction null
20:09:21.334 [DEBUG] [ding.zwave.handler.ZWaveSerialHandler] - Message SENT
20:09:21.337 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Received msg: ACK
20:09:21.343 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage started: TID 27183: [WAIT_RESPONSE] priority=Immediate, requiresResponse=true, callback: 195
20:09:21.347 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
[... cut ...]
20:09:21.420 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 12: Added 27185 to queue - size 13
20:09:21.430 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - TID 27183: Advanced to WAIT_REQUEST
20:09:21.433 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start. Holdoff false.
20:09:21.437 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 12: TID 27183: Transaction not completed
20:09:21.446 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty

cdjackson, 2018-10-24:
#720 did add the report function to send a multilevel sensor report to a remote - it added the sensor_report channel to do exactly this. If it’s not there now, then I’ll have to see if I can find the source :(

cdjackson, 2018-10-24:
Ok, found it. It wasn’t merged into master for some reason.

I’ll push the code and you might like to take a look - I’ll need to rebase as it’s 11 months old.

maxberger, 2018-10-24:
Cool, thanks! I'll wait for your push and then validate

maxberger, 2018-10-26:
See comment on pill request, this works and is merged!

closed 2018-10-26 by maxberger