openhab/org.openhab.binding.zwave #1051: Command Poll Period default doesn't work great with WD500Z-1

bigboxer23, 2018-11-02:
The Command Poll Period default of 1500 doesn't work great combined with the linear [WD500Z-1](https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/6) wall switch.  I have a rule triggering to turn some of these switches to off, and i see the command logged successfully to shut them off, but then a little over 1.5sec later i see another item state change hitting somewhere around 48-53 brightness for the light.  I believe this is because the switch has a dim transition that lasts longer than the 1.5 sec from start to finish, so the switch reports its current state when asked again.  Im able to either disable the CPP completely or set it to something longer to get around this issue.  I also see the WD500Z5-1 has config options to change that transition period, but the WD500Z-1 does not, so my only option to fix is to manually update all the CPP timings (which is not ideal with 100 switches).  This issue did not exist with the zwave 2.3 binding.  I've worked around it by modifying ZWaveConfigProvider, updating the default CPP value provided and building the jar myself.

Logs showing the rule triggering
2018-11-02 11:42:38.721 [WARN ] [se.smarthome.model.script.Bigboxer23] - MudRoomMotionTimer Off, Sensor Off, shutting off lights
2018-11-02 11:42:38.725 [ome.event.ItemCommandEvent] - Item 'GarageLights' received command 0
2018-11-02 11:42:38.726 [vent.ItemStateChangedEvent] - GarageLights changed from 53 to 0
2018-11-02 11:42:40.321 [vent.ItemStateChangedEvent] - GarageLights changed from 0 to 52

OpenHAB 2.3
Zwave binding 2.4.0 (used both the 10-31-18 version, and presently using current github branch code with modified ZWaveConfigProvider to automatically set the CPP to a higher default value).

cdjackson, 2018-11-02:
I’m not sure what you want changed? If the default doesn’t work well, then you can change it for devices that have issues - that’s why it’s changeable. What are you suggesting should be done (I don’t think there is any real suggestion)?

bigboxer23, 2018-11-02:
Suggestion would be to have a better default for these dimmers (or dimmers in general if other manufactures display similar behavior).  Having a default value which requires manual configuration (and more importantly a general awareness of what's even happening) seems incorrect to me.  At best users will figure this out and have to spend effort configuring around this, at worst they'll assume the binding is buggy and decline to use the platform.

cdjackson, 2018-11-02:
It’s not possible to have different defaults for different devices. The default is set to be good for most devices - if that’s not the case, then we can change it, but it doesn’t make sense to change the default for 1 device when we have over 800 others.

bigboxer23, 2018-11-02:
Totally get it.  Perhaps some sort of warning is warranted in the WD500Z-1 database doc is my only further recommendation to at least guide users down the right path.  Thanks for listening!

cdjackson, 2018-11-02:
Sure - we can add a comment in the database - there is a section to add user notes. Can you suggest something? Is this also linked to a device setting - eg a transition speed or something?

bigboxer23, 2018-11-02:
That's the problem, on that version of the switch there isn't a user settable transition speed (whereas the WD500Z5-1 has capacity to set that via params 7/8).

I'd say something like:

```
Note:  The default Command Poll Period of 1500ms can cause issues with state updating properly after a remote command is received due to the transition period between states of around 2.5s.  This manifests as a switch turned fully off in real life, but showing as 50% on in the OpenHAB UI. 
 Recommended to update this setting to > 2500ms to provide more consistent results.
```

cdjackson, 2018-11-03:
Thanks. I've added this to the database, and this will in turn automatically end up in the documentation.

closed 2018-11-03 by cdjackson