openhab/org.openhab.binding.zwave #1053: [ZWave] SceneNumber item not updated (Nodon CRC-3100)

ugomeda, 2018-11-04:
After a few hours of digging i think there is a bug in the ZWave binding. I'll report 2 bugs (Error 500 + item not updated) that are probably linked.

## Expected Behavior

When editing a Thing, the edit succeeds and the SceneNumber item is updated when I click on my device.

## Current Behavior

When adding the device, the SceneNumber is *initially* updated properly when i click on a button :

```
20:22:23.036 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:1d9a1461:node3' has been updated.
20:22:23.820 [INFO ] [smarthome.event.ItemStateChangedEvent] - Remote_SceneNumber changed from 3.0 to 1.0
20:22:24.766 [INFO ] [smarthome.event.ItemStateChangedEvent] - Remote_SceneNumber changed from 1.0 to 3.0
20:22:24.794 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:1d9a1461:node3' has been updated.
```

But once i edit anything on the settings, the item is not updated anymore :

```
20:23:48.871 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:1d9a1461:node3' has been updated.
20:23:49.997 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:1d9a1461:node3' has been updated.
20:23:50.936 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:1d9a1461:node3' has been updated.
20:23:51.875 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:1d9a1461:node3' has been updated.
20:23:52.932 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:1d9a1461:node3' has been updated.
```

Additionally, when i edit the name of the thing, i get "ERROR: 500 - Internal Server Error", with this json returned :

```json
{  
   "error":{  
      "message":"Duplicate channels zwave:device:1d9a1461:node3:scene_number",
      "http-code":500,
      "exception":{  
         "class":"java.lang.IllegalArgumentException",
         "message":"Duplicate channels zwave:device:1d9a1461:node3:scene_number",
         "localized-message":"Duplicate channels zwave:device:1d9a1461:node3:scene_number"
      }
   }
}
```

I was not able to find a stacktrace in the logs server-side. Both channels are displayed in the UI with the same name :

[image]

## Possible Solution

This device supports both "Central Scene" and "Scene Activation" modes. It might be possible the binding tries to create an item for both modes with the same name ?

## Steps to Reproduce (for Bugs)

I was able to reproduce this way :

1. Remove the existing remote from the things
2. Hard Reset the USB Z-Way controller
3. Re-pair the remote
4. Create an item on the SceneNumber channel
5. Click on the remove => The logs shows the SceneNumber item was activated
6. Edit the configuration of the device (modified the LED settings)
7. Click on the remote => SceneNumber is not updated anymore

## Your Environment

- openhab 2.3.0 (docker image openhab/openhab:2.3.0-amd64-debian), freshly installed
- binding-zwave - 2.3.0
- Nodon CRC310x (https://nodon.fr/support/NodOn-CRC-3-1-xx-UserGuide-171010-EN-interactive.pdf)

cdjackson, 2018-11-04:
Please use the latest binding - 2.3 is very old and I suspect that this will be resolved already if you use the snapshot.

ugomeda, 2018-11-05:
2.4.0-M4 didn't work either, i'll try 2.4.0-snapshot and keep you updated.

cdjackson, 2018-11-05:
Note that you will need to delete the thing and add it back so that the new definition is read in.

ugomeda, 2018-11-06:
Setup 2.4.0-snapshot brand new. Good things :

- Error 500 is gone
- Scene Number is not duplicated

But the SceneNumber is not updated once i update the name of the thing. Before changing the name :

```
21:35:37.906 [INFO ] [smarthome.event.ItemStateChangedEvent] - ZWaveNode002_SceneNumber changed from NULL to 1.0
21:35:37.922 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:e2017e67:node2' has been updated.
21:35:41.481 [INFO ] [smarthome.event.ItemStateChangedEvent] - ZWaveNode002_SceneNumber changed from 1.0 to 2.0
21:35:41.504 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:e2017e67:node2' has been updated.
21:35:44.970 [INFO ] [smarthome.event.ItemStateChangedEvent] - ZWaveNode002_SceneNumber changed from 2.0 to 2.3
21:35:44.999 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:e2017e67:node2' has been updated.
21:35:47.598 [INFO ] [smarthome.event.ItemStateChangedEvent] - ZWaveNode002_SceneNumber changed from 2.3 to 1.3
21:35:49.275 [INFO ] [smarthome.event.ItemStateChangedEvent] - ZWaveNode002_SceneNumber changed from 1.3 to 2.3
21:35:49.300 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:e2017e67:node2' has been updated.
21:36:00.713 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:e2017e67:node2' has been updated.
```

After changing the name :

```
21:36:04.387 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:e2017e67:node2' has been updated.
21:36:08.108 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:e2017e67:node2' has been updated.
21:36:11.709 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:device:e2017e67:node2' has been updated.
```

ugomeda, 2018-11-06:
Please let me know if i can provide any info that would help for resolving this issue.

cdjackson, 2018-11-06:
Please provide a log file so I can see what is happening.

ugomeda, 2018-11-06:
Configuration changed at 22:14:57.132

[events.log](https://github.com/openhab/org.openhab.binding.zwave/files/2555214/events.log)
[openhab.log](https://github.com/openhab/org.openhab.binding.zwave/files/2555215/openhab.log)

cdjackson, 2018-11-06:
The log is empty.

ugomeda, 2018-11-06:
Sorry i didn't know how to enable full logging for zwave only. Here are the full logs with `log:set DEBUG org.openhab.binding.zwave`.

- Added the remote at 23:33:27
- Edited the settings at 23:34:12

[events.log](https://github.com/openhab/org.openhab.binding.zwave/files/2555270/events.log)
[openhab.log](https://github.com/openhab/org.openhab.binding.zwave/files/2555271/openhab.log)

cdjackson, 2018-11-06:
Since the device is working, I would guess that you have a configuration issue rather than it being a binding bug.

ugomeda, 2018-11-07:
You're right. I was able to reset the device and it started working again. Re-configure... down.
Turns out there is a "Lifeline" setting that has to be set to "controller". Once this is set, it works fine :).

The settings displayed in the UI are not fetched from the device, it's only the default settings before the first configuration ?

Anyway, thank you so much for the help ! And sorry again for not properly reading the readme yesterday.

closed 2018-11-07 by ugomeda