openhab/org.openhab.binding.zwave #1058: Fibaro FGT-001 with sensor - Generating message failed for command class = THERMOSTAT_MODE, endpoint = 0

Tymo3, 2018-11-08:
Hi,
Make long story short - cannot pass value "3" to thermostat_mode.

Full story:
In Z-Wave binding there are two values for Thermostat_Mode for Fibaro FGT-001 Thermostat Valve
0 - Off
1 - Heating
But in my case, valve at "Heating" state is not taking care for paired senor temperature (i.e. valve is heating at full open state).
In manual you can find that thermostat_mode can be also "Auto".
So I am wondering if passing "3" can solve problem.
Can you change it accordingly? 
(or you have already tested that "Auto" does not work with this valve).

OH2 version: 2.3.0
Z-Wave Binding version: 2.3.0
Device name: Raspberry Pi, manual installation 

Passing value "3" makes an warning:
[...]
2018-11-08 23:33:22.886 [ome.event.ItemCommandEvent] - Item 'Heater_ThermostatMode1' received command 3

==> /var/log/openhab2/openhab.log <==
2018-11-08 23:33:22.905 [WARN ] [nverter.ZWaveThermostatModeConverter] - NODE 36: Generating message failed for command class = THERMOSTAT_MODE, endpoint = 1

==> /var/log/openhab2/events.log <==
2018-11-08 23:33:23.396 [vent.ItemStateChangedEvent] - Heater_ThermostatMode1 changed from 1 to 3
[...]

**ITEMS file**
Group grp_TEST "Test group"
Switch   Heater_SensorStatus               "SensorStatus              [MAP(kaloryfery.map):%s]"      (grp_TEST) {channel="zwave:device:e765eb82:node36:sensor_binary"}
Number   Heater_ThermostatMode             "ThermostatMode            [%s]"            <radiator>    (grp_TEST) {channel="zwave:device:e765eb82:node36:thermostat_mode"}
Number   Heater_ThermostatMode1            "ThermostatMode1           [%2.0f]"         <radiator>    (grp_TEST) {channel="zwave:device:e765eb82:node36:thermostat_mode1"}
Number   Heater_ThermostatSetpointHeating  "ThermostatSetpointHeating [%2.0f°C]"       <heating>     (grp_TEST) {channel="zwave:device:e765eb82:node36:thermostat_setpoint_heating"}
Number   Heater_ThermostatSetpointHeating1 "ThermostatSetpointHeating1[%2.0f°C]"       <heating>     (grp_TEST) {channel="zwave:device:e765eb82:node36:thermostat_setpoint_heating1"}
Switch   Heater_AlarmPower                 "AlarmPower                [%s]"            <alarm>       (grp_TEST) {channel="zwave:device:e765eb82:node36:alarm_power" }
Switch   Heater_AlarmPower1                "AlarmPower1               [%s]"            <alarm>       (grp_TEST) {channel="zwave:device:e765eb82:node36:alarm_power1" }
Switch   Heater_AlarmSystem                "AlarmSystem               [%s]"            <alarm>       (grp_TEST) {channel="zwave:device:e765eb82:node36:alarm_system" }
Switch   Heater_AlarmSystem1               "AlarmSystem1              [%s]"            <alarm>       (grp_TEST) {channel="zwave:device:e765eb82:node36:alarm_system1" }
Number   Heater_BatteryLevel               "BatteryLevel              [%d%%]"          <battery>     (grp_TEST) {channel="zwave:device:e765eb82:node36:battery-level"}
Number   Heater_BatteryLevel1              "BatteryLevel1             [%d%%]"          <battery>     (grp_TEST) {channel="zwave:device:e765eb82:node36:battery-level1"}
Number   Heater_TimeOffset                 "TimeOffset                [%d]"            <clock>       (grp_TEST) {channel="zwave:device:e765eb82:node36:time_offset"}
Number   Heater_TimeOffset1                "TimeOffset1               [%d]"            <clock>       (grp_TEST) {channel="zwave:device:e765eb82:node36:time_offset1"}
Number   Heater_Temperature                "Temperature               [%2.0f°C]"       <temperature> (grp_TEST) {channel="zwave:device:e765eb82:node36:sensor_temperature"}
Number   Heater_Temperature2               "Temperature2              [%2.0f°C]"       <temperature> (grp_TEST) {channel="zwave:device:e765eb82:node36:sensor_temperature2"}
Switch   Heater_AlarmPower2                "AlarmPower2               [%s]"            <alarm>       (grp_TEST) {channel="zwave:device:e765eb82:node36:alarm_power2" }
Number   Heater_BatteryLevel2              "BatteryLevel2             [%d%%]"          <battery>     (grp_TEST) {channel="zwave:device:e765eb82:node36:battery-level2"}

**SITEMAP file**
        Text item=grp_TEST {
            Default   item=Heater_SensorStatus              
            Selection item=Heater_ThermostatMode               mappings=[ 0="Off", 1="Heat", 3="Auto", 5="Resume",11="Heat Economy"]    
            Selection item=Heater_ThermostatMode1              mappings=[ 0="Off", 1="Heat", 3="Auto", 5="Resume",11="Heat Economy"]
            Setpoint  item=Heater_ThermostatSetpointHeating    minValue=15 maxValue=30 step=1
            Setpoint  item=Heater_ThermostatSetpointHeating1   minValue=15 maxValue=30 step=1
            Default   item=Heater_AlarmPower                
            Default   item=Heater_AlarmPower1               
            Default   item=Heater_AlarmSystem               
            Default   item=Heater_AlarmSystem1              
            Default   item=Heater_BatteryLevel              
            Default   item=Heater_BatteryLevel1             
            Default   item=Heater_TimeOffset                
            Default   item=Heater_TimeOffset1               
            Default   item=Heater_Temperature               
            Default   item=Heater_Temperature2              
            Default   item=Heater_AlarmPower2               
            Default   item=Heater_BatteryLevel2                         
        }

cdjackson, 2018-11-08:
> Number Heater_ThermostatMode "ThermostatMode [%s]" (grp_TEST) {channel="zwave:device:e765eb82:node36:thermostat_mode"}
> Number Heater_ThermostatMode1 "ThermostatMode1 [%2.0f]" (grp_TEST) {channel="zwave:device:e765eb82:node36:thermostat_mode1"}
> Number Heater_ThermostatSetpointHeating "ThermostatSetpointHeating [%2.0f°C]" (grp_TEST) {channel="zwave:device:e765eb82:node36:thermostat_setpoint_heating"}
> Number Heater_ThermostatSetpointHeating1 "ThermostatSetpointHeating1[%2.0f°C]" (grp_TEST) {channel="zwave:device:e765eb82:node36:thermostat_setpoint_heating1"}
> Switch Heater_AlarmPower "AlarmPower [%s]" (grp_TEST) {channel="zwave:device:e765eb82:node36:alarm_power" }
> Switch Heater_AlarmPower1 "AlarmPower1 [%s]" (grp_TEST) {channel="zwave:device:e765eb82:node36:alarm_power1" }
> Switch Heater_AlarmSystem "AlarmSystem [%s]" (grp_TEST) {channel="zwave:device:e765eb82:node36:alarm_system" }
> Switch Heater_AlarmSystem1 "AlarmSystem1 [%s]" (grp_TEST) {channel="zwave:device:e765eb82:node36:alarm_system1" }
> Number Heater_BatteryLevel "BatteryLevel [%d%%]" (grp_TEST) {channel="zwave:device:e765eb82:node36:battery-level"}
> Number Heater_BatteryLevel1 "BatteryLevel1 [%d%%]" (grp_TEST) {channel="zwave:device:e765eb82:node36:battery-level1"}
> Number Heater_TimeOffset "TimeOffset [%d]" (grp_TEST) {channel="zwave:device:e765eb82:node36:time_offset"}
> Number Heater_TimeOffset1 "TimeOffset1 [%d]" (grp_TEST) {channel="zwave:device:e765eb82:node36:time_offset1"}
> Number Heater_Temperature "Temperature [%2.0f°C]" (grp_TEST) {channel="zwave:device:e765eb82:node36:sensor_temperature"}
> Number Heater_Temperature2 "Temperature2 [%2.0f°C]" (grp_TEST) {channel="zwave:device:e765eb82:node36:sensor_temperature2"}
> Switch Heater_AlarmPower2 "AlarmPower2 [%s]" (grp_TEST) {channel="zwave:device:e765eb82:node36:alarm_power2" }
> Number Heater_BatteryLevel2 "BatteryLevel2 [%d%%]" (grp_TEST) {channel="zwave:device:e765eb82:node36:battery-level2"}

Please update to a more recent version, and also provide a the XML file from the device so I can see what it supports.

Tymo3, 2018-11-09:
Sorry for my ignorance - should I upgrade to 2.4.0 or newest 2.3.0? As after upgrade only binding to 2.4.0, (without upgrading OH from 2.3 to 2.4) only zwave serial controller was online, rest z-wave things was UNINITIALIZED - HANDLER_CONFIGURATION_PENDING.

Tymo3, 2018-11-10:
Ok, updated to 2.4 M5 (I made clean install)
openHAB Core 	2.4.0.M5
ZWave Binding	2.4.0.M5

XML here:
[network_f3240cb4__node_36.zip](https://github.com/openhab/org.openhab.binding.zwave/files/2568280/network_f3240cb4__node_36.zip)

Items, sitemap, transformations
[test.zip](https://github.com/openhab/org.openhab.binding.zwave/files/2568284/test.zip)

openhab.log
2018-11-10 10:49:47.439 [ome.event.ItemCommandEvent] - Item 'Heater_ThermostatMode1' received command 3
2018-11-10 10:49:47.444 [nt.ItemStatePredictedEvent] - Heater_ThermostatMode1 predicted to become 3
2018-11-10 10:49:47.480 [vent.ItemStateChangedEvent] - Heater_ThermostatMode1 changed from 0 to 3

==> /var/log/openhab2/openhab.log <==
2018-11-10 10:49:47.479 [ERROR] [lass.ZWaveThermostatModeCommandClass] - NODE 36: Unsupported mode type 3
2018-11-10 10:49:47.484 [WARN ] [nverter.ZWaveThermostatModeConverter] - NODE 36: Generating message failed for command class = COMMAND_CLASS_THERMOSTAT_MODE, endpoint = 1

==> /var/log/openhab2/events.log <==
2018-11-10 10:49:55.274 [vent.ItemStateChangedEvent] - Heater_ThermostatMode1 changed from 3 to 0
2018-11-10 10:49:55.576 [vent.ItemStateChangedEvent] - Heater_ThermostatSetpointHeating1 changed from NULL to 22
2018-11-10 10:49:55.869 [vent.ItemStateChangedEvent] - Heater_TimeOffset1 changed from NULL to 0
2018-11-10 10:49:56.173 [vent.ItemStateChangedEvent] - Heater_Temperature2 changed from 21.3 to 21.6

[test.zip](https://github.com/openhab/org.openhab.binding.zwave/files/2568284/test.zip)