openhab/org.openhab.binding.zwave #1061: Typo in "zwave_lastheat" in ZWaveThingHandler

mhilbush, 2018-11-15:
Shouldn't this be PROPERTY_LASTHEAL?

https://github.com/openhab/org.openhab.binding.zwave/blob/master/src/main/java/org/openhab/binding/zwave/handler/ZWaveThingHandler.java#L1585

cdjackson, 2018-11-15:
Yep - thanks.
Was there a reason you found this? ie is there a bug (other than the incorrect name) or did you just stumble across this?

I'm not sure why it's removing this property here, and given it's currently not doing anything, I intend to remove this line.

mhilbush, 2018-11-15:
Stumbled on it completely by accident.

As I was messing with the ZSMOKE device, I was wondering what was producing this line when I changed the name of the Thing. The typo was nearby caught my eye, so I figured I'd log it.

```
2018-11-15 10:24:20.613 [DEBUG] [rg.openhab.binding.zwave.handler.ZWaveThingHandler] - NODE 108: Update ASSOCIATION group_ZWaveAssociationGroup [index=1, name=null, profile1=null, profile2=null, associations=[node_1]]: Adding node_1
```

cdjackson, 2018-11-15:
Ok, thanks. I will remove this then.

PR #1062 (2018-11-16): Remove property that is set in error.
Closes #1061
closed 2018-11-16 by cdjackson