openhab/org.openhab.binding.zwave #1065: Offline / Failed Devices Are Not Marked Offline / Failed

fh-eureka, 2018-11-17:
I'm opening a new issue because I cannot reopen the old issue #1024
Unfortunately the last merged pull request #1035 did not fix this issue.

Here is the current situation (copied from my last comment):
I have attached the log file (starting from the time of zwave binding restart and ending after 2 hours)

Looking at one example node:
**Node 9** (Fibaro Roller Shutter) (this should be offline)
This node was marked offline at 11:07:18.711 (3 minutes after restart)
`NODE 9: Setting OFFLINE`
Then immediately at 11:07:18.871 it was changed to REQUEST_NIF
`NODE 9: Node advancer - advancing to REQUEST_NIF`
Is this the correct behavior? After it was set to OFFLINE it was changed to REQUEST_NIF?
Then at 11:08:18.990:
`NODE 9: No data from device, but it was ACK'd. Possibly not supported? (Try 0)`
Can you get ACK'd if the device is completely offline (power disconnected)?
Then at 11:41:31.639 (37 minutes after restart):
`NODE 9: Node advancer: Restored from file - skipping static initialisation`
This looks strange to me. That after 37 minutes it was decided to skip static initialisation and load from file!
Then the last Node advancer entry at 11:43:44.915:
`NODE 9: Node advancer - advancing to DONE`
I guess this did set the node to ONLINE! (HABmin shows that the node is online)
Then after that, there is a polling request every 30 minutes
And also a sendTransaction every ~15 minutes
All of them are failing but the device remains online.

Log file:
[openhab.zip](https://github.com/openhab/org.openhab.binding.zwave/files/2526218/openhab.zip)

fh-eureka, 2018-12-06:
Hi @cdjackson 
Did you get a chance to look at the attached log?

cdjackson, 2018-12-06:
No - not yet.

flozano, 2021-06-28:
I'm having this issue, all my (fibaro) devices are unplugged and stored in a box, but all of them show up as ONLINE.