openhab/org.openhab.binding.zwave #1071: [Bug] Some Z-Wave item values are not displayed in Paper UI

bodiroga, 2018-11-22:
Hi @cdjackson!

I have seen this problem many times, so it is time to report it. There are some Z-wave channels whose item values are not displayed in the control tab of Paper UI. Here you have an image where you can see what I mean:

[image]

It seems that Switch items with a non-default StateDescription ("Ok", "Alarm" in most alarm channels) are empty in Paper UI.

I suppose that this is a known issue (it's very common in multisensors), but can we fix it? Is it a Z-Wave problem or a Paper UI error?

Thanks for your work Chris!

Aitor

cdjackson, 2018-12-01:
The channels that are missing are probably notification channels - they can't be polled, so if they don't send a notification, then they won't be updated.

bodiroga, 2018-12-02:
Hi @cdjackson!

No, no, items are correctly updated, the device is sending the information. The problem is that Paper UI is not able to render the state of the item and I think that it's related to the custom StateDescription implemented on those channels ("ok"/"alarm" instead of "on"/"off").

cdjackson, 2018-12-02:
Please provide a logfile so I can see what is happening.

bodiroga, 2018-12-03:
Hi @cdjackson!

It seems that this is the same problem I'm seeing: https://community.openhab.org/t/solved-fibaro-fgms-001-cannot-see-alarm-off-on-in-gui-paperui/25685

The binding behaves correctly (channel state updates are sent to the framework and the item is updated), but for some reason Paper UI doesn't like the stateDescription transformation you are using for alarm channels ("ON" -> "Alarm", "OFF" -> "OK"). It's a pitty because it is very confusing for new users. Perhaps I should create a new issue in the ESH repository, what do you think? I remember reading a conversation between you and Kai about this, but now I'm not able to find it :disappointed:

Thanks!

cdjackson, 2019-08-11:
This seems to be a problem with PaperUI rather than the binding so I'd propose to close this.

cdjackson, 2019-08-11:
Feel free to reopen if there is a binding issue.

closed 2019-08-11 by cdjackson