openhab/org.openhab.binding.zwave #1076: Need a way to add new items to operational openhab database between releases

llamahunter, 2018-11-25:
Device is not in the 2.3.0 database, and no way to add xml file exported from online database into running openhab system.

There really needs to be a way to add devices without switching to snapshot builds.  Just read a directory to pick up extra database xml files to augment the ones that are built in.

* OH2 Version:
2.3

* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console):
2.3.0

If the problem is associated with a device, please provide the following -:

* Device name: 
Dome Motion detector
* Device version: 
1.0?
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list):
https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/797
* If the device is not in the database, provide a link to the manual.
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.

cdjackson, 2018-11-25:
A release version can not have new things added to it. This is a constraint with ESH and is applicable for all bindings. The database links in with the java code - changes to the database may require code changes and this is also not possible with a released version. 

If you want to use the newer devices then you need to update to a newer binding that has the device available with coherent code and database.

llamahunter, 2018-11-30:
So... this seems like a design flaw, no?  In ESH perhaps, but still. New zwave devices come out all the time.  What's the recommended way to gain access to devices that are in the online database, but not in the released zwave binding software?

cdjackson, 2018-11-30:
The XML files trigger features in the Java code - if we separate the two (XML and code) then we have the problem that people will expect to be able to update the XML files without changing the Java, and this is not always true. We then have a configuration nightmare to make sure people run the right versions of code and XML.

Think about other bindings (eg Hue) - when a new device comes out, it requires a binding update to add support - it’s exactly the same here only you think that because much of the coding is done in XML that the two can be split, and this is not true.

llamahunter, 2018-11-30:
Ok, well, given the rapid rate of zwave device releases, perhaps there need to be much more frequent openhab releases?  I think the last one was approx 6 months ago. and that one doesn't appear to have zwave database update from more recent than about 8-9 months ago.

mhilbush, 2018-11-30:
Snapshot releases come out almost daily. Milestone releases about once a month. I've been running snapshot releases for 2 years. I'm just careful about which ones I install in my production systems.

cdjackson, 2018-11-30:
Maintenance releases are every 4 weeks. Snapshots every few days, and major releases every 6 months.

The ZWave database for V2.3, as with the Java code in 2.3, is from the release date of 2.3, which was updated in June or July when 2.3 was released so it’s not 9 months old.

llamahunter, 2018-12-01:
Ok, well, I switched all of openhab to a snapshot release 2.4M6, and now the device is being recognized by its type:id.  The xml file was first added to the zwave binding on Mar 11, 2018, and was *last* touched in the database on May 19, 2018, so unclear how it's not 8-9 months old in the 2.3 release, where it's completely unknown.
https://github.com/openhab/org.openhab.binding.zwave/commits/master/ESH-INF/thing/elexa/dmms1_0_0.xml

llamahunter, 2018-12-01:
And it's not a very user friendly experience to say that you have to run unstable snapshots in order to use readily available consumer devices.

cdjackson, 2018-12-01:
2.3 was released in May, so this just missed the cutoff by about 4 days.  Glad it’s working now though.

cdjackson, 2018-12-01:
You can use the milestone releases if you don’t like the snapshots…

You’re happy instead that people can mix and match untested binding and definitions, with the ultimate result that there will be incompatibilities though, and things won’t work ;)

closed 2018-12-01 by cdjackson

AngelosF, 2018-12-01:
> I switched all of openhab to a snapshot release 2.4M6

He is using the Milestone 6 (testing) but he is confusing this with the unstable (snapshot) builds branch

https://community.openhab.org/t/oh-2-4-0-m6-testing-results/57321

AngelosF, 2018-12-01:
snapshot != milestone
milestone != unstable
but... whatever you want to believe 📦