openhab/org.openhab.binding.zwave #1085: OSGI-INF/org.openhab.binding.zwave.discovery.ZWaveUsbSerialDiscoveryParticipant still being built

cdjackson, 2018-12-08:
@kaikreuzer the UsbDiscoveryParticipant was removed from the binding recently, but CI is still building the XML file which is causing problems when the binding starts. It looks to me like the service has been properly removed, and the dependencies also removed from the manifest, so I'm not sure why this is still being created.

I wondered if there was some caching in Jenkins, but can't see anything. Do you have any idea.

cdjackson, 2018-12-09:
https://community.openhab.org/t/z-wave-oh2-4-0-s1454-classnotfoundexception/59058

cdjackson, 2018-12-09:
Tried adding a file to the OSGI-INF folder to see if the absence of this folder was an issue - this didn't help.

I note that the XML file isn't being generated during the build, and does seem to be stuck in cache somewhere -:

[image]

kaikreuzer, 2018-12-09:
I can see the file at https://ci.openhab.org/job/openHAB2-Bundles/ws/bindings/org.openhab.binding.zwave/OSGI-INF/.
I guess it isn't cleaned by the build - I will manually delete the workspace now, so I'd hope that the file is gone from the next build.

codeaholics, 2018-12-13:
@kaikreuzer I can't see the Jenkins configuration, but normally in the Git plugin there's an option to run a clean before and/or after pull. I'm not sure why it's an option because I've never yet encountered a situation when doing a clean is a bad idea! Really, it should be enabled after the pull as it avoids precisely this kind of issue where deleted files are left in the Jenkins workspace.

kaikreuzer, 2018-12-17:
Yes, this would be an option. For our bigger repos, we prefer saving the time, but for smaller ones, it should not have much impact.

closed 2018-12-17 by kaikreuzer

codeaholics, 2018-12-17:
A clean shouldn’t add very much to the build time, especially when weighed up against the correctness of actually building the code that’s in the repo and not the code that happens to be lying around on the build server!