openhab/org.openhab.binding.zwave #1091: Issues with LOCK configuration

PavelasV, 2018-12-18:
System: windows 10/ RPI 3 b+ 
OH cersion: 2.4.0
z-wave binding 2.4.0 - downloaded from paperui

Device name: Keyless Connected Smart Door Lock
https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/948

I have issues with this specific lock, it works, I can lock/unlock it, can see alarm status and I can change user codes with paper ui/API/habmin, but it works randomly, sometimes binding changes all codes sometime not. Very often user code does not change at all, or if I trying to change, let's say 10 user codes, I get 3 or 4 changed from 10, or just 1.  the number of successfully changed codes is random. Sometimes binding left old ones, sometimes the binding puts empty codes just changing the user label to new one. 

Then I am trying to change user codes by setting new config throw API with POST to "/rest/things/"+thingUID + "/config" URL I getting the response with success code, and right after this I am trying to get thing json from API - it responds with new user codes(they not work), but if I wait some time - I getting response with wrong ones, so as I understand this is not API or user interface issue.

I attached my log file and xml of the lock, I am using Aeon Labs Z-Stick Gen 5 with windows and tested it with RPI 3 b+ - same issues on both platforms. 

Thank you!

cdjackson, 2019-02-09:
Please provide a logfile showing the issue.

cdjackson, 2019-08-11:
Closing this as no information was provided. If the problem persists, please reopen and provide the requested information.

closed 2019-08-11 by cdjackson