openhab/org.openhab.binding.zwave #1099: Raising minimum temperature offset for Fibaro Motion Sensor FGMS-001 from 1°C to 0.1°C

KaiLuettmann, 2018-12-28:
* OH2 Version: 2.4.0.1
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): release
* Device name: Fibaro Motion Sensor FGMS-001
* Firmware version: 3.3
* Link to [database](https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/115)

Improvement-Issue regarding [this](https://community.openhab.org/t/raising-minimum-temperature-offset-for-fibaro-motion-sensor-fgms-001-from-1-c-to-0-1-c/61422) thread 

[image]

Currently it is only possible to set an offset to 1°C. In my optinion nobody use a offset from ±1000°C. A good way is to use 0,1°C steps instead of 1°C.

cdjackson, 2018-12-28:
As discussed on the thread, this is not something the binding can change as this is a configuration parameter in the device itself and NOT the binding.

KaiLuettmann, 2018-12-28:
Sorry, I missunderstand you. I thought the issue should be labeled with Fibaro. The relevant [file](https://github.com/openhab/org.openhab.binding.zwave/blob/master/ESH-INF/thing/fibaro/fgms001_3_2.xml) is in this repository. Where should I create the Issue?

cdjackson, 2018-12-28:
Sorry for the confusion. There is nothing that we can do here - this is solely interpreted by the device - the binding is simply passing the data to the device.

Please read the manual from the device - it is the same I think as shown in the database. It is already showing that the offset is in degrees. We can't therefore change this to provide a better resolution as this calibration is performed inside the device.

[image]

KaiLuettmann, 2018-12-28:
Ok, now I understand your valid objection. Thank you very much.

closed 2019-01-06 by cdjackson