openhab/org.openhab.binding.zwave #1107: Shock sensor is not triggering alarms. What should I fix?

JesperWe, 2019-01-03:
I am running `org.openhab.binding.zwave-2.5.0-SNAPSHOT.jar` from Jan 2 2019 and trying to debug why alarms triggered on a Vision ZS5101 shock sensor is not reported in OpenHAB.

Since I found several potential issues I have to ask some questions (mostly for @cdjackson  i suppose...) before I can submit a fix.

When the sensor alarm triggers I get the following log:
```
WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 10 00 04 00 06 0A 71 05 02 00 00 FF 07 00 01 02 6A 
[nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=6, callback=0, payload=00 06 0A 71 05 02 00 00 FF 07 00 01 02 
[nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=6, callback=0, payload=00 06 0A 71 05 02 00 00 FF 07 00 01 02 
[nal.protocol.ZWaveTransactionManager] - lastTransaction null
[nal.protocol.ZWaveTransactionManager] - NODE 6: Application Command Request (ALIVE:DONE)
[ng.zwave.internal.protocol.ZWaveNode] - NODE 6: resetResendCount initComplete=true isDead=false
[ng.zwave.internal.protocol.ZWaveNode] - NODE 6: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
[ng.zwave.internal.protocol.ZWaveNode] - NODE 6: SECURITY NOT required on COMMAND_CLASS_ALARM
[tocol.commandclass.ZWaveCommandClass] - NODE 6: Received COMMAND_CLASS_ALARM V4 NOTIFICATION_REPORT
[.commandclass.ZWaveAlarmCommandClass] - NODE 6: NOTIFICATION report - 2 = 0, event=0, status=255, plen=1
[.commandclass.ZWaveAlarmCommandClass] - NODE 6: Alarm Type = BURGLAR (2)
[ding.zwave.handler.ZWaveThingHandler] - NODE 6: Got an event from Z-Wave network: ZWaveAlarmValueEvent
[ding.zwave.handler.ZWaveThingHandler] - NODE 6: Got a value event from Z-Wave network, endpoint = 0, command class = COMMAND_CLASS_ALARM, value = 255
[ternal.converter.ZWaveAlarmConverter] - NODE 6: Alarm converter processing NOTIFICATION
[ternal.converter.ZWaveAlarmConverter] - NODE 6: Alarm converter NOTIFICATION event is 0, type OnOffType
[ternal.converter.ZWaveAlarmConverter] - NODE 6: Alarm converter NOTIFICATION event is 0, channel alarm_access is not implemented.
[ternal.converter.ZWaveAlarmConverter] - NODE 6: Alarm converter processing NOTIFICATION
[ternal.converter.ZWaveAlarmConverter] - NODE 6: Alarm converter NOTIFICATION event is 0, type OnOffType
[ding.zwave.handler.ZWaveThingHandler] - NODE 6: Updating channel state zwave:device:8ceebde6:node6:alarm_tamper to OFF [OnOffType]
[nal.protocol.ZWaveTransactionManager] - NODE 6: Commands processed 1.
[nal.protocol.ZWaveTransactionManager] - NODE 6: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@3e06b91b.
[nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
[nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
[nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
[nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
```
...so...

`ZWaveIncomingEvent` gets a `ZWaveAlarmValueEvent` of command class `COMMAND_CLASS_ALARM`.

This fails all the cases in the `switch (event.getCommandClass()) {}` and moves on to the line 

`State state = channel.getConverter().handleEvent(channel, event);`

...where it is somehow decided that this is a `NOTIFICATION`, not an `ALARM`?

QUESTION 1: Is this correct?

Ok, so we move on to `handleNotificationReport()` which says that the event is `0`and type is `OnOffType`

The ESH_INF database has:
```
<channel id="alarm_access" typeId="alarm_access">
        <label>Shock Alarm</label>
        <properties>
          <property name="binding:*:OnOffType">COMMAND_CLASS_ALARM;type=BURGLAR, event=2</property>
        </properties>
</channel>
```
QUESTION 2: There seems to be a mismatch in the event (`0`vs `2`) here. Who is correct?

Somehow this event is still classified (from the ESH_INF file or somewhere else??) as `alarm_access`, and then the following `switch (channelType) {}` fails to match any case and reports "not implemented", then returns null which causes `ZWaveIncomingEvent` to not update any states. The event has now for all practical purposes been ignored.

QUESTION 3: This also seems wrong in maybe more than one way. Could you explain the correct behavior? Then I could submit a patch.

cdjackson, 2019-01-05:
> QUESTION 1: Is this correct?

Yes, this is considered a NOTIFICATION (as per the logs).

> QUESTION 2: This seems wrong in maybe more than one way. 

We should change the channel type to ```alarm_burglar``` and then it should work ok. Please can you update the database (or if not, let me know and I will do it).

JesperWe, 2019-01-05:
Ok, but there are around 50 devices in the database with `alarm_access` channels. Should all of these be `alarm_burglar` instead?

cdjackson, 2019-01-05:
No - not unless they are sending the same reports and you probably don’t know what the other devices are sending do you?

Please just update the one device unless you know the type of reports sent by all other devices, and please don’t do a PR to update XMLs themselves as these are auto generated - it needs to be done directly in the database.

Thanks.

JesperWe, 2019-01-05:
But there must be something wrong with all those others, no? `alarm_access` events are not supported at all in the static event HashMap, so all of those will fail as well.

cdjackson, 2019-01-05:
Ok, if you actually know what the notifications are, then please update them. However I doubt that you know all these other devices do you?  If you go and change something that is wrong, to something else that is wrong, have you really achieved anything? I think not.

Changing all the others will not mean that they work any better (ie it does not mean it will work at all), so I would suggest to leave them unless you actually know what the devices are sending (ie the exact notifications and events). Randomly changing channels without clear information in my mind is not a good idea.

JesperWe, 2019-01-05:
I agree completely, I was just wondering whether the `access_alarm` channels was actually *supposed* to be implemented but was lost in some previous update. If this is not the case I will just update the two devices I have that currently are specifying `alarm_access` channels.

cdjackson, 2019-01-05:
Are you sure that both your two devices are sending the same events? If not, the channels could be different. What I’ve suggested was based on the log that you sent - it is not necessarily the same for a different device, so please check.

JesperWe, 2019-01-05:
No, I am only sure that they have `alarm_access` channels defined in the database and that this channel is not implemented in the code so those events are ignored. 

For the Shock sensor it is the Shock alarm, I have requested a DB update for that. 

For the door lock I will investigate further.

JesperWe, 2019-01-05:
So the other device I have that was using the unimplemented `alarm_access` channel is an [ID Lock 150](https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/887)

In this case the event is not really an alarm, so I tried to change it to a `notification_access_control` event, which the DB complained about as being deprecated. In this case I think the channel should actually be removed, since it has been replaced by the `alarm_raw` channel. But there is no UI that I can see for removing a channel.

JesperWe, 2019-01-05:
The lock sends events when it is locked or unlocked, either manually or using codes or keys. Using the current device xml these events will display "channel alarm_access not implemented" (main/java/org/openhab/binding/zwave/internal/converter/ZWaveAlarmConverter.java line 337).

I did get rid of this by changing the xml to use the `notification_access_control` channel, but the DB does not like that (deprecated) so I think the channel should be removed since it is redundant anyway (the event information is available from the `alarm_raw` channel).

cdjackson, 2019-01-05:
> 
> The lock sends events when it is locked or unlocked

Sure - but what events? 

Can you either be more specific, or provide a log so that I can understand what it is actually doing. Otherwise it’s hard to comment on what should be implemented.

JesperWe, 2019-01-05:
This is an example with my modified XML that has alarm_access changed to notification_access_control:

```
2019-01-05 14:07:26.927 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 23 00 04 00 07 1D 98 81 58 6C 91 26 3D 96 2F 05 D9 00 32 DE 62 F8 63 54 52 2A 6D 55 57 6D 52 BD AB 29 5C 0A 
2019-01-05 14:07:26.928 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=7, callback=0, payload=00 07 1D 98 81 58 6C 91 26 3D 96 2F 05 D9 00 32 DE 62 F8 63 54 52 2A 6D 55 57 6D 52 BD AB 29 5C 
2019-01-05 14:07:26.928 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=7, callback=0, payload=00 07 1D 98 81 58 6C 91 26 3D 96 2F 05 D9 00 32 DE 62 F8 63 54 52 2A 6D 55 57 6D 52 BD AB 29 5C 
2019-01-05 14:07:26.928 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction null
2019-01-05 14:07:26.929 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Application Command Request (ALIVE:DONE)
2019-01-05 14:07:26.929 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: resetResendCount initComplete=true isDead=false
2019-01-05 14:07:26.929 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: Decapsulating COMMAND_CLASS_SECURITY
2019-01-05 14:07:26.929 [DEBUG] [mmandclass.ZWaveSecurityCommandClass] - NODE 7: SECURITY_RXD 71 05 00 00 00 FF 06 02 00 
2019-01-05 14:07:26.929 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
2019-01-05 14:07:26.930 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 7: Received COMMAND_CLASS_ALARM V4 NOTIFICATION_REPORT
2019-01-05 14:07:26.930 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 7: NOTIFICATION report - 0 = 0, event=2, status=255, plen=0
2019-01-05 14:07:26.930 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 7: Alarm Type = ACCESS_CONTROL (0)
2019-01-05 14:07:26.930 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got an event from Z-Wave network: ZWaveAlarmValueEvent
2019-01-05 14:07:26.930 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got a value event from Z-Wave network, endpoint = 0, command class = COMMAND_CLASS_ALARM, value = 255
2019-01-05 14:07:26.931 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 7: Alarm converter processing NOTIFICATION
2019-01-05 14:07:26.931 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 7: Alarm converter NOTIFICATION event is 2, type DecimalType
2019-01-05 14:07:26.931 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Updating channel state zwave:device:145bfeec:node7:notification_access_control to 2 [DecimalType]
2019-01-05 14:07:26.931 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 7: Alarm converter processing NOTIFICATION
2019-01-05 14:07:26.932 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 7: Alarm converter NOTIFICATION event is 2, type OnOffType
2019-01-05 14:07:26.932 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 7: Alarm converter processing NOTIFICATION
2019-01-05 14:07:26.932 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 7: Alarm converter NOTIFICATION event is 2, type OnOffType
2019-01-05 14:07:26.932 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 7: Alarm converter processing NOTIFICATION
2019-01-05 14:07:26.933 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 7: Alarm converter NOTIFICATION event is 2, type StringType
2019-01-05 14:07:26.933 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Updating channel state zwave:device:145bfeec:node7:alarm_raw to {"notification":"ACCESS_CONTROL__MANUAL_UNLOCK","level":"0","type":"ACCESS_CONTROL","event":"2","status":"255"} [StringType]
2019-01-05 14:07:26.934 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Commands processed 1.
```

cdjackson, 2019-01-05:
Please use channel type alarm_entry

JesperWe, 2019-01-05:
Makes sense. One question though: I have not understood how DB channel names are mapped to the property type (as in `<property name="binding:*:OnOffType">`)

For this channel I would like the data type to be `DecimalType`, since it is getting different values depending on the cause (manual/key/code).

There are currently examples of `alarm_entry` with both `DecimalType` and `OnOffType` in various thing xml's. How do I control that from the DB?

cdjackson, 2019-01-05:
The mapping is done in the database exporter.

The value is either an ON or OFF, so other data is not really applicable - you can’t use a number to say the person X opened the door - you could say that the door was opened or closed, or you can say who by - not both. That is the typically the purpose of the raw channel where more information is made available.

JesperWe, 2019-01-05:
Ok, fair enough. Although that makes the channel not so useful for this device, it will basically always be ON. But since I can't delete it I have requested an update to `alarm_entry`. 

I guess you have to trigger the update from DB to XML's in this repo? In that case I will close this issue once the updates I requested have propagated.

cdjackson, 2019-01-05:
Really - it doesn’t report unlock state as well as lock state (as is defined by the protocol)? That would be a little strange, but if the device isn’t sending the states, then yes, it would make it a little useless!

JesperWe, 2019-01-05:
Sorry, you are correct, I was mixing up different parts of the event. alarm_entry does follow the lock status, it's just the cause that gets lost and has to be fetched from alarm_raw.

cdjackson, 2019-01-05:
Yes, although it doesn’t exactly “get lost” - an item state can only carry one piece of information, so it’s not able to pass both the state of the lock, and information such as who opened the lock.

JesperWe, 2019-01-07:
Fixed in #1110.

closed 2019-01-07 by JesperWe