openhab/org.openhab.binding.zwave #1111: ZD2102-5 Door Sensor: Update Items Partially

Bassysta, 2019-01-06:
Hi. 
Recently i had to reset my Z-Wave UZB and re-add all the nodes. Before the reset my ZD2102-5 Work perfectly, but now it update the sensor items only when door is closed, same for the tamper item.

OH Detect it as ZD2102-5:
https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/713

But i've bought it on amazon as ZHS09:
https://gzhls.at/blob/ldb/1/b/3/4/6ae52d05e4b0a0b34d21a6bf6eab44537d5f.pdf

Here the Debug Log when i open/close the contact:

019-01-06 13:31:11.175 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 9: Application Command Request (ALIVE:DONE)

2019-01-06 13:31:11.178 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 9: Starting initialisation from DONE

2019-01-06 13:31:11.186 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 9: Incoming command class BASIC

2019-01-06 13:31:11.189 [TRACE] [ssage.ApplicationCommandMessageClass] - NODE 9: Found Command Class BASIC, passing to handleApplicationCommandRequest

2019-01-06 13:31:11.193 [DEBUG] [.commandclass.ZWaveBasicCommandClass] - NODE 9: Received Basic Request

2019-01-06 13:31:11.199 [DEBUG] [.commandclass.ZWaveBasicCommandClass] - NODE 9: Basic Set sent to the controller will be processed as Basic Report

2019-01-06 13:31:11.203 [DEBUG] [.commandclass.ZWaveBasicCommandClass] - NODE 9: Basic report, value = 0x00

2019-01-06 13:31:11.209 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Got an event from Z-Wave network: ZWaveCommandClassValueEvent

2019-01-06 13:31:11.213 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Got a value event from Z-Wave network, endpoint = 0, command class = BASIC, value = 0

2019-01-06 13:31:11.239 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 9: Application Command Request (ALIVE:DONE)

2019-01-06 13:31:11.241 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 9: Starting initialisation from DONE

2019-01-06 13:31:11.245 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 9: Incoming command class ALARM

2019-01-06 13:31:11.247 [TRACE] [ssage.ApplicationCommandMessageClass] - NODE 9: Found Command Class ALARM, passing to handleApplicationCommandRequest

2019-01-06 13:31:11.249 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 9: Received ALARM command V4

2019-01-06 13:31:11.251 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 9: Process NOTIFICATION_REPORT V4

2019-01-06 13:31:11.252 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 9: NOTIFICATION report - 6 = 0, event=23, status=255

2019-01-06 13:31:11.255 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 9: Alarm Type = ACCESS_CONTROL (6)

2019-01-06 13:31:11.260 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Got an event from Z-Wave network: ZWaveAlarmValueEvent

2019-01-06 13:31:11.263 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Got a value event from Z-Wave network, endpoint = 0, command class = ALARM, value = 255

2019-01-06 13:31:11.276 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Updating channel state zwave:device:f8d6bc6b:node9:sensor_door to CLOSED [OpenClosedType]

2019-01-06 13:31:12.602 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 9: Application Command Request (ALIVE:DONE)

2019-01-06 13:31:12.606 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 9: Starting initialisation from DONE

2019-01-06 13:31:12.615 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 9: Incoming command class BASIC

2019-01-06 13:31:12.621 [TRACE] [ssage.ApplicationCommandMessageClass] - NODE 9: Found Command Class BASIC, passing to handleApplicationCommandRequest

2019-01-06 13:31:12.625 [DEBUG] [.commandclass.ZWaveBasicCommandClass] - NODE 9: Received Basic Request

2019-01-06 13:31:12.629 [DEBUG] [.commandclass.ZWaveBasicCommandClass] - NODE 9: Basic Set sent to the controller will be processed as Basic Report

2019-01-06 13:31:12.633 [DEBUG] [.commandclass.ZWaveBasicCommandClass] - NODE 9: Basic report, value = 0xFF

2019-01-06 13:31:12.642 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Got an event from Z-Wave network: ZWaveCommandClassValueEvent

2019-01-06 13:31:12.644 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Got a value event from Z-Wave network, endpoint = 0, command class = BASIC, value = 255

2019-01-06 13:31:12.671 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 9: Application Command Request (ALIVE:DONE)

2019-01-06 13:31:12.673 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 9: Starting initialisation from DONE

2019-01-06 13:31:12.678 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 9: Incoming command class ALARM

2019-01-06 13:31:12.680 [TRACE] [ssage.ApplicationCommandMessageClass] - NODE 9: Found Command Class ALARM, passing to handleApplicationCommandRequest

2019-01-06 13:31:12.682 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 9: Received ALARM command V4

2019-01-06 13:31:12.684 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 9: Process NOTIFICATION_REPORT V4

2019-01-06 13:31:12.687 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 9: NOTIFICATION report - 6 = 255, event=22, status=255

2019-01-06 13:31:12.689 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 9: Alarm Type = ACCESS_CONTROL (6)

2019-01-06 13:31:12.699 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Got an event from Z-Wave network: ZWaveAlarmValueEvent

2019-01-06 13:31:12.701 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Got a value event from Z-Wave network, endpoint = 0, command class = ALARM, value = 255

---------------------------------------------------------
I've tried to remove,reset and re-add the sensor many times, but it still not work correctly

Hope you can help me.
Have a nice day.

cdjackson, 2019-01-06:
It is probably better to discuss this on the forum. Please also provide information such as what version of the binding you are using as this is quite important.

Bassysta, 2019-01-06:
I've the 2.3 version. Ok i'll try into the forum, but can you just tell me how can update the binding if i've not the snapshot version?

Tnk!

cdjackson, 2019-01-06:
Ok, 2.3 is very old so I would definitely recommend using the latest - at least use 2.4, or if you have problems, then try the snapshot.

closed 2019-01-06 by cdjackson