openhab/org.openhab.binding.zwave #1124: Z-Wave Security S2 implementation

pizerg, 2019-02-01:
Since the number of products supporting S2 + Smart Start ara increasing in the market and more are being announced, I was wondering if there are any plans to implement those features on the binding. 

If so, I'd be more than happy to assist either with coding or with testing as I have access to some of the new devices.

cdjackson, 2019-02-10:
@dbadia was working on S2 - I'm not sure of the status.

I know that in future, things may have to change quite significantly, so I'm not sure working on these features is wise right now until things become clearer.

dbadia, 2019-02-12:
Just posted an update on the OH community site about this work:
https://community.openhab.org/t/z-wave-s2-security-work/52060

I will certainly need help testing once it's ready

theyoz, 2021-01-20:
Hi,
It sounds like S2 is not yet implemented on OpenHab. If not, is there any plans?

dbadia, 2021-11-20:
@dw-8  thanks for the tip!  I will have a look