openhab/org.openhab.binding.zwave #1128: OpenHab Rule not Updating CT-100 Thermostat Setpoint

martyphelan, 2019-02-09:
While the thermostat setpoint can be successfully controlled using the PaperUI, using a rule does not work. When the rule is fired, only the thing state and PaperUI seems to be updated. 
The new setpoint shows up in the PaperUI, but does not get sent to the CT-100 thermostat.

The DEBUG level logging for Z-Wave was used to observe the flow of Z-Wave activity.

Environment
-----------
Raspberry Pi 3 using OpenHabian Stable 2.4.0
Aeotec Z-Stick Gen5, Z-Wave Plus USB
CT-100 Thermostat Z-Wave

=================================
TEST RULE - FAILED
=================================
Expect rule to fire and update thing, PaperUI and CT-100 thermostat

Initial State
-------------
PaperUI shows Setpoint 21
CT-100 Thermostat setpoint/target temp at 70°F (21°C)

Rule
----
rule "Test Rule 1"
when
        Time cron "0 21 14 ? * MON-SUN"
then
        ThermostatMasterBedroom_SetpointHeating.postUpdate(60|°F)
end

Log when rule fired:
--------------------
2019-02-08 14:20:12.459 [INFO ] [el.core.internal.ModelRepositoryImpl] - Refreshing model 'thermostat.rules'
2019-02-08 14:21:00.457 [vent.ItemStateChangedEvent] - ThermostatMasterBedroom_SetpointHeating changed from 21.1111111111111111111111111111111 °C to 15.5555555555555555555555555555556 °C
(nothing else follows)

State after rule fired:
-----------------------
PaperUI shows Setpoint 16
CT-100 Thermostat setpoint/target temp at 70°F (21°C)
FAILED

=================================
TEST THERMOSTAT - SUCCESS
=================================
Expect changing setpoint on CT-100 Thermostat to update thing and PaperUI

Initial State
-------------
PaperUI shows Setpoint 16
CT-100 Thermostat setpoint/target temp at 70°F (21°C)

Action
------
Change setpoint/target temp at CT-100 Thermostat to 68°F

Log after action
----------------
2019-02-08 14:41:50.785 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0B 00 04 08 02 05 43 03 01 09 44 F3 
2019-02-08 14:41:50.798 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=2, callback=8, payload=08 02 05 43 03 01 09 44 
2019-02-08 14:41:50.806 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=2, callback=8, payload=08 02 05 43 03 01 09 44 
2019-02-08 14:41:50.808 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0B 00 04 00 02 05 43 03 01 09 44 FB 
2019-02-08 14:41:50.812 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction null
2019-02-08 14:41:50.816 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 2: Application Command Request (ALIVE:DONE)
2019-02-08 14:41:50.818 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=2, callback=0, payload=00 02 05 43 03 01 09 44 
2019-02-08 14:41:50.821 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 2: resetResendCount initComplete=true isDead=false
2019-02-08 14:41:50.825 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 2: Incoming command class COMMAND_CLASS_THERMOSTAT_SETPOINT, endpoint 0
2019-02-08 14:41:50.830 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 2: SECURITY not supported
2019-02-08 14:41:50.834 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 2: Received COMMAND_CLASS_THERMOSTAT_SETPOINT V1 THERMOSTAT_SETPOINT_REPORT
2019-02-08 14:41:50.839 [DEBUG] [.ZWaveThermostatSetpointCommandClass] - NODE 2: Thermostat Setpoint report Scale = 1
[... 54 lines cut ...]
2019-02-08 15:03:44.922 [ome.event.ItemCommandEvent] - Item 'ThermostatMasterBedroom_SetpointHeating' received command 18 °C
==> /var/log/openhab2/openhab.log <==
2019-02-08 15:03:44.933 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Command received zwave:device:770c2841:node2:thermostat_setpoint_heating --> 18 °C [QuantityType]
2019-02-08 15:03:44.946 [DEBUG] [ter.ZWaveThermostatSetpointConverter] - NODE 2: Thermostat command received for 18 °C
==> /var/log/openhab2/events.log <==
2019-02-08 15:03:44.946 [nt.ItemStatePredictedEvent] - ThermostatMasterBedroom_SetpointHeating predicted to become 18 °C
==> /var/log/openhab2/openhab.log <==
2019-02-08 15:03:44.950 [DEBUG] [.ZWaveThermostatSetpointCommandClass] - NODE 2: Creating new message for command THERMOSTAT_SETPOINT_SET
2019-02-08 15:03:44.953 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 2: SECURITY not supported
==> /var/log/openhab2/events.log <==
2019-02-08 15:03:44.956 [vent.ItemStateChangedEvent] - ThermostatMasterBedroom_SetpointHeating changed from 20.00 °C to 18 °C
==> /var/log/openhab2/openhab.log <==
2019-02-08 15:03:44.957 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 2: Command Class COMMAND_CLASS_THERMOSTAT_SETPOINT is NOT required to be secured
2019-02-08 15:03:44.959 [DEBUG] [ter.ZWaveThermostatSetpointConverter] - NODE 2: Sending Message: org.openhab.binding.zwave.internal.protocol.transaction.ZWaveCommandClassTransactionPayload@641c1f
2019-02-08 15:03:44.961 [DEBUG] [.ZWaveThermostatSetpointCommandClass] - NODE 2: Creating new message for application command THERMOSTAT_SETPOINT_GET (Heating)
2019-02-08 15:03:44.964 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 2: SECURITY not supported
2019-02-08 15:03:44.966 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 2: Command Class COMMAND_CLASS_THERMOSTAT_SETPOINT is NOT required to be secured
2019-02-08 15:03:44.968 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 2: Adding to device queue
... (remainder omitted for brevity)

State after rule fired:
-----------------------
PaperUI shows Setpoint 18
CT-100 Thermostat setpoint/target temp at 64°F (18°C)
SUCCESS

cdjackson, 2019-02-09:
You might be better off to discuss this on the forum. If the command works ok when using the UI, but doesn’t work with a rule, then it is not a problem with the binding as the binding knows nothing about rules - just commands. Probably there is a problem with the rule and you will be better to discuss this on the forum.

martyphelan, 2019-02-09:
Thanks - Problem solved. As you indicated, the rule was the problem.
The corrected rule is:
rule "Test Rule 1"
when
Time cron "0 21 14 ? * MON-SUN"
then
ThermostatMasterBedroom_SetpointHeating.sendCommand(60|°F)
end

I have been searching for days without luck. Today, I was in the process of posting when I found a close example and discovered my mistake. I was using postUpdate when it should have been sendCommand. The postUpdate did change the thing state but does not trigger sending a Z-Wave message.

Thanks,
Marty

closed 2019-02-09 by martyphelan