openhab/org.openhab.binding.zwave #1137: Shock sensor alarms not triggered

ncister, 2019-02-18:
Hi.
I've without problems configured ZS5101 Shock Sensor (https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/442) using a Z-Wave USB stick (https://z-wave.me/uzb) installed on OH2 (v. 2.4.0-1) binding-zwave v. 2.4.0.
The "_Tamper alarm_" events works fine (are correctly triggered and catched by OpenHAB event log ...) while "_Shock Alarm_" events **are ignored** ! ....
I've already tried to set the sensibility ... without results ... 
Have you some suggestion on OpenHAB integration of this device ?
Thanks.
Nicola Cisternino.

```
openhab> smarthome:things show zwave:device:fa62667f:node2
UID: zwave:device:fa62667f:node2
Type: zwave:vision_zs5101_00_000
Label: Z-Wave Shock Sensor 001
Status: ONLINE: Node initialising: REQUEST_NIF
Bridge: zwave:serial_zstick:fa62667f

Properties:
	zwave_class_basic : BASIC_TYPE_ROUTING_SLAVE
	zwave_class_generic : GENERIC_TYPE_SENSOR_NOTIFICATION
	zwave_frequent : false
	zwave_neighbours : 
	zwave_lastwakeup : 2019-02-18T07:25:15Z
	modelId : ZS5101
	zwave_version : 16.6
	zwave_listening : false
	zwave_plus_devicetype : NODE_TYPE_ZWAVEPLUS_NODE
	manufacturerId : 0109
	manufacturerRef : 2003:0306,2003:0307
	dbReference : 442
	zwave_deviceid : 775
	zwave_nodeid : 2
	vendor : Vision Security
	defaultAssociations : 1
	zwave_routing : true
	zwave_plus_roletype : ROLE_TYPE_SLAVE_SLEEPING_REPORTING
	zwave_beaming : true
	zwave_secure : false
	zwave_class_specific : SPECIFIC_TYPE_NOTIFICATION_SENSOR
	zwave_manufacturer : 265
	zwave_devicetype : 8195

Configuration parameters:
	wakeup_node : 1
	wakeup_interval : 21750
	group_1 : [node_1]
	node_id : 2

Channels:
	ID: sensor_binary
	Label: Binary Sensor
	Type: zwave:sensor_binary
	Description: Indicates if a sensor has triggered

	ID: alarm_access
	Label: Shock Alarm
	Type: zwave:alarm_access
	Description: Indicates if the access control alarm is triggered

	ID: alarm_tamper
	Label: Tamper Alarm
	Type: zwave:alarm_tamper
	Description: Indicates if the tamper alarm is triggered

	ID: battery-level
	Label: Battery Level
	Type: system:battery-level
```

Dopeyr, 2019-02-18:
Hi, I don't think your node is initialised yet:

> Status: ONLINE: Node initialising: REQUEST_NIF

Try to keep on waking it until it's complete.

ncister, 2019-02-18:
Hi Dopeyr, and thank you for replay ....
> Try to keep on waking it until it's complete.

You are right ! ... it seems that node is not completely initialized ...
How can keep on waking it ?

Dopeyr, 2019-02-18:
Hi, it should keep waking by itself every few hours, or you can speed it up by waking it manually. Manual wake up is device specific, so it should be in the manual. The link in your first post says:

>”Awake” mode: it is to leave the “Sleep” mode by removing the cover of ZD5101,
to allow the Z-Wave™ Interface Controller to do “Inclusion”, “Exclusion”,
“Association” and to reply and receive the commands from controller

So I *think* that means just leave the cover off? 

You may want to [enable debug logging](https://www.cd-jackson.com/index.php/openhab/5-zwave-debugging-openhab) to see if it's progressing.

ncister, 2019-02-19:
Hi ...
Now Status is "ONLINE" ... but when I trigger tamper the log print:
`Thing 'zwave:device:fa62667f:node2' has been updated. ` 
(.... not ON / OFF)
... while when i "stress" the sensor the log print:
```
2019-02-19 08:21:20.410 [vent.ItemStateChangedEvent] - zwave_serial_zstick_fa62667f_serial_ack changed from 55 to 56
2019-02-19 08:21:20.435 [vent.ItemStateChangedEvent] - zwave_serial_zstick_fa62667f_serial_sof changed from 193 to 194
2019-02-19 08:21:20.439 [vent.ItemStateChangedEvent] - zwave_serial_zstick_fa62667f_serial_sof changed from 194 to 195
```
that is related to usb z-wave stick activity ....

cdjackson, 2019-02-19:
It would be better to discuss this on the community forum where you will have a wider support audience. This list is normally for bugs rather than support issues. 

You should also provide a debug log so that we can see what is happening (As suggested above).

ncister, 2019-02-20:
Sorry, Chris ... :-(
... but when I've contacted the dealer, he replied:

> Hello,
> It sounds like there is a compatibility issue between OpenHAB and this particular product after v2.3
> According to this post, it was working for this user with 2.3 and after his update to 2.4 his vibration sensor started having the same problem you are reporting. I assume this is something OpenHAB needs to fix but perhaps the manufacturer can help you with this or they can help OpenHAB fix the problem. There is not much that I, as a seller, can do about this problem unfortunately. 
> You can reach the manufacturer for tech support via their website:
> 
> http://www.visionsecurity.com.tw/index.php?option=contact&lang=en
> 
> Kind regards,
> Anthony Hulsman - EliteConsole

So, I will try the sensor with OH2.3 and will provide a debug log ;-)
Thanks.

cdjackson, 2019-02-20:
Probably there is no problem, but it's hard to say without a debug log, and we can then tune the database to resolve this.

Please provide a debug log from 2.4, and please let's discuss on the forum where there are more people to help.

cdjackson, 2019-08-11:
Closing this as no information was provided. If the problem persists, please reopen and provide the requested information.

closed 2019-08-11 by cdjackson