openhab/org.openhab.binding.zwave #1167: Invalid electric meter scale warning

eshabi, 2019-05-06:
Using Qubino's 3-Phase Smart Meter to Periodically send its readings, a warning about invalid scale meter 7 is shown with each report. 
Seems like the z-wave binding does not support the extension of scales using code 7 (introduced with version 4 of command class).
The [following document](https://www.silabs.com/documents/login/miscellaneous/SDS13781-Z-Wave-Application-Command-Class-Specification.pdf) shows the scaling options starting at page 99. 
This specific device reports among other values the Reactive Power, instant and accumulated, kVar & kVarh accordingly. It uses scale code 7 and the value of "scale 2" to differentiate between the two.

* OH2 Version: 2.5.0.M1
* Binding Version: 2.5.0.M1
* Device name: ZMNHXD
* Device version: 
* Link to [database](https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/900)

[openhab.log](https://github.com/openhab/org.openhab.binding.zwave/files/3149057/openhab.log)
[network_ffe08d7b__node_7.zip](https://github.com/openhab/org.openhab.binding.zwave/files/3149063/network_ffe08d7b__node_7.zip)

cdjackson, 2019-05-26:
Please provide a log during the initialisation of the device. Delete the XML and restart, or click the reinitialisation option.

eshabi, 2019-05-27:
Attached is the log after using the 'Reinitialise' option.

[openhab.log](https://github.com/openhab/org.openhab.binding.zwave/files/3222356/openhab.log)

PR #1179 (2019-06-01): Update for Meter CC V3
Closes #1167
closed 2019-06-01 by cdjackson