openhab/org.openhab.binding.zwave #1173: Report Z-Wave 0xFE undefined state as OH2 'UNDEF' state

shutterfreak, 2019-05-23:
* **OH2 Version:** 2.5M1 (and any earlier)
* **Binding Version** (if SNAPSHOT, provide compile time as found on the filename in the console): 2.5.0.201905020404 snapshot (and any earlier)

The Z-Wave binding currently doesn't report 'undefined' 0xFE state to OH2. As a result, OH2 has incomplete state information on some Z-Wave devices that do report 0xFE to the binding. See: [Z-Wave binding_cmdrepollperiod cannot exceed 15 seconds](https://community.openhab.org/t/z-wave-binding-cmdrepollperiod-cannot-exceed-15-seconds/72599/17?u=shutterfreak)

Example: a Z-Wave roller shutter may be endpoint aware but not position aware. In this case, the Z-Wave state will be either 0x00 or 0x0F if an endpoint has been reached (fully open / closed), or 0xFE for any other position. Note that according to the Z-Wave specification, this state can also be reported in case the state is unknown to the device (e.g., after a power cut).

**Proposed solution:** have the Z-Wave binding report Z-Wave state 0xFE to OH2 as 'UNDEF' state.

cdjackson, 2019-05-23:
What channel is this for?

shutterfreak, 2019-05-23:
Command class = `SWITCH__MULTILEVEL`
Channel = `blinds_control` (linked to `SWITCH_MULTILEVEL`)
Device = https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/1036

`MULTILEVEL_GET` will return `0x00` of `0xFF` when an endpoint has been reached, and returns `0xFE` in case the shutter is partially opened. As far as I know, Z-Wave state `0x00` is translated to OH2 state `0` and Z-Wave state `0xFF` is translated to OH2 state `100`.

Z-Wave state `0xFE` should be translated as OH2 state `UNDEF`.

shutterfreak, 2019-05-27:
Is it possible that OH2 snapshot 2.5.0~S1601-1 already implements this? I saw an "UNDEF" state followed with the final state on my node-red overview (used for supporting Apple HomeKit on non-HomeKit devices).

cdjackson, 2019-05-27:
Yes, this was implemented recently.

shutterfreak, 2019-05-27:
Ok, issue closed!

closed 2019-05-27 by shutterfreak