openhab/org.openhab.binding.zwave #1177: Inclusion on newer 2.5 Builds at inclusion

RobertHerter, 2019-05-28:
OH2 Version: 2.5 Build 1601

With newer builds i got the following Error on inclusion, after that i have to restart the Z-Wave Controller thing, inclusion is not finished but found as Unkown device in the Inbox:
`28-Mai-2019 15:42:03.155 [ERROR] [wave.handler.ZWaveSerialHandler$ZWaveReceiveThread] - Exception during ZWave thread.
java.lang.IllegalStateException: Queue full
        at java.util.AbstractQueue.add(AbstractQueue.java:98) ~[?:?]
        at java.util.concurrent.ArrayBlockingQueue.add(ArrayBlockingQueue.java:312) ~[?:?]
        at org.openhab.binding.zwave.internal.protocol.ZWaveTransactionManager.processReceiveMessage(ZWaveTransactionManager.java:404) ~[355:org.openhab.binding.zwave:2.5.0.201905260928]
        at org.openhab.binding.zwave.internal.protocol.ZWaveController.incomingPacket(ZWaveController.java:1064) ~[355:org.openhab.binding.zwave:2.5.0.201905260928]
        at org.openhab.binding.zwave.handler.ZWaveControllerHandler.incomingMessage(ZWaveControllerHandler.java:643) ~[355:org.openhab.binding.zwave:2.5.0.201905260928]
        at org.openhab.binding.zwave.handler.ZWaveSerialHandler$ZWaveReceiveThread.run(ZWaveSerialHandler.java:323) [355:org.openhab.binding.zwave:2.5.0.201905260928]`

I'll attach the log from starting Inclusion to S
[t.txt](https://github.com/openhab/org.openhab.binding.zwave/files/3228146/t.txt)
topping

cdjackson, 2019-05-28:
The queue full error is a symptom of other problems in your system and is not related to inclusion. I'm not sure what is happening, but the queue is filling up and not being processed - possibly the controller needs resetting.

RobertHerter, 2019-05-28:
Ok,
it only occurs when including, every time. Else it's running fine for days with 80+ devices.
Do you mean Hard-Reset ? Softresetting is not solving the Problem nor Firmware upbor downgrade.

JohanHectors, 2019-06-18:
>  Ok,
> it only occurs when including, every time. Else it's running fine for days with 80+ devices.
> Do you mean Hard-Reset ? Softresetting is not solving the Problem nor Firmware upbor downgrade.

Hi,

https://community.openhab.org/t/error-message-in-console/76384/20

I might have the exact same problem, I start incluson true the binding, the new thing does not show up until a restart of the binding or the openhab2 service.
Also, there seems to be a LOT lag on the z wave network after the start of the inclusion, leaving it unusable for switching of/on lights ed..

Setu:
Openhabian last snapshot 
Rpi 3+
Aeon gen 5 z wave stick
about 30 z wave nodes running in network
z wave binding, homekit binding,  tinkerforge binding, and openvpn, running atm.

regards,
Johan

cdjackson, 2019-07-05:
I believe that this issue is resolved with #1192

closed 2019-07-05 by cdjackson