openhab/org.openhab.binding.zwave #1196: [enhancement] Reduce retries during STATIC_VALUES stage of initialization

mhilbush, 2019-07-07:
Reference forum discussion [here](https://community.openhab.org/t/issue-including-aeon-zw099-dimmer/77558?u=mhilbush).

During the STATIC_VALUES stage of initialization, when processing command classes for endpoints 0..n, if a command class is included in the nodes capability report, but is not supported by the device, the binding will request information for that command class 9 times before giving up. Because of the backoff timer, these 9 attempts can span 10-15 minutes, greatly lengthening the time to initialize the device. In some cases, if there are multiple unsupported command classes, device initialization for a mains-powered device can take over an hour.

While a deficiency in the device is the root cause of this issue, the user experience could be improved if the number of retries is reduced.

cdjackson, 2019-07-13:
I've reduced this from 10 retries to 5 retries. I don't want to reduce it too far or we'll have a different set of issues.

mhilbush, 2019-07-13:
That likely will make a big difference, as it should avoid the really big backoff times. Hopefully, it won't have any negative side-effects.