openhab/org.openhab.binding.zwave #1198: Zwave controller reported the last state of the sensor

thomas-tri-tran, 2019-07-10:
* OH2 Version: 2.4
* Device name: Zwave Stick Gen 5

Whenever the Openhab is restarted, the Zwave controller reported the last stage of the sensor e.g. The current door sensor is close, however, after the OpenHab is restarted, it reported the previous state of the door sensor which is Open.

Skrew77, 2019-07-10:
Is that really ZWave stick reporting that wrong states? Or you use the database to backup the items state and it got restored after the restart ? Can you provide the log ?

cdjackson, 2019-07-10:
In such instances it is not possible for the ZWave binding to know the state of the door sensor. A door sensor is a battery device, and it will sleep, so the binding can not wake it up to read the state. It can only read the state when it next wakes, or if the state changes the device will automatically send an update.

Otherwise, the item state is restored from a database, and in this case if it changes while the binding is down, there is no way to resolve this.

closed 2019-07-10 by cdjackson