openhab/org.openhab.binding.zwave #1202: Attempting to add listener when controller is null

lolodomo, 2019-07-18:
At first startup of the snapshot 1641, I got this error:
```

13:30:10.178 [ERROR] [.zwave.handler.ZWaveControllerHandler] - Attempting to add listener when controller is null
13:30:10.256 [INFO ] [ome.event.ThingStatusInfoChangedEvent] - 'zwave:serial_zstick:controller' changed from UNINITIALIZED to INITIALIZING
13:30:10.339 [INFO ] [smarthome.event.ThingUpdatedEvent    ] - Thing 'zwave:serial_zstick:controller' has been updated.
13:30:10.350 [INFO ] [ome.event.ThingStatusInfoChangedEvent] - 'zwave:serial_zstick:controller' changed from INITIALIZING to OFFLINE (BRIDGE_OFFLINE): Controller is offline
13:30:10.356 [INFO ] [ding.zwave.handler.ZWaveSerialHandler] - Connecting to serial port '/dev/ttyACM0'
13:30:10.382 [INFO ] [ding.zwave.handler.ZWaveSerialHandler] - Serial port is initialized
13:30:10.495 [INFO ] [ave.internal.protocol.ZWaveController] - Starting ZWave controller
13:30:10.499 [INFO ] [ave.internal.protocol.ZWaveController] - ZWave timeout is set to 5000ms. Soft reset is false.
```
By the way, the binding is working.

lolodomo, 2019-07-29:
This one is due to the discovery service that is activated just after the creation of the bridge handler but before this handler is fully initialized.
https://github.com/openhab/org.openhab.binding.zwave/blob/master/src/main/java/org/openhab/binding/zwave/discovery/ZWaveDiscoveryService.java#L62
@cdjackson : I let you confirm but I don"t think this is a problem and I will suggest to only suppress this log message here:
https://github.com/openhab/org.openhab.binding.zwave/blob/master/src/main/java/org/openhab/binding/zwave/handler/ZWaveControllerHandler.java#L455
You already have a WARN message in case addEventListener returns false here:
https://github.com/openhab/org.openhab.binding.zwave/blob/master/src/main/java/org/openhab/binding/zwave/handler/ZWaveThingHandler.java#L532
The only question I have is whether the discovery service will be notified correctly in this case ?

lolodomo, 2019-07-29:
I have the feeling that you are (re)doing the stuff at the end of the network initialization here:
https://github.com/openhab/org.openhab.binding.zwave/blob/master/src/main/java/org/openhab/binding/zwave/handler/ZWaveControllerHandler.java#L189
So this should be ok ?

lsiepel, 2021-03-22:
I can still see this in the eventlog @ 3.1.0.M2

lsiepel, 2023-08-04:
Still available in 4.0, it is very minor, but anoying at the same time. I'll create a PR.

PR #1881 (2023-08-05): Fix Attempting to add listener when controller is null #1202
Fixes #1202 

Looks like the handler is not yet fully initialized and logs this message. As the binding works as expected, changing the log level from info to debug is better.
closed 2023-08-05 by cdjackson