openhab/org.openhab.binding.zwave #1203: Serial Error: Port {0} does not exist

lolodomo, 2019-07-18:
When I restart OH, the thing status is OFFLINE - COMMUNICATION_ERROR Serial Error: Port {0} does not exist
If I restart the Zwave bundle, everything is working again.

This problem appeared in recent snapshots.

cdjackson, 2019-07-20:
I don't think this is a binding issue - again, this is likely to be a problem with the underlying serial driver in OH (ie the ESH serial abstraction). I seem to recall in the past there was some discussion about bindings being able to start before the serial abstraction system had loaded and you simply need to wait until the startup is complete.

I propose to close this and open an issue in the OH core.

lolodomo, 2019-07-22:
I see that your handler factory is declared with "immediate = true". Could it be the source of the problem ?
I opened an issue to get a clear status about that: https://github.com/openhab/openhab2-addons/issues/5852

lolodomo, 2019-07-22:
Except this one, I don't see other significant difference with my bindings using the serial abstraction.

cdjackson, 2019-07-22:
I don't think changing that would ensure that the serial driver is loaded before the handler is started. The HandlerFactory is not dependant on the serial driver - as far as I know it's completely separate.

That said, I'm not sure of the impact of changing it - maybe it helps by adding in a small additional delay (?) but if so I don't believe it would eliminate the issue.

Probably what is needed is to ensure that all bundles with system requirements such as the serial handlers are run with a higher start priority (lower number I think) than other bundles?

lolodomo, 2019-07-22:
I see no trace of "immediate = true" in the serial stuff itself.
Regarding the start level, I assume it has to be set in the feature file but I don't know on which bundle exactly.

cdjackson, 2019-07-22:
I'm afraid I'm not really an OSGi expert and these sort of policies probably should be defined at "system" level, but I'm not sure how this works. As you are no doubt aware, the startup of OH has always been a problem.

If there's something to change here, then I'm happy if someone can advise what to change. Personally I think that somehow, the startup of the bindings should be done after services such as serial drivers has completed, but I don't know how/where to ensure that this is the case.

cdjackson, 2019-07-29:
@lolodomo I think we can close this now with your changes - right?

lolodomo, 2019-07-29:
No, my change has only partially fixed the problem in the Z-Wave binding. In my personal case, it should fix it fully but only because I am using a workaround to delay the startup of things/items/sitemaps/rules (I am waiting for a new snapshot to confirm it).
But the problem is not yet fully resolved and could still occur for general users. We have a pending PR in the core framework that should fully fix it for any binding using the serial abstraction.
The problem is now no more in the Z-Wave binding so either we close the issue now or we wait for the framework fix.

cdjackson, 2019-07-29:
IMHO there's no need to keep an issue open here if there is nothing to do here. When the issue is being resolved in another library, we don't keep issues open, so I'm tempted to close.

lolodomo, 2019-07-29:
The problem is not yet fixed but this is now a framework bug that can be followed in this other issue:
https://github.com/openhab/openhab-core/issues/937

closed 2019-07-29 by lolodomo

lolodomo, 2019-07-30:
Using the snapshot 1650, I got no such error after 3 startup of OH.