openhab/org.openhab.binding.zwave #1212: Add RollerShutter capability for FGS222

robin-thoni, 2019-07-28:
* Device name: FGS222
* Device version: v2.2
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/124
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it: https://github.com/openhab/org.openhab.binding.zwave/blob/e964e9a5e0b0423fca53dfe99bb640c74cf5bc05/src/main/resources/ESH-INF/thing/fibaro/fgs222_0_0.xml

The FGS222 is a simple 2 relays device. Its main purpose was probably to control two lights, but it can also be used to control low voltage shutters. I have 8 shutters that works on 24v, making it impossible to use a FGR222 (without any external relays, which would take too much space).

I can use the FGS222 with 3 items and a rule:
[image]

```
rule "LivingRoomShutterRule"
when
    Item LivingRoomShutter received command DOWN or
    Item LivingRoomShutter received command UP or
    Item LivingRoomShutter received command STOP
then
    if (receivedCommand == DOWN)
    {
       LivingRoomShutter_SwitchUp.sendCommand(OFF)
       LivingRoomShutter_SwitchDown.sendCommand(ON)
    }
    else if (receivedCommand == UP)
    {
       LivingRoomShutter_SwitchDown.sendCommand(OFF)
       LivingRoomShutter_SwitchUp.sendCommand(ON)
    }
    else if (receivedCommand == STOP)
    {
       LivingRoomShutter_SwitchDown.sendCommand(OFF)
       LivingRoomShutter_SwitchUp.sendCommand(OFF)
    }
end
```

Having a builtin support for shutters would be MUCH easier than creating 3 items and a rule...

Thanks

cdjackson, 2019-07-28:
Unfortunately I don't think this device is a roller shutter device is it? It's a switch. Therefore it's not really easy to support what you are asking for - it would require custom code for this device in the binding, and would mean that everyone with this device has a roller shutter showing up instead of (or maybe as well as) the switches.

I'm afraid that given you are doing something non-standard, the rules are the best way to solve this.

robin-thoni, 2019-07-29:
> Unfortunately I don't think this device is a roller shutter device is it? It's a switch.
True, it's a simple double switch.

> it would require custom code for this device in the binding
Ah. I don't really know implementation details, I was hopping it would be a simple interface to implement or something as easy.

> or maybe as well as
Yes, that was my idea to add the possibility to use it both as a switches and roller shutter. Both devices are just two relays, so it would make sense to allow to use both of them as simple switches AND roller shutters. What do you think?

> I'm afraid that given you are doing something non-standard, the rules are the best way to solve this.
Yeah, I was afraid of that :/

robin-thoni, 2019-08-02:
Closing, as it is not a supported use-case

closed 2019-08-02 by robin-thoni