openhab/org.openhab.binding.zwave #1213: Add serial port reconnection logic

wborn, 2019-07-29:
The binding would be more reliable and easier to use when it has its own serial port reconnection logic (see https://github.com/openhab/openhab-core/issues/937#issuecomment-513843563).

The binding would be able to recover from issues such as:
* starting openHAB without the Z-Wave USB stick inserted and inserting it at a later time
* serial port permission issues that occur and are fixed while using the binding
* unplugging/reinserting the Z-Wave USB stick while using the binding
* other bundles providing serial comm dependencies going unavailable because of them being restarted/updated while using the binding

cdjackson, 2019-07-29:
While I agree, the slightly annoying thing here is that this has only become an issue since we changed to use the ESH serial abstraction :(.

Anyway, I will take a look at this in the near future - it requires some rearchitecting of the binding.

wborn, 2019-07-29:
Great! The workaround is now available in build 1650 (and newer) so that should buy us all some time. Looks like my issue is similar to #979 (predating the ESH serial abstraction).

bodiroga, 2019-07-29:
Hi guys!

@cdjackson, the code provided in https://github.com/openhab/org.openhab.binding.zwave/pull/980#issuecomment-444482366 did work correctly in the tests I made some months ago. As far as I know, with it all node handlers are properly shutdown and reinitialized correctly when the bridge is online again.

I will try to setup de IDE in the next days and made some tests, let's see if we can finally implement this feature in the binding :+1:

cdjackson, 2019-08-10:
I don't see how the code you linked can reinitialise the controller. Maybe there is more code other than the code you referenced, but this is only in the thing handlers.

The controller MUST be completely reinitialised. It may work under some circumstances and some use cases without this, but not all. One example that people talk about a lot is to remove the dongle, include a new device, or exclude a current device, and plug back in to the PC. These will simply not work unless the controller, and the things are reinitialised. In these cases the binding will still need to be restarted to account for the changes in the network, and that defeats the purpose.

I don't want to implement something that isn't done properly - past experience has shown that this results in more problems, and more of my time to resolve than doing it properly for starters.

I'm happy if you want to look at this - just let me know. Otherwise I will look at it shortly.

cdjackson, 2019-08-11:
Duplicate of #979

bodiroga, 2019-08-11:
Working on it @cdjackson, give me some hour to prepare a WIP pull request :wink:

bodiroga, 2019-08-12:
I thought that I had it, but no, this is harder than expected. Well, the network is shut down correctly and all node threads and transactions are stopped, but I have a 100% cpu load problem when the stick is unplugged :cry:  The problem seems to be related to the "serialEvent" sleep workaround, because the first time I unplug the stick everything works fine, but when the stick is plugged again, I don't see the "RXTX library CPU load workaround, sleep forever" message in the log file. In fact, I can reproduce the issue without starting the ZWaveReceive thread, commenting the receiveThread.start() method, so its completely unrelated to the ZWave network behavior.

And the IDE configuration issues don't help, I'm building the binding by hand through maven and moving the .jar file to openHAB's addon folder everytime I want to try something new :angry: 

Let's keep trying :+1:

PR #1222 (2020-01-04): Add serial port reconnection logic (Closes #1213)
Hi @cdjackson!

Let's see what you think about the implementation, fingers crossed :crossed_fingers:

Some parts of the code are taken from the great work done by @Hilbrand on #1210, so thanks Hilbrand for your job!

A couple of important notes:

- I have set the "notifyOnDataAvailable" flag to false because we are not using the serialEvent callback of the SerialPortEventListener interface to check for new data, as we are reading the serial port in a while loop. The "forever" sleep (```Thread.sleep(Long.MAX_VALUE);```) in serialEvent causes a 1 [... cut]
commit 81674b73 (2020-01-04): Add serial port reconnection logic (Closes #1213) (#1222)

Add serial port reconnection logic
closed 2020-01-04 by cdjackson