openhab/org.openhab.binding.zwave #1214: [enhancement] Command Channel

tmrobert8, 2019-07-31:
So I was thinking about [this thread](https://community.openhab.org/t/zwave-things-config-parameters/78472/19) and how, if you have manually specified the things/items that you can't set the 'command' configuration properties (not really sure if that's the right terminology).

As an alternative - would you be able to create a 'command' channel that would allow someone to send a 'key=value' string to the command channel and have it process it as a command.  Example: if, in the paperui, I enter 'config_27_1=1' - then would have the same effect as setting the config parameter for the thing in paperui.

Would be really nice because I could use some rules to change the configuration if needed on some zwave devices (or even do it from the karaf console).

Not sure if I'm making myself clear..

openhab-5iver, 2019-08-01:
Channels can be added through the device database for configuration parameters. However, a ThingAction that could be used for any device and any configuration parameter would be a nice addition.

lsiepel, 2019-08-05:
If i understand it right, your requirement is altering the Thing configuration from a rule. Because that is impossible, you suggest some kind of 'raw command'-channel for every device.
I'd prefer to have the configuration changable from a rule, but i think that is discussed before (and not related to z-wave but to core?) so this would be a good workaround!

My use case:
Let's say i have a LUX sensor, and i need it to be accurate around certain times (the morning and evening). To save battery usage, it would be cool to set it to update its state every minute when it's needed and set it to an hour or so when the accuracy is not needed. Not sure if the device's lifetime (memory wear?) will be shortened and how the wakeup stuff would work for a battery device like this.

cdjackson, 2019-08-05:
> If i understand it right, your requirement is altering the Thing configuration from a rule.

My personal preference for this sort of thing is to add this to the core so that it's available for all bindings since this is not really a ZWave issue (IMHO ;) ).

> it would be cool to set it to update its state every minute when it's needed and set it to an hour or so when the accuracy is not needed.

This is not really possible - at least not very well. ZWave battery devices only wake up on a set schedule (say every hour, or every day). This is set by the wakeup configuration and obviously this can only be changed at the next wakeup.

This is also not how ZWave devices should be operated - you should configure them to report on change so that you get timely updates when the sensor data changes by a value that is of interest. Changing the wakeup period will again not impact this at all, and if you are considering using polling, then this is a bad idea anyway.

lsiepel, 2019-08-05:
> My personal preference for this sort of thing is to add this to the core so that it's available for all bindings since this is not really a ZWave issue (IMHO ;) ).

100% agree. I know it was discussed before somewhere, but can't find a reference about a maintainer's point of view. 

I did find a (hacky)  workaround: https://community.openhab.org/t/thing-configuration-from-rules/41228/13 so maybe that is good enough to use until someone makes a decission to implement a native method in rules (or not).

About the use case. I know that it would also be dependent on the wake up, but i think it can work, maybe hacky but it 'll work. Let's not get distracted by my (not the very best) example. This issue should be about changing the configuration from rules or as workaround implement a generic command channel.

cdjackson, 2019-09-15:
There are moves to resolve this properly in the core, so I'm closing this here.

https://github.com/openhab/openhab-core/pull/583
https://github.com/openhab/openhab-core/pull/1004

closed 2019-09-15 by cdjackson