openhab/org.openhab.binding.zwave #1220: NEXA Motion sensor not updating channel on trigger

JC-85, 2019-08-10:
**Expected Behavior**
Thing channel should be updated to ON when motion sensor trips and change to OFF after set timer expires.

**Current Behavior**
OpenHAB receives a NOTIFICATION event 8 when sensor is triggered but does NOT set channel to ON. After timer lapses a event 0 is received which does update the channel to OFF.

The behavior seems to correspond to this issue https://github.com/openhab/org.openhab.binding.zwave/issues/1113.

**Additional info**

The branding of the device is Nexa, but OpenHAB identifies it as a Everspring SP816.
Link to database entry: https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/855

**Logfile**
```
2019-08-10 21:03:48.870 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=12, callback=0, payload=00 0C 09 71 05 00 00 00 FF 07 08 00
2019-08-10 21:03:48.881 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction null
2019-08-10 21:03:48.890 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 12: Application Command Request (ALIVE:DONE)
2019-08-10 21:03:48.900 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 12: resetResendCount initComplete=true isDead=false
2019-08-10 21:03:48.906 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 12: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
2019-08-10 21:03:48.915 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 12: SECURITY NOT required on COMMAND_CLASS_ALARM
2019-08-10 21:03:48.920 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 12: Received COMMAND_CLASS_ALARM V4 NOTIFICATION_REPORT
2019-08-10 21:03:48.925 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 12: NOTIFICATION report - 0 = 0, event=8, status=255, plen=0
2019-08-10 21:03:48.931 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 12: Alarm Type = BURGLAR (0)
2019-08-10 21:03:48.930 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 09 00 04 00 05 03 25 03 FF 2D
2019-08-10 21:03:48.936 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 12: Got an event from Z-Wave network: ZWaveAlarmValueEvent
2019-08-10 21:03:48.952 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 12: Got a value event from Z-Wave network, endpoint = 0, command class = COMMAND_CLASS_ALARM, value = 255
2019-08-10 21:03:48.958 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 12: Alarm converter processing NOTIFICATION
2019-08-10 21:03:48.963 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 12: Alarm converter NOTIFICATION event is 8, type OnOffType
2019-08-10 21:03:48.968 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 12: Alarm converter processing NOTIFICATION
2019-08-10 21:03:48.973 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 12: Alarm converter NOTIFICATION event is 8, type OnOffType
2019-08-10 21:03:48.975 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=5, callback=0, payload=00 05 03 25 03 FF
2019-08-10 21:03:48.979 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 12: Commands processed 1.
2019-08-10 21:03:48.989 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 12: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@d2dfed.
2019-08-10 21:03:48.996 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2019-08-10 21:03:49.001 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
--
2019-08-10 21:04:19.418 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=12, callback=0, payload=00 0C 0A 71 05 00 00 00 FF 07 00 01 08
2019-08-10 21:04:19.434 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction TID 146: [WAIT_REQUEST] priority=Poll, requiresResponse=true, callback: 145
2019-08-10 21:04:19.443 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 12: Application Command Request (ALIVE:DONE)
2019-08-10 21:04:19.452 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 12: resetResendCount initComplete=true isDead=false
2019-08-10 21:04:19.461 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 12: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
2019-08-10 21:04:19.468 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 12: SECURITY NOT required on COMMAND_CLASS_ALARM
2019-08-10 21:04:19.473 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 12: Received COMMAND_CLASS_ALARM V4 NOTIFICATION_REPORT
2019-08-10 21:04:19.477 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 09 00 04 00 05 03 25 03 00 D2
2019-08-10 21:04:19.478 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 12: NOTIFICATION report - 0 = 0, event=0, status=255, plen=1
2019-08-10 21:04:19.489 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 12: Alarm Type = BURGLAR (0)
2019-08-10 21:04:19.495 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 12: Got an event from Z-Wave network: ZWaveAlarmValueEvent
2019-08-10 21:04:19.503 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 12: Got a value event from Z-Wave network, endpoint = 0, command class = COMMAND_CLASS_ALARM, value = 255
2019-08-10 21:04:19.505 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=5, callback=0, payload=00 05 03 25 03 00
2019-08-10 21:04:19.511 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 12: Alarm converter processing NOTIFICATION
2019-08-10 21:04:19.521 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 12: Alarm converter NOTIFICATION event is 0, type OnOffType
2019-08-10 21:04:19.526 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 12: Updating channel state zwave:device:980c6d82:node12:alarm_burglar to OFF [OnOffType]
2019-08-10 21:04:19.533 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 12: Alarm converter processing NOTIFICATION
2019-08-10 21:04:19.540 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 12: Alarm converter NOTIFICATION event is 0, type OnOffType
2019-08-10 21:04:19.545 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 12: Commands processed 1.
2019-08-10 21:04:19.550 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 12: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@33a9f5.
2019-08-10 21:04:19.554 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 1
2019-08-10 21:04:19.560 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 1
```

cdjackson, 2019-08-10:
Possibly this is caused by the channel being burglar and not motion, but please advise what version you are using first.

JC-85, 2019-08-10:
Sorry, I missed to include the version-info.

* OH2 Version: openHAB 2.4.0 Release Build
* ZWave Binding: 2.4.0

cdjackson, 2019-08-10:
I will update the binding tomorrow which will hopefully resolve this. Please provide a complete and detailed debug log showing the initialisation.

JC-85, 2019-08-10:
I removed and re-added the Thing and have attached the log-file. The device in question is "NODE 12".

[openhab.log](https://github.com/openhab/org.openhab.binding.zwave/files/3489365/openhab.log)

cdjackson, 2019-08-11:
I have not yet updated the binding. I will do that later today, so please do the update and grab the log on Monday.

cdjackson, 2019-09-15:
Can this be closed?

cdjackson, 2019-09-15:
This should be resolved, so closing. If there are still issues, please provide further information and reopen.

closed 2019-09-15 by cdjackson