openhab/org.openhab.binding.zwave #1223: Unclear log message

lsiepel, 2019-08-16:
When trying to set this node as failed by HABmin (build 1662):
a message showed up in the log file:

`2019-08-16 18:35:42.231 [ERROR] [essage.ReplaceFailedNodeMessageClass] - NODE 43: Replace failed node failed as node if functioning!
`
Not sure about the root cause yet, if needed i will add another issue or go to the community. But this message is abracadabra to me. Could this log message be improved to tell what is wrong?

cdjackson, 2019-08-16:
It should say ```NODE 43: Replace failed node failed as node is functioning!```.

This means that you tried to set a device as failed, and it was rejected because the device has not failed. This will only happen if you manually did this through HABmin or PaperUI.

lsiepel, 2019-08-16:
Yes that is what i did …. **_trying to set this node as failed by HABmin_**…
Ok, so maybe you can fix the typo some day. Not a big thing.

Why is the binding saying the node is functional, and what does functional mean in this context? 

As a background: this node (Fibaro FGDS002 smoke sensor) is hard reset without proper exclusion. 
So there is no working assocation. Ok in theory the binding doesn't know that the device is hard reset yet so it might think it is fully functional. On the other hand the node was reset several days ago , so the binding should know it is not communicating anymore.

So while typing this comment:
enable DEBUG mode. 
start openHAB, 
tried to remove the device. (log noted failed. Problem 1). 
tried to mark it as failed (log noted: failed because its functional . Problem 2)

Full log:
[zwave.txt](https://github.com/openhab/org.openhab.binding.zwave/files/3510550/zwave.txt)
How can i remove this device from the controller?

cdjackson, 2019-08-16:
The binding is not saying the node is functional - the controller is. The controller has refused to add this to the failed node list as the node is not failed. The controller will not add nodes to the failed node list if it believes they are still working.

lsiepel, 2019-08-16:
Ah, the controller does that stuff. Forgot that there is another actor :-)

Still strange as this message is also shown in the log:
> 2019-08-16 19:46:28.376 [DEBUG] [message.RemoveFailedNodeMessageClass] - NODE 43: Marking node as having failed.

Anyway, i'm stuck. How can i remove the node from the controller as that is the main objective.

cdjackson, 2019-08-16:
This message is logged by the binding when the whole process starts. The binding asks the controller to mark the node as failed - it logs this message. The controller will then return either a success or failure. In your case it returns the failure and states the reason that the device has not failed and the binding logs the other message.

I appreciate that this is a problematic area - some people have used Silabs tools and they work for some and not others - same as doing it through the binding. I’m afraid I can’t really offer a lot more help - sorry.

closed 2019-08-16 by lsiepel