openhab/org.openhab.binding.zwave #1231: Wrong parameters Neo Coolcam Neoems02z

epmsmid, 2019-09-09:
First: Thanks for the great work for the zwave binding!

I have  a Neo Coolcam motion sensor with temperature sensor.
The previous version without temperature sensor had parameters 1 till 10 and 99.
The new version with temparature version has parameters 1 till 12 and 99:
http://manuals-backend.z-wave.info/make.php?lang=en&sku=NEOEMS02Z

Is it possible to add add the device to/change the database?
This is how the device is recognized:
[image]

Regards,

Eric

cdjackson, 2019-09-09:
Hi,
Please find information on updating the database below -:

https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide

If you have any questions, please discuss on the forum where there are a few database administrators who can help

Cheers
Chris

closed 2019-09-09 by cdjackson

epmsmid, 2019-09-10:
Thanks Chris for your quick repsonse.
I'll try the link.
Regards,

Eric