openhab/org.openhab.binding.zwave #1245: Device 1089 Binding Update & Database Issue

bwosborne2, 2019-10-12:
This device need significant binding updates.

"there are significant code changes required to support this device as it uses a new command class."
from https://community.openhab.org/t/aeotech-doorbell-6/79278/42?u=bruce_osborne

Also, since the database maintenance, this device is no longer visible in the summary.

bwosborne2, 2019-10-12:
Here is the xml from OH.

>  <node>
>   <homeId>0xc9b5af4d</homeId>
>   <nodeId>7</nodeId>
>   <version>4</version>
>   <manufacturer>0x371</manufacturer>
>   <deviceId>0xa4</deviceId>
>   <deviceType>0x103</deviceType>
>   <listening>true</listening>
>   <frequentlyListening>false</frequentlyListening>
>   <routing>true</routing>
>   <security>false</security>
>   <beaming>true</beaming>
>   <maxBaudRate>40000</maxBaudRate>
>   <sleepDelay>1000</sleepDelay>
>   <nodeInformationFrame>
>     <commandClass>COMMAND_CLASS_ZWAVEPLUS_INFO</commandClass>
>     <commandClass>COMMAND_CLASS_VERSION</commandClass>
>     <commandClass>COMMAND_CLASS_CONFIGURATION</commandClass>
>     <commandClass>COMMAND_CLASS_MANUFACTURER_SPECIFIC</commandClass>
>     <commandClass>COMMAND_CLASS_ASSOCIATION_GRP_INFO</commandClass>
>     <commandClass>COMMAND_CLASS_ASSOCIATION</commandClass>
>     <commandClass>COMMAND_CLASS_POWERLEVEL</commandClass>
>     <commandClass>COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION</commandClass>
>     <commandClass>COMMAND_CLASS_MULTI_CHANNEL</commandClass>
>     <commandClass>COMMAND_CLASS_DEVICE_RESET_LOCALLY</commandClass>
>     <commandClass>COMMAND_CLASS_TRANSPORT_SERVICE</commandClass>
>     <commandClass>COMMAND_CLASS_SECURITY</commandClass>
>     <commandClass>COMMAND_CLASS_SECURITY_2</commandClass>
>     <commandClass>COMMAND_CLASS_SUPERVISION</commandClass>
>     <commandClass>COMMAND_CLASS_FIRMWARE_UPDATE_MD</commandClass>
>     <commandClass>COMMAND_CLASS_ALARM</commandClass>
>   </nodeInformationFrame>
>   <associationGroups class="concurrent-hash-map">
>     <entry>
>       <int>1</int>
>       <associationGroup>
>         <index>1</index>
>         <maxNodes>0</maxNodes>
>         <associations/>
>       </associationGroup>
>     </entry>
>     <entry>
>       <int>2</int>
>       <associationGroup>
>         <index>2</index>
>         <maxNodes>0</maxNodes>
>         <associations/>
>       </associationGroup>
>     </entry>
>     <entry>
>       <int>3</int>
>       <associationGroup>
>         <index>3</index>
>         <maxNodes>0</maxNodes>
>         <associations/>
>       </associationGroup>
>     </entry>
>     <entry>
>       <int>4</int>
>       <associationGroup>
>         <index>4</index>
>         <maxNodes>0</maxNodes>
>         <associations/>
>       </associationGroup>
>     </entry>
>     <entry>
>       <int>5</int>
>       <associationGroup>
[... 995 lines cut ...]
>               <control>false</control>
>               <versionSupported>8</versionSupported>
>               <alarms/>
>               <v1Supported>true</v1Supported>
>               <isGetSupported>true</isGetSupported>
>             </COMMAND__CLASS__ALARM>
>           </entry>
>           <entry>
>             <commandClass>COMMAND_CLASS_ASSOCIATION</commandClass>
>             <COMMAND__CLASS__ASSOCIATION>
>               <endpoint reference="../../../.."/>
>               <version>2</version>
>               <instances>1</instances>
>               <control>false</control>
>               <versionSupported>2</versionSupported>
>               <maxGroups>2</maxGroups>
>             </COMMAND__CLASS__ASSOCIATION>
>           </entry>
>         </supportedCommandClasses>
>       </endPoint>
>     </entry>
>   </endpoints>
>   <nodeNeighbors/>
>   <lastReceived>2019-08-04 23:57:41.106 UTC</lastReceived>
> </node>

cdjackson, 2019-10-12:
As I sort of indicated earlier, please can you make these issues standalone rather than just referencing messages in the forum. It might seem fine now, but it takes a lot of time later to work out what the issue is, and it would really be appreciated if the actual issue, and the required change could be added to the issue list rather than a link to a long conversation.

Thanks.

bwosborne2, 2019-10-12:
Is there anywhere I can attach the xml file for use after the PR ?

bwosborne2, 2019-10-12:
Is there any way I can tell what classes the binding recognizes to determine what is missing? Your original posts merely said, 

"It looks like the device is using a new device class that needs to be added to the binding before the device can work. I will look into this tonight if jetlag doesn’t get the better of me!"

and later

"The XML file will not be created until I add the new device class - as mentioned in my previous message."

I can compare with the manual if I know where to look. Manual for Siren 6 ( the same device) is [here.](https://s3.amazonaws.com/cdn.freshdesk.com/data/helpdesk/attachments/production/6086177008/original/SPEC-ZW164%20Engineering%20Specification%20-%20Indoor%20Siren%206%20-%20Aeotec%20-%202019.03.01.pdf?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAJ2JSYZ7O3I4JO6DA%2F20191012%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20191012T223625Z&X-Amz-Expires=300&X-Amz-Signature=dd25854d44f53722d02a031e20737d1de6dd09b53f770f4dbcbc4c8823b7867a&X-Amz-SignedHeaders=Host&response-content-type=application%2Fpdf)

bwosborne2, 2019-10-12:
From the docs, this appears to be the command classes for this device. I am not sure which one is missing.

GENERIC_TYPE_AV_CONTROL_POINT
SPECIFIC_TYPE_SOUND_SWITCH
COMMAND_CLASS_ZWAVEPLUS_INFO
COMMAND_CLASS_SECURITY
COMMAND_CLASS_SECURITY_2
COMMAND_CLASS_SUPERVISION
COMMAND_CLASS_ASSOCIATION
COMMAND_CLASS_ASSOCIATION_GRP_INFO
COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION
COMMAND_CLASS_NOTIFICATION
COMMAND_CLASS_SOUND_SWITCH

kennetn, 2020-01-03:
I was thinking about trying to implement this, but is there a good place to start when trying to understanding the code / design ? I have been looking at the comments in the code but I was just thinking if there was a better place to start?

bwosborne2, 2020-01-03:
@cdjackson is the expert on this code. I am just a user that has been helping with the device database.

cdjackson, 2020-01-04:
@kennetn please define what channels you see as being required and I will try and look at coding up the command class soon.

kennetn, 2020-01-04:
@cdjackson thanks for your help.
When trying to figure out the channels and understanding how the device worked I was looking at the zwave debug log. I could see that the door bell was sending a notification in addition to the COMMAND_CLASS_SOUND_SWITCH when I activated the ring button. I found that the notification was using the alarmtype 14 (siren) that was not supported and causing the alarm to be null.
My initial quest was just to get the notification working when the ring button was activated, so I was just about to open a new issue when I found that you just added alarmtype 14 today in https://github.com/openhab/org.openhab.binding.zwave/commit/3d6c4ebf292a217222a3fbb0f035afc5141adca0. So maybe this will make the notification work when the ringer is activated.
It is a bit late now to be playing with the door bell but I will try it out tomorrow.

Regarding the channels for the COMMAND_CLASS_SOUND_SWITCH I am still trying to get a understanding of how this command class works.
I am a bit unsure how you normally handle information that requires a zwave command to be queried. For instance the number of tones supported by a node can be queried with a command "Sound Switch Tones Number Get Command" so it would be possible to create a channel with this information. But I guess that will require that the binding always query the node regardless if the channel is used or can you see if a channel is using in the binding ?

The primary channels of interest would be to set tone and volume but I will look more at this tomorrow.

cdjackson, 2020-01-04:
> My initial quest was just to get the notification working when the ring button was activated, so I was just about to open a new issue when I found that you just added alarmtype 14 today in 3d6c4eb <https://github.com/openhab/org.openhab.binding.zwave/commit/3d6c4ebf292a217222a3fbb0f035afc5141adca0>. So maybe this will make the notification work when the ringer is activated.

Thanks. Please can you provide the full information about the alarm that is produced - alarm type 14 does not provide enough information - I need to also know the event IDs that are sent.

If this is the main thing that is needed to get some functionality for this device, if you let me know the information above I will add this to the binding tomorrow.

Thanks
Chris

bwosborne2, 2020-01-04:
Do we need to import the xml file again into the database? I believe I still have the original files.

cdjackson, 2020-01-04:
No - importing the XML won’t help at this point. Once the XML has been imported once, the only changes that can occur by importing another XML is to update the type/ids.

kennetn, 2020-01-04:
I added another issue with the missing support for alarmtype 14 (siren) . #1284

cdjackson, 2020-01-04:
Thanks. I’ll try and have that added shortly.

kennetn, 2020-01-11:
Hi @cdjackson 

I Finally had a bit of time to look at this command class.
I only looked at version one of the command class so far.

What I can see is that the commands supported can be divided in 4 groups.
1. Number of tones that has 2 Commands
2. Get tone information that has 2 Commands
3. Change volume and tone that has 3 Commands
4. Start playing tones that has 2 Commands.

### 1. Number of tones
This is a simple value that is read from the device.
I suggest a channel called.
```
1-1. "Number of tones" with a number ( read only )
```
I am a bit unsure how you normally handle these very static channels. Is it possible that you can just read the value once after the node is connected or would you need to have another channel to indicate that the value should be read?

### 2. Get tone information
For these I would suggest one channel called
```
2-1. "Get tone information" with a number
```
This channel should be updated with the tone number that is then returned in the three channels called.
```
2-2. "Tone Identifier" with a number
2-3. "Tone Duration" with number that is playing time in sec
2-4. "Tone Name" with string
```
These three values are received in the same message so I guess that it would be most easy to use if the Identifier channel is updated last.
I can see that using this is a bit of protocol on top of protocol and I am unsure if this is the "normal" way to handle this? 

### 3. Change volume and tone
The set command can be send individually for volume and tone.
Therefore I suggest two channels
```
3-1. "Default tone" with number
3-2. "Volume" with number that is percent from 0 to 100
```
Again I am a bit unsure when the channel should be updated with status from the device I would guess this only needs to be done after a change is send to the device with a set command.

### 4. Start playing tones
I would suggest a channel called.
```
4-1. "Play tone" with a number or the value "DEFAULT" or "OFF"
```
When this channel is updated the device starts playing. Maybe you could also use DEFAULT when any value that can not be converted to a number is received. 
I do not see a reason to implement the Get and Report commands because the alarm report command class is more useful for this.

Was this the type of information you were looking for ?
For me and I guess many others the 1 and 2 is not so useful so in a first version it would be fine to skip these.

bwosborne2, 2020-01-12:
This is used for 2 variants of a device. One is an alert siren and the other is a doorbell.

chrismcgarry, 2020-01-16:
Don't know if this helps but here's the engineering spec for the doorbell. I've had this device since July so keen to help out to get it working.

There's quite a lot of info in the doorbell engineering spec and examples too is that what you need @cdjackson?
[SPEC-ZW162 Engineering Specification - Doorbell 6 - Aeotec - 2019.03.01.pdf](https://github.com/openhab/org.openhab.binding.zwave/files/4074025/SPEC-ZW162.Engineering.Specification.-.Doorbell.6.-.Aeotec.-.2019.03.01.pdf)

cdjackson, 2020-01-16:
Thanks, but the spec here is not needed.

I thought that the doorbell now worked after the notification was added a few weeks back?

kennetn, 2020-01-17:
> I thought that the doorbell now worked after the notification was added a few weeks back?

Yes the notification part is working with 2.5.1 (thanks a lot). 
However the command_class_sound is still needed because it is not possible to adjust the ring volume without the class and the default volume is really really loud. I got a round this by using some other home automation software (hoomseer), but for most users this is not really a usable solution.
In addition I would also like to use the door bell as a siren for my alarm system and this also requires the command_class_sound. Aeotec also sells a siren sepperatly that I guess also use this command class that will need this calls.
I think most users of the device will need to have the channels 3-2 and 4 that I described above.

cdjackson, 2020-01-17:
Sure - I understand that there's functionality that's not supported - I was just responding to the comment that "this device isn't working" where I think the primary function does work.

It's not about not knowing what I'm doing, or looking for information - the issue is simply time. I'm currently working through some autogenerated command classes - once that is done I will use that generated code to provide some of the functions in this class.

chrismcgarry, 2020-01-17:
Is there a steep learning curve, anything I could do to help? I've done plenty of Java development in the past and I'm very grateful for everything you've done @cdjackson so maybe time for me to try and contribute back.

cdjackson, 2020-01-17:
> Is there a steep learning curve

Yes.

However the other point here is that I don't want to waste my time doing something half baked. There are major changes required for the binding over the coming months, and to support that I need to do some background work (which I'm doing now). Once that is done I can use that to generate what's needed here.

Hopefully in a week or two I'll have that done and can look at this (hopefully it will be reasonably easy then).

bwosborne2, 2020-01-17:
It is my understanding the siren is basically the doorbell packaged without the buttons. They sell them separately to turn the siren into a doorbell.

Nadahar, 2020-02-22:
> It is my understanding the siren is basically the doorbell packaged without the buttons. They sell them separately to turn the siren into a doorbell.

This is my understanding as well. I have the "Siren 6" only, and right now I can't do anything with it. As far as I understand, "Doorbell 6" is "Siren 6" + one wireless (non z-wave) button. These buttons can also be bought separately and it is possible to "pair" up to 3 such buttons to one siren. The "button functionality" isn't really related to z-wave (or openHAB), since this is a (closed) system between the siren and the buttons. The only thing about it that concerns z-wave apparently, is that you can get a report sent when the "doorbell function" is activated (by someone pushing the button).

As a siren it doesn't seem like it can do anything at the moment using openHAB. As far as I can understand, the "sound switch" command is what is needed to be able to tell the siren to do anything at all.

yehudamm, 2020-03-27:
Hi,

Are there any updates on supporting aeotec doorbell in openhab?

Thank you

bwosborne2, 2020-03-27:
I know Chris is working on totally rewriting the Z-Wave binding.  I am helping him with some of the back-end programming changes for the new database.
Like most managers I suspect he is especially busy with his day job with people working from home.

kennetn, 2020-04-18:
> I know Chris is working on totally rewriting the Z-Wave binding. I am helping him with some of the back-end programming changes for the new database.

Do you know where this development is taking place? I can not find any branch that looks like it is being worked on

bwosborne2, 2020-04-18:
My understanding is that the current binding was developed bu reverse engineering device behaviour. Chris is working on developing a binding based on the published standards, likely in a private repository. It is expected to be quite some time before that would be usable.

In addition, I an assisting in moving the Z-Wave device database to a website with a different host and framework. It too is incomplete and unusable at the present time. I am working through some code migration issues.

kennetn, 2020-06-05:
Hi @cdjackson 
Is it possible that you can comment on the status of this "total rewrite" process?
I was starting to look at implementing this but I do not want to do a lot of work if a rewrite is just around the corner. Maybe I can help with the rewrite instead ?
Sorry if it sounds like I am pushing, that is not my intention I know we are all volunteers.

RudiTholen, 2020-08-21:
Any news on the ZW164 (Aeotec Siren 6) to get it working with openhab?
With ‘Z-Wave PC Controller 5’ I can let the ‘siren 6’ make all sounds at all volume levels, which I desire (from my PC) but with ‘openhab’ on my PI, I can only ‘monitor’ but not control the siren!

Is there any hope to get it working in OpenHab, or am I missing an important clue?
Rudi Tholen

yehudamm, 2020-11-10:
@cdjackson 

Hi Chris,,

Could you update us on when we can expected to get Openhab control the Siren of this device?
This is the main functionality of this device and I believe most of us are looking to operate it.

Thank you

JimCzuprynski, 2020-12-12:
@cdjackson -

Chris, though I'm a noob with Openhab, I'd be more than happy to help out with building + testing the interface to the ZW164 doorbell and siren. It'd be nice to get this working so that we could control the siren via rules.

JimCzuprynski, 2020-12-12:
> I made a beta version that allows playing tones and setting the volume from rules. I put up a complied version in [my fork](https://github.com/kennetn/org.openhab.binding.zwave/releases). There are other issues with it, but for what I wanted to achieve it works. Please note you should not expect support for this version.

So for us noobs: what do we need to do to implement your solution, please? I know it's not supported but it'll be a worthwhile experiment on my part if I can help out the community.

bwosborne2, 2020-12-12:
I believe the current support work is already in the Z-Wave binding. It is only partial though. The full solution requires more work and the binding is due to be rewritten.

JimCzuprynski, 2020-12-13:
Thanks for the response! I've scanned through most of the listings and code here, but to me it's unclear that there's a way to activate any of the doorbell or siren features from within a rule or script. Is that an accurate assessment? Or can anyone point me to an OpenHAB example of actually activating the siren / doorbell in this manner?

kennetn, 2020-12-13:
This is how I configured my items.
Switch alarm_system_siren "Alarm siren" { channel="zwave:aeotec_zw164_00_000:controller:node12:notification_siren1" }
Switch doorbell_ringing "Door ring" { channel="zwave:aeotec_zw164_00_000:controller:node12:notification_siren3" }
Number doorbell_play_tone "Playing tone" { channel = "zwave:aeotec_zw164_00_000:controller:node12:play_tone" }
Dimmer  alarm_volume  "Alarm volume" { channel = "zwave:aeotec_zw164_00_000:controller:node12:volume" }

In the rule I write:
   doorbell_play_tone.sendCommand(17)

chrismcgarry, 2020-12-13:
Is that using your fork @kennetn?

I've had the doorbell sitting in a box now for 1.5 years so would really like to get it working!

JimCzuprynski, 2020-12-13:
Awesome! I saw your previous posts as well – I’ll give this a try. Thanks for the quick feedback!

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

JimCzuprynski, 2020-12-13:
Awesome, @kennetn! I saw your previous posts as well – I’ll give this a try. 

To echo @chrismcgarry tho - do we need to apply the code in your fork?

kennetn, 2020-12-14:
> Is that using your fork @kennetn?
> 
> I've had the doorbell sitting in a box now for 1.5 years so would really like to get it working!

Yes thai is with my fork.

JimCzuprynski, 2020-12-15:
So just a dumb question from a newbie again: Do we just shut down OpenHAB, install your JAR in place of the existing one, and restart? Sorry, didn't see installation instructions.

alamers, 2020-12-15:
I am trying to get @kennetn 's solution working but only partially succeeded so far. Here's what I did. Since I am running 2.5.11, I forked his fork, merged in the 2.5.11 changes and compiled it. But you may as well try the compiled jar from kennetn.

* I placed the jar in /usr/share/openhab2/addons 
* log into the openhab console (something like: `ssh -p 8101 openhab@localhost`)
* find and disable the original 2.5.11 zwave bundle (`bundle:list|grep ZWave` and `bundle:uninstall <id>`) so only the new bundle is active
* Created items for the channels `play_tone`, `volume` and `notification_siren3`
* Created a rule that triggers on some arbitrary button and does: 
```
        Siren6_Volume.sendCommand(5)
        Siren6_PlayTone.sendCommand(17)
```

Findings so far:
* triggering the rule does seem to change the volume and play an alarm sound
* triggering the doorbell button plays a doorbell sound at max volume and haven't been able to change that 
* i also see a lot of nullpointer exceptions in the log:
```
2020-12-15 20:06:58.736 [ERROR] [nal.common.AbstractInvocationHandler] - An error occurred while calling method 'ThingHandler.handleComm
and()' on 'org.openhab.binding.zwave.handler.ZWaveThingHandler@b001ed': null
java.lang.NullPointerException: null
        at org.openhab.binding.zwave.internal.converter.ZWaveSoundSwitchConverter.receiveCommand(ZWaveSoundSwitchConverter.java:89) ~[?:
?]
        at org.openhab.binding.zwave.handler.ZWaveThingHandler.handleCommand(ZWaveThingHandler.java:1181) ~[?:?]

```

nza-dk, 2021-07-02:
I just spend a whole day setting up a build environment for openHAB version 3.1, downloading the source for the z-wave binding, merging @kennetn's changes and getting it all compiled.

I uninstalled the original 3.1.0 binding and copied my compiled jar to the bindingsfolder, re-included my Aeotec Siren 6, and it just works beautifully! 

I get no errors in my z-wave log, no errors in openHAB and it just responds fast and accurate.

Why has this not been added to the official repo?
May I try to submit your work in a pull-request @kennetn?

kennetn, 2021-07-04:
> I just spend a whole day setting up a build environment for openHAB version 3.1, downloading the source for the z-wave binding, merging @kennetn's changes and getting it all compiled.
>
> I uninstalled [...]

Hi

I have No issues with you making a pull-request. I think there are somethings regrading automatic discovery that is not working and status.

brydling, 2021-07-23:
> I just spend a whole day setting up a build environment for openHAB version 3.1, downloading the source for the z-wave binding, merging @kennetn's changes and getting it all compiled.
>
> I uninstalled [...]

Hi @nza-dk,

Any chance you can upload that jar for me to test?

Would be awesome if you want to make a pull request out of @kennetn's work 👍 

Regards,
Niclas

JimCzuprynski, 2021-07-23:
So if this did get fixed, would we be able to include it in the OpenHAB 3.2 beta release? How does all that work, folks?

nza-dk, 2021-07-23:
Sure @brydling, I called it version 3.1.1 to easily identify if it's the right one loaded in openHAB: https://www.dropbox.com/s/551p7y5z4gwhpmc/org.openhab.binding.zwave-3.1.1.jar?dl=1

I haven't gotten around all the requirements for a pull-request, I've never done one to openHAB before...
But yes @JimCzuprynski, if it gets accepted as a pull-request, it gets included in the following beta release.

brydling, 2021-07-26:
@nza-dk 
It works perfectly! Of course the sound configuration settings are still missing (being able to set the default tone and volume for different events) but I guess that's mainly important when using it as a doorbell, not as a siren (to set which tone and volume shall play when the button is pressed).

I actually modified the xml some (database). There were two notifications missing (notification_siren3 tells when button #1 is pressed but notification_siren4 and notification_siren5 was missing so you can't get a notification when button #2 or #3 is pressed). I also changed the names and texts of most channels to make it more clear when mapping the channels to the documentation.

If you want to I can send it to you to include in your pull request? I have a 4 month baby, which is great, but I unfortunately don't have the time to investigate how to make a pull-request on my own 😄

JimCzuprynski, 2021-07-26:
I'm actually upgrading my OpenHAB configuration from 2.5.1 to 3.1.0 this coming week, so I'll be in a good position to help test this when the beta drops. Perhaps I can even help with expanding some of the other functions? I'd really like to be able to use the device as a siren for burglaries, as well as a "tone" when someone opens or closes a door (or forgets to shut the d*mn thing). ;)

nza-dk, 2021-07-27:
@brydling xml updates needs to come from the source database at https://www.opensmarthouse.org/zwavedatabase/1089 🙂
Once (if) the pull-request gets accepted, the channels for playing tones and setting volume will also have to be added to this database.

@JimCzuprynski I did the same upgrade not so long ago, and I can't say it was all straight forward 😄

brydling, 2021-07-27:
@nza-dk ah, didn't know that. Thanks for the info. I have found more problems in the database for this device. Have opened a question in the OpenHAB forums on how to fix it: https://community.openhab.org/t/aeotec-doorbell-6-button-power-notifications-not-working/124921

brydling, 2021-07-27:
@nza-dk 
I checked your pull-request. Nice!
I see you have set the volume using the configuration message instead of the "Play Command Tone Volume" field in the regular "Tone Play Set" command. I think this actually means that it is possible to add this channel to every endpoint in the database, which means it will be possible to set the default volume of each endpoint, but using a channel rather than as a parameter in OpenHAB.
My question is: Would it be possible to also add the default tone parameter (Default Tone Identifier) to the configuration message, maybe using a different TYPE, like TYPE=DEFAULT_TONE instead of TYPE=VOLUME that you have used to set the volume.
So it will be like one channel TYPE=VOLUME that send a config message with payload (vol,0) like you do now and another channel with TYPE=DEFAULT_TONE that send a config message with payload (0,default_tone)?

kennetn, 2022-01-05:
Hi

I have being looking at this again in the hope of creating a more polished version that can be merged. I also found an issue that I have fixed but before I publish this I need to add some unit testing.
It still only supports playing tones and setting the volume.
To do this I have added two new channels in the ...\thing\channels.xml like this:
`    

    
    <channel-type id="sound_tone_play">
        <item-type>Number</item-type>
        <label>Sound tone play</label>
        <description>The Sound tone that will play. The allowed tone numbers depend on the device</description>
        <category></category>
    </channel-type>

    
    <channel-type id="sound_volume">
        <item-type>Dimmer</item-type>
        <label>Sound volume</label>
        <description>The sound volume channel allows control of the loudness</description>
        <category></category>
    </channel-type>
`
But it would be nice if @cdjackson or someone else could comment on if this is the right way to implement this?

bwosborne2, 2022-01-05:
I believe that node-zwave-js written in javascript & typescript supports this command. Perhaps it would be useful to look at their code for hints?

brydling, 2022-01-06:
@kennetn
Nice!
What are your thoughts about my suggestion?
That way the bindning can be used to fully configure the siren as a doorbell and also dynamically change the configuration, for example lower the volume of the doorbell endpoints when the baby is asleep 🙂
For the volumes I think it is just a matter of adding that to every endpoint in the database, but for the tone there is also code updates needed as I suggested above. 

BR
Niclas

kennetn, 2022-01-07:
Hi

I create a new release that I think should be ok to merge.
https://github.com/kennetn/org.openhab.binding.zwave/releases/tag/v3.3.0-v1

I will run it the next few days and then I will create a pull request.
Therefore I suggest to delete you pull request @nza-dk.

I did not add support for setting the default tone. Honestly I do not see much point in that at the moment.

brydling, 2022-01-07:
Hi!

The point is for the other endpoints. That is how you change for example what tone the siren will play when one of the doorbell buttons are pressed, or an alarm is received. Currently this has to be configured using another gateway or with the Z-wave PC Controller application because this bindning does not support it. Setting the default tone for the endpoint used to play tones from OH is pointless as you say.

kennetn, 2022-01-07:
Is it not just because the parameters are incomplete in the z-wave database?
There is a "Tone Index" for parameters 2 to 8 described in the "Engineering Specification - Aeotec Doorbell 6 .pdf".
If that is changed would it not work then ?

brydling, 2022-01-07:
I never noticed this while I read the specification before but now I see it. You are probably right. That should hopefully do the trick!

bwosborne2, 2022-01-07:
> There is a "Tone Index" for parameters 2 to 8 described in the "Engineering Specification - Aeotec Doorbell 6 .pdf".

Where is that document? It is not in out database, I just used the manual for a reference. That was a LONG time ago though,

[23 more replies]

PR #1715 (2022-04-19): Add support for COMMAND_CLASS_SOUND_SWITCH
Adds new COMMAND_CLASS_SOUND_SWITCH needed to be able to trigger siren and volume on Aeotec Siren/Doorbell 6.
This pull request is more polished than the previous one and I am not aware of any bugs. I also tried to keep in line with the rest of the code.

The ZWaveSoundSwitchCommandClass.java is strait forward and made like all the other command classes.
But because the command class supports getting and setting status of both volume and the tone to play / being played.
I needed to differentiate between the two in the ZWaveSoundSwitchConverter.java. I [... cut]
closed 2022-04-19 by cdjackson