openhab/org.openhab.binding.zwave #1247: Support for bridge controllers

pizerg, 2019-10-15:
Since new Z-Wave 700 series controllers only come with BRIDGE firmware and not STATIC, it would be good to add support for them (and maybe for the new S2 security and Smart Start features too).

cdjackson, 2019-10-15:
Clearly this is a sticky area since Silabs are no longer supporting this API at all and have stated that it is not possible to develop or certify any products that use this. It's difficult based on this to know where to take this as documentation is no longer available, and certification is rejected.

cdjackson, 2019-10-31:
> I contacted sigma about the issue and they confirmed that Z/IP Gateway software must be used from now on (that is if the software needs to be certified).
>
> They provide (free access) to the source code which also includes pre compiled binaries for linux and the raspberry pi / beaglebone boards. The only problem, as usual, is that the documentation is scattered through several files and some times incomplete or outdated. They do provide a python and c sample clients (https://github.com/Z-WavePublic/libzwaveip <https://github.com/Z-WavePublic/libzwaveip>) and the PC Controller app also supports Z/IP interface (and the source is available as well).
>
> Haven't thorougly tested / explored it yet (but I will), it should include S2 + Smart Start out of the box and probably simplify the Z-Wave client code / binding as Z/IP software is supposed to be taking care of several things.

Yes, I’m fully aware of all of this. However the binding will require a major rewrite to use Z/IP (as I’ve said - I just can’t remember if it was in this issue or another one). It’s very messy to use the Z/IP software and I think it’s a major step backwards to the “bad old days” of ZWave being a closed protocol. I’ve unfortunately not had any success in convincing Silabs of this though :(

cdjackson, 2020-01-17:
This requirement will be reversed in future so we'll look to support this when the binding is rewritten.

Fishwaldo, 2020-02-20:
I've managed to Reverse engineer the Bridge Serial API - Essentially only two major changes as far as I can see:
FUNC_ID_SEND_DATA - Change to FUNC_ID_SEND_DATA_BRIDGE and add a Source Node ID to preamble and
FUNC_ID_APPLICATION_COMMAND_HANDLER - Change to FUNC_ID_APPLICATION_COMMAND_HANDLER_BRIDGE which just contains the "destination" node ID in the preamble.

You can watch https://github.com/OpenZWave/open-zwave/projects/2 as I push in changes to OZW to support it.

Simple enough really. Thought I'd share with you guys.

Fishwaldo, 2020-02-20:
Seems you already have half the work done in #821

cdjackson, 2020-02-20:
Thanks. It’s not really about the protocol - I have this working here also. The big push these days (well for the past year or so) is certification and up until the recent announcements that has only been possible with Z/IP so I’ve not progressed this.

Fishwaldo, 2020-02-21:
We are pushing on the OZW side on this, as there are sticks hitting the market now with 700 Series controllers and they are bridge firmware only.

Never been about the certification for me - Although SiLabs have upped their cooperation in the last 2 months or so to help me out where they can... with their goal of Certification :)

cdjackson, 2020-02-21:
Unfortunately I have to sort out certification so it requires a different approach. 

Yes, I saw an email from you on the OZW list about talking with someone at Silabs. My guess is we are talking to the same chap :)

pizerg, 2020-03-28:
Hello @cdjackson 

I've been following some comments on openhab's forums and here on some open issues and I was wondering if you could clarify some details about the future of ZWave binding.

From what I understood, the binding is being rewritten and it seems that the main objective is certification:

* Last time I spoke with Sigma, they seemed pretty adamant on Z/IP Gateway software being mandatory for 700 series certification (500 series can still be certified on serial API as you already know), but I understood from some of your comments that Z/IP Gateway requirement might be dropped for certification in the near-mid future. Do you have any news or comments on that coming from Silabs?

* From some other comments I assume the binding is being rewritten without using Z/IP Gateway software, could you summarize a bit what's being changed or what's "wrong" with the current implementation, if I'm not mistaken most issues regarding certification right now have to do with S2 support (including Transport Service encapsulation) and assuming roles different from primary controller (when being added on another controller's network as Seconary, Inclusion or SIS).

Thanks and have a nice weekend!

cdjackson, 2020-03-28:
Z/IP will not be required in future - given that other companies don't support it they can't make this mandatory. There is no way the binding will get through certification - there are just too many old classes etc.

S2 is not required for certification against 500 series chipsets. My current need is to target a system using the 500 series.

At the moment, there are no guidelines for the new certification program. Clearly it will change a lot, but my last telecon with SiLabs they were still thinking about this, and still thinking about what API they will support. Clearly with the opening of the standard there's going to be a lot of change.

The last I spoke to the guys doing this, it was planned for late summer (slipping a little from "middle of the year"). That was 3 weeks ago, and I suspect that a lot has changed since then. We have been starting the certification paperwork, and I know from this that many of the Silabs people are in Singapore and this impacted them a few weeks back - now that Europe and the US is also affected, I could expect this to slip to the end of the year (but I'm guessing here!).

pizerg, 2020-03-30:
Hi thanks for the clarification,

I will contact with Silabs to see if I can get first hand information as it's not in line with what they told us last time.

Regarding 500 series certification, unless something strange happened or they lifted the restriction specially for openhab, S2 is a strict requirement for new certifications (but SmartStart is optional) as can be seen in both, silabs website and in the certification website (not sure if you already have access to this or not).

[Source](https://www.silabs.com/products/wireless/mesh-networking/z-wave/certification)
> All new 500 series Z-Wave Plus devices MUST be based on SDK v6.81.03 or newer and implement S2 Security. Only active SDKs published by Silicon Labs can be used for certification of new devices